kizniche/Mycodo #275: Mycodo hangup

reymart081678, 2017-07-20:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.49
 
#### Problem Description
Please list:

 Activate relay using timer with time interval of 50s on and 10 seconds off. ( exhaust fan)
 No sensors only the timer. RPI 2B 
 Activated daemon in debug mode to determine the cause of error.

 
### Errors
2017-07-21 01:52:08,242 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:52:18,053 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:52:18,154 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:53:08,182 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:53:18,055 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:53:18,222 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:54:08,322 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:54:18,082 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:54:18,238 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:55:08,324 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:55:18,106 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:55:18,249 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:55:18,733 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:54:18.336542Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:56:08,296 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:56:18,147 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:56:18,311 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:56:18,513 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:55:18.309603Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:56:18,652 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:55:18.314227Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:57:08,347 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:57:18,149 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:57:18,356 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:57:18,613 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:56:18.392893Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:57:18,616 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:56:18.339763Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:58:08,377 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:58:18,200 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:58:18,318 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:58:18,593 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:57:18.380139Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:58:18,750 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:57:18.386872Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:59:08,441 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 01:59:18,229 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 01:59:18,460 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 01:59:18,762 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:58:18.450653Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 01:59:18,940 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:58:18.491156Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-21 02:00:08,552 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-21 02:00:18,320 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-21 02:00:18,464 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) on for 50.0 seconds.
2017-07-21 02:00:18,817 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-20T17:59:18.553120Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-07-20:
Looks like influxdb is not running

reymart081678, 2017-07-20:
Hi Kizniche,
   I saw a restart script if daemon is not running, can we have also for influxdb to ensure that it is running?
it is repetitive.

kizniche, 2017-07-20:
Can you confirm the influxd process is indeed not running?

```ps aux | grep influxd```

reymart081678, 2017-07-20:
The rpi is not responsive when i checked in the morning, so i restarted it already before sending the issue

kizniche, 2017-07-21:
InfluxDB has a resolved [issue](https://github.com/influxdata/influxdb/issues/8500) that was included version 1.3.1, currently unreleased ([changelog](https://github.com/influxdata/influxdb/blob/master/CHANGELOG.md)). I suspect this is the issue. I actually had this issue happen to me just now.

kizniche, 2017-07-21:
Perhaps downgrading to 1.2.4 might fix the issue until they make their next release with the fix included.

kizniche, 2017-07-21:
```bash
wget --quiet https://dl.influxdata.com/influxdb/releases/influxdb_1.2.4_armhf.deb
sudo dpkg -i influxdb_1.2.4_armhf.deb
rm -rf influxdb_1.2.4_armhf.deb
service influxdb restart
```

kizniche, 2017-07-21:
The downgrade worked for me (that is, Mycodo is still writing and reading from the database correctly).

reymart081678, 2017-07-21:
Hi kizniche,
  Thank you.  I'll downgrade it later.

kizniche, 2017-07-23:
I just noticed that influxdb 1.3.1 was released 3 days ago (without updating their changelog for me to notice). I just upgraded and it appears to be working. I used the following to upgrade:

```bash
wget https://dl.influxdata.com/influxdb/releases/influxdb_1.3.1_armhf.deb
sudo dpkg -i influxdb_1.3.1_armhf.deb
sudo service influxdb restart
rm -rf influxdb_1.3.1_armhf.deb
```

reymart081678, 2017-07-24:
Hi Kizniche,
   Okay, i already upgraded to 1.3.1 and observe the response, it's been running 1.2.4.1 without any hangup issue since the downgrade. let me observe for few days.

reymart081678, 2017-07-25:
Hi Kizniche,
   I will revert to 1.2.4.1, 1.3.1 hangup in less than 24 hours.
2017-07-25 17:21:28,244 - mycodo.relay - DEBUG - Relay 5 (exhaustFan) turned off.
2017-07-25 17:22:20,487 - mycodo.timer_1 - DEBUG - Turn relay a807e1f3-01a0-4cb0-bfa2-a9aae06c960f on for 50.0 seconds, then off for 10.0 seconds
2017-07-25 17:31:36,042 - mycodo.influxdb - DEBUG - Failed to write measurement to influxdb (Device ID: a807e1f3-01a0-4cb0-bfa2-a9aae06c960f). Data that was submitted for writing: [{'fields': {'value': 50.0}, 'tags': {'device_id': u'a807e1f3-01a0-4cb0-bfa2-a9aae06c960f'}, 'time': '2017-07-25T09:20:28.989072Z', 'measurement': 'duration_sec'}]. Exception: {"error":"timeout"}

2017-07-25 18:00:32,809 - mycodo.timer_22 - DEBUG - At 17:00, turn relay 9d5ab9fc-e6d3-41d0-8fe2-e501dc76dc29 on
2017-07-25 18:00:24,357 - mycodo.timer_24 - DEBUG - At 18:00, turn relay 9d5ab9fc-e6d3-41d0-8fe2-e501dc76dc29 on
2017-07-25 18:00:36,692 - mycodo.relay - DEBUG - Relay 2 (humidifierFan) ON at 2017-07-25 18:00:36.620275.
2017-07-25 18:00:39,834 - mycodo.timer_31 - DEBUG - Relay ef15a342-0ac4-453e-b6ab-8b25ee003925 should be on, but is off. Turning on.
2017-07-25 18:00:43,484 - mycodo.relay - WARNING - Relay 2 (humidifierFan) is already on.
2017-07-25 18:00:44,297 - mycodo.relay - DEBUG - Relay 3 (humidifierSupply) ON at 2017-07-25 18:00:43.939820.
2017-07-25 18:01:35,853 - mycodo.relay - DEBUG - Relay 1 (humidifier) ON at 2017-07-25 18:01:35.811704.
2017-07-25 18:01:37,418 - mycodo.relay - DEBUG - 2017-07-25 18-01-35
[Relay Conditional 1] humidityCntrlOn
If relay 2 (humidifierFan) turns on, Then turn relay 1 (humidifier) on.

kizniche, 2017-07-25:
I only see one influx error. If there is only one, there's probably nothing to worry about.

reymart081678, 2017-07-25:
The problem is that the mycodo is so slow to respond through web and ssh, need to hard reset it

kizniche, 2017-07-25:
Sounds like it may be a network issue. My Pi 2s and 3s are very responsive, even with 10+ sensors and multiple PWM and relays in use, while streaming logs via SSH. Only my Pi 1 is slow for me.

reymart081678, 2017-07-25:
No, after the hard reset , RPI is back to normal speed

kizniche, 2017-08-09:
I believe this issue was the bad Influxdb version, and it has been resolved.

closed 2017-08-09 by kizniche