kizniche/Mycodo #276: SHTx2 and SHTx7 share same library ?

wangine, 2017-07-22:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 Mycodo 5.0.49
 Mycodo 4.0.22
#### Problem Description
Please list:
I try just to understand what going wrong with my sensors 
### Errors
Daemon crash randomly , 2h 10h , some days.
### Steps to Reproduce the issue:
How can this issue be reproduced?
Just to activate SHT75 sensor and daemon will crash very soon. Also if i puled out the sensor
daemon crash very soon. Can be something wrong with my PI ?
### Additional Notes
First of all , i need my controled room on that period and i say to install a fresh raspberry jessie image. That was done easy, but when i try to install mycodo i just get a error. The last Jessie image have a update mirror bug , so Mycodo cant update the RPi system and fail trying to fech some archive . Easy solution was to edit the hosts file .
`
One way is to add following lines to /etc/hosts file

Open /etc/hosts file with sudo rights $sudo nano /etc/hosts
Paste the following lines at the end of /etc/hosts file
93.93.128.193   mirrordirector.raspbian.org

93.93.128.191   archive.raspbian.org

sudo apt-get update -y && sudo apt-get upgrade -y && sudo apt-get autoremove -y
`
So far so good. OK , i try to install again , without delete Mycodo folder first.
I found that in install log , so i doit 
`To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove
$sudo reboot
`
I just notice , cant acces SSH anymore , so i activated again with raspi-config.
After reboot i get succes on install. 
Now going to test sensors. I just install several versions of Mycodo , until i found on any version, daemon will crash on some randomly time after i activate the SHT75 sensor (i want to say , that never happen on thats versions of Mycodo). Sience only SHT75 was attached to my board (other sensors was pulled out) , i just pull out from my board the SHT75 sensor to made a close inspection on him connections and i pull in the sht25 sensor (on different male conector, each sensor have unique conector, just do dont mess up in mistake) , so SHT75 are on GPIO26 and CKL 19 / sht25 are on native I2C pins conected.
I return in the room to check the web interface and i notice something weird, A previously SHT75 activated sensor on web interface show the measurement data on live page (not real one , almost 15 *C less than should be) collecting data of a sht25 atached to the board. of course on different RPi pins. 
How that is possible. I know the SHT75 dont have a native I2C protocol but also i know in my experience 
is almost the same protocol. It is possible to both sensors to share same librarry and daemon to mistake? Maybe that can be the crash reason because i inspected the SHT75 sensor and i tested to other board (dsPIC33F) and works perfectly. Also i'm sure is not Mycodo fault , but can be a bug or sometrhing ?

wangine, 2017-07-22:
Also my confussion . A activated SHT75 sensor (and hes missig on hardware ) pin 26 and CLK 19 ,
on that picture is only atached a sht25 pins 5 and 3 (I2C pins) .
[image]

[image]
Any sugestion are welcome , i'm realy frustrated after 50 images flashed .

kizniche, 2017-07-23:
> Just to activate SHT75 sensor and daemon will crash very soon. Also if i puled out the sensor
daemon crash very soon.

Is it only the SHT75 sensor that causes your daemon to crash? Not also the SHT2x?

I went back as far as version 4.0.26 and that version of Mycodo uses the same code for the SHT1x7x and SHT2x sensors as the latest version (5.0.49).

I also tried to add the SHT1x7x sensor with pins not connected to anything and got the expected error:

```
2017-07-22 22:27:20,897 - mycodo.sensors.sht1x_7x - ERROR - SHT1x7xSensor raised an exception when taking a reading: Command ACK failed on step-2
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht1x_7x.py", line 111, in read
    self._temperature) = self.get_measurement()
  File "/var/www/mycodo/mycodo/sensors/sht1x_7x.py", line 96, in get_measurement
    temperature = sht_sensor.read_t()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sht_sensor/sensor.py", line 300, in read_t
    t_raw = self._get_meas_result(self.cmd.t)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sht_sensor/sensor.py", line 240, in _get_meas_result
    self._send(cmd)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sht_sensor/sensor.py", line 203, in _send
    raise ShtCommFailure('Command ACK failed on step-2')
ShtCommFailure: Command ACK failed on step-2
```

kizniche, 2017-07-23:
If you have other Python code that works with the SHT1x7x sensors, I'll put that in the sensor module and see if it fixes your issues.

kizniche, 2017-07-23:
> How that is possible. I know the SHT75 dont have a native I2C protocol but also i know that in my experience it's almost the same protocol. It is possible for both sensors to share the same library and daemon to mistake?

I have no idea how that is happening with your sensor. I don't have any o these sensors to test, and the sht2x module uses a data and clock GPIOs while the sht1x7x module uses I2C.

wangine, 2017-07-23:
No no no  , sht25 have a native I2C protocol and SHT75 have clk and data , but in teory is almost like the same type of comunication as I2C. Anyway that's sensors mess me up in last days. Let made a deal , send me on my email wangine.tb@gmail.com your adress and i will buy for you thats sensors and i will send you on your adress at home. They are the best and accurate sensors ever and you deserve for your support here.
I dont know , i made a complex hardware projects in last 20 years but that "weird" errors made me to fell my self like a dumb. I change the RPi board, sensors (both) , Mycodo, Jessie full, lite . Everything are OK. Single board i didn't check is the interface between relays and RPi, because are full isolated with optocuplers. I will take a look tomorow , maybe water sleeps there where i didn't see.

wangine, 2017-07-23:
Also i remember long time ago , i made my own driver for SHT2x series for Microchip in C and that code can read corectly even the SHT7x sensors, but the driver maded for SHT75 can't read SHT2x series.
I remember for sure , they use almost same code , SHT75 is more complex , can take a look 
https://www.ccsinfo.com/forum/viewtopic.php?t=54354&highlight=sht75
That why i ask you if they share same code

wangine, 2017-07-23:
I can not find a explication. I inspected the interface between sensors and RPi . Is OK. Still i notice something, with other SHT2x sensor daemon crash on fixed interval. 3h and 48min . 3 times daemon crash on same timing interval . Installing now last version .
[image]

kizniche, 2017-07-23:
I just pushed a commit to the PWM_support branch that adds more exception logging to the daemon. I would ```git pull``` this branch and start using this code, with hope we can catch the issue you've been experiencing.

>  ```WriteByte(0xFE); // in case if comunication is lost```

What exactly does this do? If the communication is lost, how can you write a Byte to the sensor?

kizniche, 2017-07-23:
That's very strange that it crashes after the same period of time.

wangine, 2017-07-23:
yes, i just updated to last PWM version , and strange , mycodo stop on same time period , realy annoing. More annoing when i try to look on logs , they are empty , no error code. (i just delete the logs when daemon working , to see clear the error)
[image]

**> What exactly does this do? If the communication is lost, how can you write a Byte to the sensor?**
That code will initialise the sensor to start normal reading. Is like any type of asincronous comunicattion , if the device dont get the init code or adress , will do nothing. So , if the comunication are lost the sensor will keep send data (exception when the sensor is damaged or hardware issue) but the system will dont sync that data. Is like a software resset of the sensor , after an error oncured . I dont know how SHT2x library write data on sensor but i will take a look in 5 min.

kizniche, 2017-07-23:
Try this. Add this after line 88 in ```~/Mycodo/mycodo/sensors/sht2x.py```

```python
        bus.write_byte(self.i2c_address, 0xFE)
```

So the code looks like:

```python
    def get_measurement(self):
        """ Gets the humidity and temperature """
        bus = smbus.SMBus(self.i2c_bus)
        bus.write_byte(self.i2c_address, 0xFE)
        # Send temperature measurement command
        #       0xF3(243)   NO HOLD master
        bus.write_byte(self.i2c_address, 0xF3)
        time.sleep(0.5)
...
```

Save, then restart the daemon

```sudo service mycodo restart```

kizniche, 2017-07-23:
Also, I pushed the commit with more logging less than an hour ago, so I don't think you have been running that new version.

wangine, 2017-07-23:
No , i try to commit now, Is strange some time i can git pull without sudo privileges and sometime not. 
`pi@raspberrypi:~/Mycodo $ git pull
error: cannot open .git/FETCH_HEAD: Permission denied
`
with sudo work.
Any clue? 
Also where need to write that code, always confuse me but i forget to ask you , 
in /var/vvv/mycodo/sensors....
or in /home/pi/Mycodo/sensors..........

kizniche, 2017-07-23:
/var/www/mycodo is justa  symlink to /home/pi/Mycodo

I usually use ~ and put commands like ```vim ~/Mycodo/mycodo/sensors/sht2x.py```

try this to pull the latest and hen restore proper permissions:

```bash
cd ~/Mycodo
sudo git pull
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
```

wangine, 2017-07-23:
I just look in daemon log, no any error about influxDB or SHT2x
I pull the last git
`2017-07-24 02:41:23,362 - mycodo - INFO - Mycodo daemon v5.0.49 starting
2017-07-24 02:41:23,534 - mycodo - INFO - Anonymous statistics enabled
2017-07-24 02:41:23,773 - mycodo.output - INFO - Relay 1 setup on pin 5 and turned OFF (OFF=LOW)
2017-07-24 02:41:23,775 - mycodo.output - INFO - Relay 2 setup on pin 6 and turned OFF (OFF=LOW)
2017-07-24 02:41:23,789 - mycodo.output - INFO - Output controller activated in 92.1 ms
2017-07-24 02:41:23,789 - mycodo - INFO - All activated timer controllers started
2017-07-24 02:41:24,024 - mycodo.sensor_1 - INFO - Activated in 184.6 ms
2017-07-24 02:41:24,026 - mycodo - INFO - All activated sensor controllers started
2017-07-24 02:41:24,210 - mycodo.pid_1 - INFO - Activated in 137.6 ms
2017-07-24 02:41:24,413 - mycodo.pid_2 - INFO - Activated in 149.7 ms
2017-07-24 02:41:24,416 - mycodo - INFO - All activated PID controllers started
2017-07-24 02:41:24,417 - mycodo - INFO - All activated LCD controllers started
2017-07-24 02:41:24,418 - mycodo - INFO - Mycodo daemon v5.0.49 started in 1.056 seconds
2017-07-24 02:41:24,428 - mycodo - INFO - 42.048 MB ram in use
2017-07-24 02:44:30,800 - mycodo - INFO - Received command to terminate daemon
2017-07-24 02:44:30,900 - mycodo - INFO - All LCD controllers stopped
2017-07-24 02:44:30,901 - mycodo - INFO - All Timer controllers stopped
2017-07-24 02:44:30,962 - mycodo.pid_1 - INFO - Deactivated in 61.7 ms
2017-07-24 02:44:30,980 - mycodo.sensor_1 - INFO - Deactivated in 80.0 ms
2017-07-24 02:44:30,982 - mycodo - INFO - All Sensor controllers stopped
2017-07-24 02:44:31,051 - mycodo.pid_2 - INFO - Deactivated in 150.7 ms
2017-07-24 02:44:31,052 - mycodo - INFO - All PID controllers stopped
2017-07-24 02:44:31,118 - mycodo.output - INFO - Output controller deactivated in 65.3 ms
2017-07-24 02:44:31,119 - mycodo - INFO - Mycodo terminated in 0.319 seconds`

kizniche, 2017-07-23:
I see the daemon start and then it was instructed to stop, and didn't start up again. Are you saying there is an issue?

wangine, 2017-07-23:
No, i stopped to update git , i'm saying , i dont see the error when daemon stop and why.

kizniche, 2017-07-24:
You may need to add a sleep period after the soft reset command, like this:

```python
    def get_measurement(self):
        """ Gets the humidity and temperature """
        bus = smbus.SMBus(self.i2c_bus)
        bus.write_byte(self.i2c_address, 0xFE)
        time.sleep(0.1)
        # Send temperature measurement command
        #       0xF3(243)   NO HOLD master
        bus.write_byte(self.i2c_address, 0xF3)
        time.sleep(0.5)
```

wangine, 2017-07-24:
Just to know , maybe help . Every time i install Mycodo on a fresh jessie system , i get first time an error, about fetch some archives (solved after) and about crontab entry. So i clear crontab entry , restart and install again. Now my BIG question . How can delete Mycodo without remain pieces behind system. It is possible or i need to put a fresh jessie again. That matter ?
Ok i will add the sleep

kizniche, 2017-07-24:
I recently (last month) did a fresh install of Mycodo and corrected a few issues with the install process. Before I release the PWM_support branch as v5.1.0, I'll do a fresh install and root out any issues.

You can do this which should remove the majority of what Mycodo does:

```bash
sudo apt-get remove apache2
sudo rm -rf ~/Mycodo
```

Then remove the crontab entry with ```sudo crontab -e```

There will still be some changes that Mycodo introduced. I've never created an uninstall script because of how easy writing a new Raspbian image is.

wangine, 2017-07-24:
Almost 200 eror block loke the below one after i insert the init SHT2x command . 
```
2017-07-24 03:02:11,839 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-07-24 03:02:24,939 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-24 03:02:24,939 - mycodo.pid_1 - WARNING - No data returned from influxdb
2017-07-24 03:02:25,083 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-07-24 03:02:25,084 - mycodo.pid_2 - WARNING - No data returned from influxdb
2017-07-24 03:02:26,849 - mycodo.sensors.sht2x - ERROR - SHT2xSensor raised an exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 119, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 92, in get_measurement
    bus.write_byte(self.i2c_address, 0xF3)
  File "/usr/local/lib/python2.7/dist-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/usr/local/lib/python2.7/dist-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
2017-07-24 03:02:41,917 - mycodo.sensors.sht2x - ERROR - SHT2xSensor raised an exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 119, in read
    try:
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 92, in get_measurement
    #       0xF3(243)   NO HOLD master
  File "/usr/local/lib/python2.7/dist-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/usr/local/lib/python2.7/dist-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
```
but after sleep line seems to be ok . 
```
2017-07-24 03:03:17,179 - mycodo - INFO - Mycodo daemon v5.0.49 starting
2017-07-24 03:03:17,353 - mycodo - INFO - Anonymous statistics enabled
2017-07-24 03:03:17,582 - mycodo.output - INFO - Relay 1 setup on pin 5 and turned OFF (OFF=LOW)
2017-07-24 03:03:17,583 - mycodo.output - INFO - Relay 2 setup on pin 6 and turned OFF (OFF=LOW)
2017-07-24 03:03:17,585 - mycodo.output - INFO - Output controller activated in 74.3 ms
2017-07-24 03:03:17,585 - mycodo - INFO - All activated timer controllers started
2017-07-24 03:03:17,784 - mycodo.sensor_1 - INFO - Activated in 157.0 ms
2017-07-24 03:03:17,784 - mycodo - INFO - All activated sensor controllers started
2017-07-24 03:03:18,025 - mycodo.pid_1 - INFO - Activated in 200.4 ms
2017-07-24 03:03:18,435 - mycodo.pid_2 - INFO - Activated in 301.8 ms
2017-07-24 03:03:18,439 - mycodo - INFO - All activated PID controllers started
2017-07-24 03:03:18,440 - mycodo - INFO - All activated LCD controllers started
2017-07-24 03:03:18,441 - mycodo - INFO - Mycodo daemon v5.0.49 started in 1.262 seconds
2017-07-24 03:03:18,456 - mycodo - INFO - 41.964 MB ram in use
```
Now i just wait to see how much will take to daemon stop.

kizniche, 2017-07-24:
It was probably writing the commands too quickly, without the sleep line.

Let's hope it works 🤞

wangine, 2017-07-24:
Yea , i never check with my scope on the end of the wire connector of my sensor. Anyway tomorow i will made a new wire connectors from my sensors and i will check that on the both interfaces , my old dsPIC33F interface and on RPi . I'm realy curious how the edges look because i still suspect my RPi have an unknow hardware isue, because the other RPi still working continuosly from last year without any crash or an update.

wangine, 2017-07-24:
Yes , my RPi board have an hardware issue, some pins are below 2.7v on "1" state, Thats mean is very close to neutral zone of "1" and "0" . I will fix with a repeater , stiil i dont figure out what couse that issue . Now i unpacking the new RPi. Also the code 
```         #bus.write_byte(self.i2c_address, 0xFE)
        #time.sleep(0.1) 
```
is in rong place , in reading function. I just take a look over the SHT2x.py .  I think here should be the init code. What do that function ? 
```
    def next(self):
        """ Get next measurement reading """
        if self.read():  # raised an error
            raise StopIteration  # required
        return dict(dewpoint=float('{0:.2f}'.format(self._dew_point)),
                    humidity=float('{0:.2f}'.format(self._humidity)),
                    temperature=float('{0:.2f}'.format(self._temperature)))
```
Is correct if i doing that ? I want to resset sensor only if raise an error on reading.
```
    def next(self):
        """ Get next measurement reading """
        if self.read():  # raised an error
            raise StopIteration  # required
        return  bus.write_byte(self.i2c_address, 0xFE) # reinit sensor
                    time.sleep(0.1)
                    dict(dewpoint=float('{0:.2f}'.format(self._dew_point)),
                    humidity=float('{0:.2f}'.format(self._humidity)),
                    temperature=float('{0:.2f}'.format(self._temperature)))
```

kizniche, 2017-07-24:
I just made an edit to the sensor module that *should* work. It will attempt a read, and if that fails, it will send the reset command and attempt a read again. I believe this is what you're trying to accomplish.

wangine, 2017-07-24:
Yes yes , that command was magic for me on PIC device because , i was to lazy to put a buffer between an 5v microcontroller and SHT2xx 3.3v , and some ms wasted on resset command dont hurt if is an error already. Also i found the error at install Mycodo. If you try now with a fresh jessie image to install Mycodo that will doesn't work , because some archive will fail to download. If that happen i think some packages will get conrupted and everything will go wrong. Today was first time after some hard days trying an install without error. 
```
One way is to add following lines to /etc/hosts file
Open /etc/hosts file with sudo rights $sudo nano /etc/hosts
Paste the following lines at the end of /etc/hosts file
93.93.128.193   mirrordirector.raspbian.org

93.93.128.191   archive.raspbian.org

sudo apt-get update -y && sudo apt-get upgrade -y && sudo apt-get autoremove -y
```
That will help alot. Anyone who try to install now will get an error , and after some try an restart will be ok , but the Mycodo behavior will be incomprehensible . Shoul say that in instructions page , because the actual sources on jessie image have a bug.

kizniche, 2017-07-24:
Just pushed another update for the sensor module, that fixes the logging references.

I believe the archive download issue may be specific to your network. I don't have an issue resolving either domains. I also haven't heard anyone else having this issue. I'll be attempting a fresh install in a few days, before I push the 5.1.0 release, so I'll see then if I have this problem.

wangine, 2017-07-25:
No no , because i was thinking on that before , so i try to install when i was on my service and there i have professional connection , other provider 1GB up/down. Same issue , also i found on internet the solution.
And is an raspian update issue , not archieve download , he come as ZIP so any byte lost will be an fail extract. Ok. i will give a try on my new RPi. and i hope tomorow morning daemon will be alive . Thanks

wangine, 2017-07-25:
The RPi is new one , sht25 is the same , a new one also the interface between relays and RPi . Jessie was fresh , Mycodo without any error. Still after 2h of work test , daemon still running but i notice some error in log.
```
2017-07-25 20:53:06,175 - mycodo - INFO - Mycodo daemon v5.0.49 starting
2017-07-25 20:53:06,543 - mycodo - INFO - Anonymous statistics enabled
2017-07-25 20:53:06,856 - mycodo.output - INFO - Output controller activated in 88.6 ms
2017-07-25 20:53:06,856 - mycodo - INFO - All activated timer controllers started
2017-07-25 20:53:07,062 - mycodo.sensor_1 - INFO - Activated in 161.5 ms
2017-07-25 20:53:07,065 - mycodo - INFO - All activated sensor controllers started
2017-07-25 20:53:07,066 - mycodo - INFO - All activated PID controllers started
2017-07-25 20:53:07,077 - mycodo - INFO - All activated LCD controllers started
2017-07-25 20:53:07,078 - mycodo - INFO - Mycodo daemon v5.0.49 started in 0.903 seconds
2017-07-25 20:53:07,090 - mycodo - INFO - 41.952 MB ram in use
2017-07-25 20:54:55,993 - mycodo.sensor_2 - INFO - Activated in 150.3 ms
2017-07-25 20:56:16,118 - mycodo.output - INFO - Relay 1 setup on pin 6 and turned OFF (OFF=LOW)
2017-07-25 20:57:06,573 - mycodo.output - INFO - Relay 2 setup on pin 5 and turned OFF (OFF=LOW)
2017-07-25 20:57:44,993 - mycodo.output - INFO - Relay 1 setup on pin 6 and turned OFF (OFF=LOW)
2017-07-25 20:57:54,475 - mycodo.output - INFO - Relay 2 setup on pin 5 and turned OFF (OFF=LOW)
2017-07-25 21:01:38,275 - mycodo.pid_1 - INFO - Activated in 191.9 ms
2017-07-25 21:03:32,619 - mycodo.pid_2 - INFO - Activated in 236.8 ms
2017-07-25 22:22:06,347 - mycodo.pid_1 - INFO - Hold
2017-07-25 22:23:05,727 - mycodo.pid_1 - INFO - Resume
2017-07-25 22:54:08,833 - mycodo.sensors.sht2x - ERROR - Exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 100, in get_measurement
    data0 = self.sht2x.read_byte(self.i2c_address)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 110, in read_byte
    raise IOError(ffi.errno)
IOError: 121
```
Daemon didn't stop ,yet.

wangine, 2017-07-26:
Finally hes dead after some hours of work , daemon stopped and i still don't get it , why daemoon stop . I think its some nasty in the new python library. I dont know 
```
2017-07-26 03:51:38,842 - mycodo - ERROR - Could not turn relay on: (sqlite3.OperationalError) unable to open database file
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 641, in relay_on
  File "/var/www/mycodo/mycodo/controller_relay.py", line 414, in relay_on_off
  File "/var/www/mycodo/mycodo/controller_relay.py", line 498, in check_conditionals
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2613, in all
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2761, in __iter__
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2774, in _execute_and_instances
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2765, in _connection_from_session
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 893, in connection
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 898, in _connection_for_bind
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 334, in _connection_for_bind
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 2039, in contextual_connect
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 2078, in _wrap_pool_connect
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1405, in _handle_dbapi_exception_noconnection
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 2074, in _wrap_pool_connect
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 376, in connect
    return _ConnectionFairy._checkout(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 713, in _checkout
    fairy = _ConnectionRecord.checkout(pool)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 480, in checkout
    rec = pool._do_get()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 1151, in _do_get
    return self._create_connection()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 323, in _create_connection
    return _ConnectionRecord(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 449, in __init__
    self.connection = self.__connect()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/pool.py", line 607, in __connect
    connection = self.__pool._invoke_creator(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/strategies.py", line 97, in connect
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 385, in connect
    return self.dbapi.connect(*cargs, **cparams)
OperationalError: (sqlite3.OperationalError) unable to open database file
2017-07-26 03:52:02,854 - mycodo - ERROR - Could not turn relay off: (sqlite3.OperationalError) unable to open database file
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 615, in relay_off
  File "/var/www/mycodo/mycodo/controller_relay.py", line 414, in relay_on_off
    self.check_conditionals(relay_id, duration)
  File "/var/www/mycodo/mycodo/controller_relay.py", line 498, in check_conditionals
    for each_conditional in conditionals.all():
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2613, in all
    return list(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2761, in __iter__
    return self._execute_and_instances(context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2774, in _execute_and_instances
    close_with_result=True)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2765, in _connection_from_session
    **kw)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 893, in connection
    execution_options=execution_options)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 898, in _connection_for_bind
    engine, execution_options)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 334, in _connection_for_bind
    conn = bind.contextual_connect()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 2039, in contextual_connect
    self._wrap_pool_connect(self.pool.connect, None),
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 2078, in _wrap_pool_connect
    e, dialect, self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1405, in _handle_dbapi_exception_noconnection
    exc_info
[... cut ...]
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 369, in _recv
    data = self._channel.recv()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/channel.py", line 50, in recv
    header = self.stream.read(self.FRAME_HEADER.size)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/stream.py", line 213, in read
    raise EOFError("connection closed by peer")
EOFError: connection closed by peer
2017-07-26 05:31:39,629 - mycodo - INFO - Mycodo daemon v5.0.49 starting
```

kizniche, 2017-07-26:
I've never experienced this on any of my systems. This occurs when nothing is enabled?

wangine, 2017-07-26:
No no , if i enabled only SHT2x or 7x or any type of sensor , is ok. but soon i enabled the relay and PID the errors will come soon as you can see. I dont think now is some issue about sensors , i believe is from system itself or some pakage. To be onest in last week that RPI's spend all my time and trying to make me crazy. Tomorow i will call my internet provider to change my DNS to other country , becaause soon i try to install the fresh downloades jessie image , i cant update if i dont edit host file with some sources. I cant send you the sensor from US ebay to receive it faster "dont let me to choose" i send you already from Germany , so will be some days delay until you will receive. Realy thanks for your support

kizniche, 2017-07-26:
What are your relay and PID settings? I'll try to induce the error on my system.

wangine, 2017-07-26:
Two relays GPIO 5 and 6 , and simple PID raise . Nothing special. 
That remain single error now , after i  update again the RPi . The error repeat on 2 - 3 h interval.
```2017-07-26 16:55:14,936 - mycodo.sensors.sht2x - ERROR - Exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 96, in get_measurement
    self.sht2x.write_byte(self.i2c_address, 0xF3)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
```

kizniche, 2017-07-26:
Are you sure the daemon isn't running, or are you just basing it on the color of the green/red indicator at the top-left of the web user interface? Use this to determine if the daemon is still running:

```ps aux | grep mycodo_daemon.py```

wangine, 2017-07-26:
I tolk today with my provider and switch my connection under Switzeland DNS. I will put a new jessie and i try to update without edit the sources. I will let you know.
Yes daemon stop, red indicator in the left , also controllers stop al.I restart RPi and everything became to normal until next error come. Now daemon still runing , strange , pass already 6 hours, normaly should be down until now. Just 1 error in 6 hours, but now still running.
```
2017-07-26 19:10:50,494 - mycodo.pid_2 - INFO - Hold
2017-07-26 19:11:08,188 - mycodo.pid_2 - INFO - Resume
2017-07-26 23:20:55,340 - mycodo.sensors.sht2x - ERROR - Exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 96, in get_measurement
    self.sht2x.write_byte(self.i2c_address, 0xF3)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
```
[image]

wangine, 2017-07-26:
??? I dont get it. On web page daemon is running , when i put your command mycodo_daemon.py return with red on SSH terminal. Is ok ? Or the web interface lie ?
```
pi@raspberrypi:~ $ ps aux | grep mycodo_daemon.py
root      1248  134  6.6 270640 58816 ?        Ssl  Jul26 1517:16 /var/www/mycodo/env/bin/python /var/www/mycodo/mycodo/mycodo_daemon.py
pi       30401  0.0  0.1   5932  1408 pts/0    S+   00:40   0:00 grep --color=auto mycodo_daemon.py
```

kizniche, 2017-07-26:
The web UI does not look at the daemon lock file and checn check that PID to see if the daemon is active, it executes the mycodo_client.py communication protocol to the daemon, and if it can communicate to the daemon, indicates it is running at the top-left of the UI. It essentially issues the following command:

```
~/mycodo/env/bin/python ~/mycodo/mycodo_client.py --checkdaemon
```

The errors you've been receiving, and similar:

```
File "/var/www/mycodo/mycodo/mycodo_client.py", line 102, in relay_off
    return self.rpyc_client.root.relay_off(relay_id, trigger_conditionals)
...
File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/stream.py", line 213, in read
    raise EOFError("connection closed by peer")
EOFError: connection closed by peer
```

are a result of the rpyc server no longer accepting connections. The rpyc server is how the PID thread (and others, such as the timer controllers, web UI, etc.) communicates to the relay thread, for instance.

Now that we know the daemon is indeed not crashing, that the communication thread of the daemon has stopped working, I can focus on trying to figure out why. I'll see if I can add more logging lines and find out exactly why it's crashing, or institute a way to check fi the communication thread has stopped and relaunch it.

kizniche, 2017-07-27:
I think I misunderstood what you were saying in your last post. You showed a screenshot of the web UI with green top-left, and a terminal screenshot of the daemon running.

I need you to run the ```ps aux | grep mycodo_daemon.py``` command after when find what you believe the daemon to have crashed. The red is only the matched string you put into ```grep``` and is normal.

wangine, 2017-07-27:
I understand now. About my network you has right. That why jessie fail to update. Behind my network are some very secure website, so my provider block by default some ports. I unblock everything behind other router, and now i can update without error. I still have doubt that was a reason for daemon and Mycodo fail on install also some randoms errors. Right now seems to be ok. but if daemon stop , i will change the Rpi with fresh Mycodo and Jessie, because now i get succes on first install and deserve a try.
Is crazy , no crash yet and hes running almost for 12h .

kizniche, 2017-07-27:
I just made some progress on understanding the issue with communication. I sill be pushing an update to the PWN_support branch in a minute that includes rpyc logging. Be warned though that it will produce a lot of lines n your log. This will only be temporary while we investigate this issue.

kizniche, 2017-07-27:
I'm also about to push an update that will upgrde rpyc to the latest version, but you'll need to include this command in your normal upgrade procedure for it to take effect, then restart the web UI and daemon.

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-pip-packages
```

kizniche, 2017-07-27:
I found how to get the rpyc errors to log but not all those info lines that are no doubt plentiful. I'll push an update in a minute. This should save your logs and having to hunt for errors though millions of info lines.

wangine, 2017-07-27:
ok

wangine, 2017-07-27:
Ok i update and i will delete the log , to see when i wake up if something change
```
2017-07-27 05:53:08,159 - mycodo - INFO - Mycodo daemon v5.0.49 starting
2017-07-27 05:53:08,337 - mycodo - INFO - Anonymous statistics enabled
2017-07-27 05:53:08,557 - mycodo.output - INFO - Relay 1 setup on pin 6 and turned OFF (OFF=LOW)
2017-07-27 05:53:08,558 - mycodo.output - INFO - Relay 2 setup on pin 5 and turned OFF (OFF=LOW)
2017-07-27 05:53:08,560 - mycodo.output - INFO - Output controller activated in 69.1 ms
2017-07-27 05:53:08,560 - mycodo - INFO - All activated timer controllers started
2017-07-27 05:53:08,755 - mycodo.sensor_1 - INFO - Activated in 152.7 ms
2017-07-27 05:53:08,997 - mycodo.sensor_2 - INFO - Activated in 170.5 ms
2017-07-27 05:53:09,004 - mycodo - INFO - All activated sensor controllers started
2017-07-27 05:53:09,222 - mycodo.pid_1 - INFO - Activated in 167.9 ms
2017-07-27 05:53:09,572 - mycodo.pid_2 - INFO - Activated in 289.3 ms
2017-07-27 05:53:09,576 - mycodo - INFO - All activated PID controllers started
2017-07-27 05:53:09,579 - mycodo - INFO - All activated LCD controllers started
2017-07-27 05:53:09,582 - mycodo - INFO - Mycodo daemon v5.0.49 started in 1.423 seconds
2017-07-27 05:53:09,620 - mycodo - INFO - 42.124 MB ram in use
2017-07-27 05:54:08,587 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
```

wangine, 2017-07-27:
I dont know where to look.
```
2017-07-27 23:47:38,523 - mycodo.sensors.sht2x - ERROR - Exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 96, in get_measurement
    self.sht2x.write_byte(self.i2c_address, 0xF3)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
2017-07-27 23:50:02,217 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 00:20:01,804 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 00:23:42,691 - mycodo.sensors.sht2x - ERROR - Exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/sht2x.py", line 96, in get_measurement
    self.sht2x.write_byte(self.i2c_address, 0xF3)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121
2017-07-28 00:50:02,596 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
```
Entire log below

[mycodo.zip](https://github.com/kizniche/Mycodo/files/1181457/mycodo.zip)

kizniche, 2017-07-27:
So it doesn't look like the daemon or the communication server has stopped. Looks like everything is still functioning.

wangine, 2017-07-27:
yes and no . Was a power line failure so Mycodo restart itself. If you restart RPi after daemon stop , everything will become to normal until will stop again. 
 I found a way for you  how to simulate the error. Just try to add a SHT2x sensor and activate it, made a PID and use with that sensor (random data) , on beginning , is normal and the sensor will read 0.0 , . After that restart RPi and you will see the errors flowing, and in the end daemon will crash , is like a simlink between errors, i dont know. I just tryit now on fresh Mycodo without any sensor attached.
Now in my mind, the code should be ,,,,
If any type of sensor crash or miss , data will be = with 0, or null. (that i see already in code)
Daemon should try to read sensor on the choosed timming (ex. 15 sec)
Any controller ex. PID , should hold and wait untill data will be different than 0.0.
Thats it, i still dont understand how one single read fail can transform everything in a mess with exp errors. 
And is realy very difficult to help you some how , cecause i didn't understand how the entire blocks of code work .

wangine, 2017-07-28:
Also i want to ask you . How to made a translation ? If work if i put a edited file with my language in
``` 
home/pi/Mycodo/mycodo/mycodo-flask/translations/ro/LC_MESSAGES/messages.po (with my language)
```
and run 
```
sudo service mycodo stop
sudo /etc/init.d/apache2 stop
cd Mycodo
sudo /bin/bash ./mycodo/scripts/upgrade_commands.sh initialize
sudo /bin/bash ./mycodo/scripts/upgrade_commands.sh restart-web-ui
sudo /bin/bash ./mycodo/scripts/upgrade_commands.sh restart-daemon
```
or i need to compile ?
Also im keep waiting , for some reason the RPi without sensor atached  still working, don't want to crash.
I put to read on 1 second , maybe the errors will acumulate more faster.

kizniche, 2017-07-28:
I have a wiki page explaining how to make a new translation.

https://github.com/kizniche/Mycodo/wiki/Translations

If you do make a translation, please send it to me to include in Mycodo or make a pull request with the new file.

kizniche, 2017-07-28:
I'm pushing an update in a moment that should prevent 0 values from being returned when there is a sensor error.

wangine, 2017-07-28:
Ok , i updated . Now i wait , on two RPi , one with sensor attached and other without. I assume now , the PID or other controll dont stop , just hold until get a new data , right? 
On translation , i will translate in Romanian and my son will translate in German, so soon i get time i will send you the two translations.

wangine, 2017-07-28:
I just look in translations, and Spanish translation dont have a valid python strings on some fields , like 
```
relé
Método
```
Some translations dont end in same way as the original string (space)
Also the french have a none UTF-8 types. Can affect that in any way the server or daemon?

kizniche, 2017-07-28:
What do you mean "valid python strings?"

Only the web UI uses translations. Also, I've rooted out the issues with use of different character sets. There shouldn't be any characters that crash the system.

wangine, 2017-07-28:
I just try to find out any issue, Just open the Spanish .po file with Poedit and try to validate . And you will see what i want to mean with no UTF-8 type. If you look on "é" letter in "relé" word , that is not accepted in C, or python.

kizniche, 2017-07-28:
I don't think it needs to be. It works. I'm not a translation expert, but there's not an issue with it in Mycodo.

wangine, 2017-07-29:
I test one RPi with PID and SHT2x sensor activated (without atached sensor) , 2 days and i wait to crash or daemon stop. I delete 1 millions errors (normal because sensor missing) but the system didn't crash in two days . I was thinking finaly is ok and  i delete all unnecesary error logs , i shutdown and i atach the sensor. Soon i atach the sensor on my first look on page i get unresponsive page , something about Apache server refuse to connect. I pull out the power, i start again and i look in log file.
```
2017-07-28 19:48:20,186 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 20:18:20,744 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 20:48:20,471 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 21:18:20,251 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 21:48:20,068 - mycodo.rpyc - ERROR - TESTING: (30 min timer) rpyc communication thread: closed=False
2017-07-28 22:05:28,712 - mycodo.pid_1 - ERROR - Run Error: 
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_pid.py", line 229, in run
    self.manipulate_output()
  File "/var/www/mycodo/mycodo/controller_pid.py", line 413, in manipulate_output
    min_off=self.raise_min_off_duration)
  File "/var/www/mycodo/mycodo/mycodo_client.py", line 105, in relay_on
    return self.rpyc_client.root.relay_on(relay_id,
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/netref.py", line 148, in __getattribute__
    return syncreq(self, consts.HANDLE_GETATTR, name)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/netref.py", line 71, in syncreq
    return conn.sync_request(handler, oid, *args)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/rpyc/core/protocol.py", line 467, in sync_request
    raise socket.timeout
timeout
2017-07-29 00:21:07,772 - mycodo.sensor_1 - ERROR - Error: can't start new thread
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_sensor.py", line 404, in run
    self.add_measure_influxdb()
  File "/var/www/mycodo/mycodo/controller_sensor.py", line 452, in add_measure_influxdb
    write_db.start()
  File "/usr/lib/python2.7/threading.py", line 745, in start
    _start_new_thread(self.__bootstrap, ())
error: can't start new thread
2017-07-29 03:09:55,137 - mycodo.daemon - INFO - Mycodo daemon v5.0.49 starting
```
I'm so confused , and realy i dont understand . Why that happen . I think i will wait until you will receive the sensor , because i'm overloaded right now.

wangine, 2017-07-29:
And that is not all. I restart the RPi and when i try to log in on SSH , surprise .
That can't be possible. 
```
wangine@wangine-PC ~
$ ssh pi@192.168.10.114
pi@192.168.10.114's password:
Permission denied, please try again.
pi@192.168.10.114's password:
Permission denied, please try again.
pi@192.168.10.114's password:
Permission denied (publickey,password).
```
That dont have any link with Mycodo or with sensor. I think is the system itself. RPi is new , unboxed 3 days ago, new SD card. Any clue?

kizniche, 2017-07-29:
Those are some new errors: "timeout" and "can't start new thread"

I'll be out of town the next week for the Society for Industrial Microbiology and Biotechnology (SIMB) conference in Denver, CO, so I may not be able to make any headway on this until after I return.

kizniche, 2017-07-29:
Those errors appear to have only happened once, and didn't disrupt the normal operation of the daemon.

[13 more replies]

closed 2018-04-06 by kizniche