kizniche/Mycodo #277: Fresh installation ERROR: Install Aborted

stminm, 2017-07-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: Latest version: 5.0.49
 
#### Problem Description
I tried multiple times to do a fresh install of Mycodo on Raspberry Pi 3 but with the same result: ERROR: Install Aborted! 

  - what were you trying to do?
I followed the installation instruction step-by-step.
  - specific setup details that are involved:
Raspberry Pi 3 Model B Board. 16GB SD card.
Fresh installed Raspbian Jessie Lite. Version: July 2017. Release date: 2017-07-05. Kernel version: 4.9

### Errors

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

### Steps to Reproduce the issue:
step 1: Deploy Raspbian Jessie Lite image with Win32DiskImager on SD card then boot the Raspberry Pi.
step 2: Follow the installation instruction step-by-step (starting from Configure raspi-config...)

Attached are the commands I entered to start the installation with the output and setup log.

[mycodo_install_commands.txt](https://github.com/kizniche/Mycodo/files/1178299/mycodo_install_commands.txt)
[setup.log.txt](https://github.com/kizniche/Mycodo/files/1178297/setup.log.txt)

kizniche, 2017-07-27:
Looking at your setup log, it appears to have installed properly. Restart your Pi and see if you can connect to http://0.0.0.0/ replacing 0.0.0.0 with your Pi's IP address.

If it still doesn't work, can you look at the contents of these two logs?

```
/var/log/mycodo/mycodo.log
/var/log/apache2/error.log
```

kizniche, 2017-07-27:
You may also have to execute this to fix a permission issue that may be present (this was the last command the setup.sh script didn't execute because it failed with the web UI restart.

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions
```

kizniche, 2017-07-27:
Curious. Can you execute this and tell me what it returns:

```
wget --version
```

stminm, 2017-07-27:
I followed your instructions and seems like the installation issue was fixed.
Below is the order of my actions:
After reboot when connecting to http://mycodo_IP, after redirection to https and accepted security exception, I got "Internal Server Error"
Here are the logs wget version output:
[mycodo.log.txt](https://github.com/kizniche/Mycodo/files/1178438/mycodo.log.txt)
[error.log.txt](https://github.com/kizniche/Mycodo/files/1178439/error.log.txt)
[wget_version.txt](https://github.com/kizniche/Mycodo/files/1178453/wget_version.txt)
After "sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-permissions" I got access to web interface and I was able to create user and login. 
Then I created and saved a sensor (Rasspberry PI: CPU Temperature) but when I tried to activate the sensor I got "Error: Error: Daemon: Connection refused. Is the daemon running?"
After another reboot I was able to activate the sensors and create a graph.

Now l have another issue: the CPU is getting hot. The max temperature reached 81.67 C:
[image]

With "top" I see that the CPU is heavily loaded by "mycodo_daemon.py":
[image]

Let me know if I have to create a "New issue" for high CPU temperature.

closed 2017-07-27 by stminm

kizniche, 2017-07-27:
80C!! Is that too high? I commonly have python (mycodo) process running >100% CPU, but I don't see mine go above 60C. My room temperature is 25C and I just have it open on my desk, flat with the CPU facing upwards, no heatsink or fan. What are the conditions like for your Pi?

stminm, 2017-07-27:
Room temp ~ 24C. It's on my desk in a case with heatsinks like these: https://www.amazon.com/Raspberry-Model-Protective-Heatsinks-Clear/dp/B01CDVSBPO

kizniche, 2017-07-27:
I think if you have both a heatsink and the case closed, the temperature is building up on the CPU. Try opening the case up so airflow helps lower the temperature. The heat from the heatsink should be enough to cause convection currents to bring cool air in.

I've found I only need to attempt cooling when I'm using it in very hot environments (closed shed, unairconditioned, in direct sunlight), or an unconditioned room in a house in an electrical case. The first case is expected, but I found adding the Pi to any enclosure will permit heat buildup.

stminm, 2017-07-27:
You are right. With the open case, the temperature dropped to 75C.
I have plans to use my Pi in an electrical metal box. Based on what you are saying and what I've seen I may need a small cooler and/or a bigger heatsink :( ...

Thanks for the prompt support and useful advice!

kizniche, 2017-07-27:
I would recommend a small fan attached to the electrical box and if possible, positioned so the air being pulled in gets pulled over the Pi.

kizniche, 2017-07-27:
A 40 mm fan is about the smallest I would use to move air from an electrical box, though I did get a laugh when I saw this while searching for fan images.

[image]

kizniche, 2017-07-27:
Thanks for the issue report. Glad it was easily fixed. I believe it was a fluke and not an easily reproducible issue, so I'm not sure if I can do anything about it. I'll be going through a full install of Mycodo (which I do every few months) to see if I get any install errors that I can then diagnose.