kizniche/Mycodo #278: New install fails

WiehanAu, 2017-07-31:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.49
 
#### Problem Description
New install fails on vanilla Raspbian Lite on Raspberry Pi 3

Install Mycodo following instructions as listed.

- Flashed Raspbian Lite using Etcher
- Insert SD Card into RPi 3 and power up
- Connect RPi3 to Wifi Network

- run 'sudo raspi-config'
- configure exactly as stated:

'Change User Password
Localisation Options -> Change Locale (set and select en_US.UTF-8, if US)
Localisation Options -> Change Timezone
Interfacing Options -> SSH -> Enable
Interfacing Options -> Camera -> Enable (required if using a Pi camera)
Interfacing Options -> I2C -> Enable (required if using I2C sensors/devices)
Advanced Options -> Expand Filesystem (required)
Reboot (required)'

- Run install steps as indicated:

sudo apt-get install -y jq
cd ~
curl -s https://api.github.com/repos/kizniche/Mycodo/releases/latest | \
jq -r '.tarball_url' | wget -i - -O mycodo-latest.tar.gz
mkdir Mycodo
tar xzf mycodo-latest.tar.gz -C Mycodo --strip-components=1
rm -f mycodo-latest.tar.gz
cd Mycodo/install
sudo /bin/bash ./setup.sh

 
### Errors
I've attached the full setup.log, I do not see any specific errors.

The setup fails with:

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service.

#### Creating Mycodo database if it doesn't exist

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

### Steps to Reproduce the issue:
Brand new install following steps outlined on site. 
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

[setup.zip](https://github.com/kizniche/Mycodo/files/1187921/setup.zip)

kizniche, 2017-07-31:
This appears to be a duplicate of issue #277

Go ahead and follow the recommendations in that issue, but leave this issue open and I'll investigate it after I return from traveling, next week.

WiehanAu, 2017-08-01:
My apologies, I should have checked the existing issues before posting.

I followed the same procedures as described in #277, which resolved my problems. Mycodo is now up and running and I'm able to create sensors. My experience was an exact duplicate of issue #277.

Please feel free to request any additional info should you investigate this further.

closed 2017-08-10 by kizniche