kizniche/Mycodo #279: Unable to read Senseair K-30 CO2 sensor data on Live Measurements page

Magnum-Pl, 2017-08-01:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.0.49
 
#### Problem Description

  After adding and activating a Senseair K-30 CO2 sensor, I was unable to read any data on the Live Measurements page. 

I followed the steps in the Senseair Pi.pdf to configure UART and successfully tested the sensor using the K-30 test, but was still unable to display any sensor data in Mycodo.

 
### Errors
LOG:
 2017-07-30 14:15:23,309 - mycodo - INFO - Mycodo daemon v5.0.49 starting
2017-07-30 14:15:23,731 - mycodo - INFO - Anonymous statistics enabled
2017-07-30 14:15:24,179 - mycodo.relay - INFO - Relay controller activated in 122.3 ms
2017-07-30 14:15:24,180 - mycodo - INFO - All activated timer controllers started
2017-07-30 14:15:24,517 - mycodo.sensor_1 - INFO - Activated in 261.7 ms
2017-07-30 14:15:24,520 - mycodo - INFO - All activated sensor controllers started
2017-07-30 14:15:24,522 - mycodo - INFO - All activated PID controllers started
2017-07-30 14:15:24,523 - mycodo - INFO - All activated LCD controllers started
2017-07-30 14:15:24,524 - mycodo - INFO - Mycodo daemon v5.0.49 started in 1.215 seconds
2017-07-30 14:15:24,527 - mycodo.sensors.k30 - ERROR - K30Sensor raised an exception when taking a reading: [Errno 2] could not open port /dev/ttyS0: [Errno 2] No such file or directory: '/dev/ttyS0'
2017-07-30 14:15:24,547 - mycodo - INFO - 41.736 MB ram in use
2017-07-30 14:15:43,979 - mycodo.sensors.k30 - ERROR - K30Sensor raised an exception when taking a reading: [Errno 2] could not open port /dev/ttyS0: [Errno 2] No such file or directory: '/dev/ttyS0'
2017-07-30 14:15:59,070 - mycodo.sensors.k30 - ERROR - K30Sensor raised an exception when taking a reading: [Errno 2] could not open port /dev/ttyS0: [Errno 2] No such file or directory: '/dev/ttyS0'
2017-07-30 14:15:59,073 - mycodo.sensor_1 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.

### Steps to Reproduce the issue:
I tried using a fresh install of Jessie and Mycodo on a new Micro SD card and had the same issue. 

### Additional Notes
I was able to get the data to display on the Live Measurements page with a workaround. The log was displaying an error (Errno2) stating it could not open port /dev/ttyS0. I noticed a few lines (21-23) in the k30.py file defining which port to use. Either /dev/ttyS0, which didn't exist, else /dev/ttyAMA0. I deleted the line for /dev/ttyS0 and was able to view the K-30 sensor data on the Live Measurement page after a reboot. 

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1189338/mycodo.log.zip)

[k30.py.zip](https://github.com/kizniche/Mycodo/files/1189339/k30.py.zip)

kizniche, 2017-08-03:
As a resolution to this I'll enable selection of the device as a sensor option, as a text field to enter the specific device. This will also allow custom UART devices to be used.

Magnum-Pl, 2017-08-11:
I'm unable to view any K30 sensor data after the update. 

Using the device location /dev/ttyS0 or /dev/ttyAMA0, no data is displayed on the Live Measurements page.

I'm still able to use the K30 test script to view the sensor data. I tried another fresh install of Jessie and Mycodo 5.1.2 and got the same results. If I restore 5.0.49 I can read the sensor data again.

The error I'm getting from the log is:

2017-08-10 12:24:24,153 - mycodo.sensors.k30./dev/ttyS0 - ERROR - Could not open "/dev/ttyS0". Check the device location is correct.

2017-08-10 12:27:55,436 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.

OR

2017-08-10 12:24:05,780 - mycodo.sensors.k30./dev/ttyAMA0 - ERROR - Could not open "/dev/ttyAMA0". Check the device location is correct.

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1216803/mycodo.log.zip)

kizniche, 2017-08-11:
I just committed a tested fix and released v5.1.4. Let me know if it's working after an update.

Magnum-Pl, 2017-08-11:
Its working great after the v5.1.4 update. 

Thanks for the quick fix.

closed 2017-08-11 by Magnum-Pl