kizniche/Mycodo #281: Sensor Addition: MH-Z16 CO2 Sensor

Magnum-Pl, 2017-08-07:
Hi Kyle, 

Can you add support for the MH-Z16 CO2 sensor from Sandbox Electronics?

I've attached a Python script from Sandbox. 

Also included are links to their GitHub repo and the sensor page on their site.

I tested the script and was able to view the sensor data.

https://github.com/SandboxElectronics/NDIR

http://sandboxelectronics.com/?product=mh-z16-ndir-co2-sensor-with-i2cuart-5v3-3v-interface-for-arduinoraspeberry-pi

[NDIR_RasPi_Python.zip](https://github.com/kizniche/Mycodo/files/1204067/NDIR_RasPi_Python.zip)

kizniche, 2017-08-09:
I included support for this sensor in the latest release (v5.1.0). Please upgrade and let me know if it works. Thanks.

Magnum-Pl, 2017-08-11:
I can't get the MH-Z16 to display any sensor data.

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1218801/mycodo.log.zip)

kizniche, 2017-08-11:
I just made a commit to fix this issue and released v5.1.5. Let me know if there's still an issue after an update.
Thanks.

kizniche, 2017-08-11:
wow, I had added support for the MH-Z19 sensor, not the MH-Z16. Let me duplicate and add support for that one

kizniche, 2017-08-11:
Okay, that was weird. I had actually added the MH-Z19 before you created this issue, and I just thought it was fortuitous. We both overlooked that version number.

Well, there now should be I2C and UART support for the MH-Z16 CO2 sensor in the latest release, v5.1.6. If you could test both interfaces, it would be appreciated. I'm about to leave my home, so I'm not sure how much I can help if there's an issue, but I'll be around to check this issue. Thanks.

Magnum-Pl, 2017-08-12:
I didn't catch the difference until I tried to add the sensor on the inputs page! I thought it might work because they're part of the same family. 

The MH-Z16 works great with UART, but I can't get any data to display with I2C.

I'm able to use the NDIR.py and example.py scripts to test the sensor and get data that way through I2C, but I can't get the sensor to display anything in Mycodo. 

The MH-Z16 datasheet says to use the address 0x9A, but that didn't work. I used "i2cdetect -y 1" to determine that the MH-Z16 is using the I2C address 0x4D, but that didn't work either.

Im able to add the sensor on the Inputs page. When I try to activate it I get a message in an orange box that says:

**Error: Could not activate Sensor controller with ID 6: 'SensorController' object has no attribute 'i2c_address'**

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1219394/mycodo.log.zip)

kizniche, 2017-08-12:
That seems like an easy fix if that's the error. Let me check the code and get back to you with what to edit and retest the I2C.

kizniche, 2017-08-12:
It appears I forgot to add ```MH_Z16_I2C``` to the list of I2C devices at line 158 of ~/Mycodo/mycodo/config.py

Line 158: ```LIST_DEVICES_I2C = [```

If you add it, so it looks like the following, it should work.

```python
LIST_DEVICES_I2C = [
    'ADS1x15',
    'AM2315',
    'ATLAS_PH_I2C',
    'ATLAS_PT1000_I2C',
    'BH1750',
    'BME280',
    'BMP',
    'BMP180',
    'BMP280',
    'CHIRP',
    'HTU21D',
    'MH_Z16_I2C',
    'MCP342x',
    'SHT2x',
    'TMP006',
    'TSL2561',
    'TSL2591'
]
```

Make sure to save, and reload the daemon

```
sudo service mycodo restart
```

Magnum-Pl, 2017-08-12:
I made that edit and got another error while activating the sensor.

**Error: Could not activate Sensor controller with ID 6: 'MHZ16Sensor' object has no attribute 'interface'**

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1219403/mycodo.log.zip)

kizniche, 2017-08-12:
In ~/Mycodo/mycodo/sensors/mh_z16.py, move line 28 (```self.interface = interface```) to after line 19, so it looks like this:

```python
        super(MHZ16Sensor, self).__init__()
        self.interface = interface
        if self.interface == 'UART':
            self.logger = logging.getLogger(
                "mycodo.sensors.mhz16.{dev}".format(dev=device_loc.replace('/', '')))
        elif self.interface == 'I2C':
            self.logger = logging.getLogger(
                "mycodo.sensors.mhz16.{dev}".format(dev=i2c_address))

        self.k30_lock_file = None
        self._co2 = 0
```

Apologies, I made this module in a hurry before leaving the house and didn't really look at the code closer than pulling the UART and I2C code from the sources and dropping it in. We at least got the UART part working on the first try ;)

Magnum-Pl, 2017-08-12:
No worries. I appreciate all your effort and I'm glad to help get things sorted. 

The sensor data is showing up on the Live Measurements page and I'm no longer getting any errors on the Inputs page after that edit, but Im still getting this error in the log:

**2017-08-11 22:08:05,602 - mycodo.sensors.mhz16.77 - ERROR - MHZ16Sensor raised an exception when taking a reading: 121
2017-08-11 22:08:05,603 - mycodo.sensor_6 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.**

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1219428/mycodo.log.zip)

kizniche, 2017-08-12:
I'm not sure what that is, so we'll have to change the logging line from error to exception. In ~/Mycodo/mycodo/sensors/mh_z16.py change line 169 from ```error``` to ```exception```, like so:

```python
            self.logger.error(
                "{cls} raised an exception when taking a reading: "
                "{err}".format(cls=type(self).__name__, err=e))
```

This will print the entire exception traceback.

Magnum-Pl, 2017-08-12:
Ok, I made the edit and this is what I'm getting in the log:

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1219444/mycodo.log.zip)

kizniche, 2017-08-12:
Okay. Here's what I would do to fix this, I'm not sure why there are read issues, but I modified the module to be more like the example script. There are too many edits to list them all, so here's the whole ~/Mycodo/mycodo/sensors/mh_z16.py module to test:

```python
# coding=utf-8

from lockfile import LockFile
import logging
import serial
import smbus
import struct
import time
from .base_sensor import AbstractSensor

from sensorutils import is_device

class MHZ16Sensor(AbstractSensor):
    """ A sensor support class that monitors the MH-Z16's CO2 concentration """

    def __init__(self, interface, device_loc=None, baud_rate=None,
                 i2c_address=None, i2c_bus=None):
        super(MHZ16Sensor, self).__init__()
        self.interface = interface
        if self.interface == 'UART':
            self.logger = logging.getLogger(
                "mycodo.sensors.mhz16.{dev}".format(dev=device_loc.replace('/', '')))
        elif self.interface == 'I2C':
            self.logger = logging.getLogger(
                "mycodo.sensors.mhz16.{dev}".format(dev=i2c_address))

        self.k30_lock_file = None
        self._co2 = 0

        if self.interface == 'UART':
            # Check if device is valid
            self.serial_device = is_device(device_loc)
            if self.serial_device:
                try:
                    self.ser = serial.Serial(self.serial_device,
                                             baudrate=baud_rate,
                                             timeout=1)
                    self.k30_lock_file = "/var/lock/sen-mhz16-{}".format(device_loc.replace('/', ''))
                except serial.SerialException:
                    self.logger.exception('Opening serial')
            else:
                self.logger.error(
                    'Could not open "{dev}". '
                    'Check the device location is correct.'.format(
                        dev=device_loc))

        elif self.interface == 'I2C':
            self.cmd_measure = [0xFF, 0x01, 0x9C, 0x00, 0x00, 0x00, 0x00, 0x00, 0x63]
            self.IOCONTROL = 0X0E << 3
            self.FCR = 0X02 << 3
            self.LCR = 0X03 << 3
            self.DLL = 0x00 << 3
            self.DLH = 0X01 << 3
            self.THR = 0X00 << 3
            self.RHR = 0x00 << 3
            self.TXLVL = 0X08 << 3
            self.RXLVL = 0X09 << 3
            self.i2c_address = i2c_address
            self.i2c = smbus.SMBus(i2c_bus)
            self.begin()

    def __repr__(self):
        """  Representation of object """
        return "<{cls}(co2={co2})>".format(
            cls=type(self).__name__,
            co2="{0:.2f}".format(self._co2))

[... 133 lines cut ...]
        time.sleep(0.001)
        self.i2c.write_byte_data(self.i2c_address, reg_addr, val)

    def send(self, command):
        if self.read_register(self.TXLVL) >= len(command):
            self.i2c.write_i2c_block_data(self.i2c_address, self.THR, command)

    def receive(self):
        n = 9
        buf = []
        start = time.clock()

        while n > 0:
            rx_level = self.read_register(self.RXLVL)

            if rx_level > n:
                rx_level = n

            buf.extend(self.i2c.read_i2c_block_data(self.i2c_address, self.RHR, rx_level))
            n = n - rx_level

            if time.clock() - start > 0.2:
                break
        return buf
```

Magnum-Pl, 2017-08-12:
The only error I'm getting in the log now is:

**2017-08-11 22:55:27,876 - mycodo.sensor_6 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.**

Everything else is working fine. I'm still getting data on the Live Measurements page.

[mycodo.log.zip](https://github.com/kizniche/Mycodo/files/1219457/mycodo.log.zip)

kizniche, 2017-08-12:
Try changing after line 155 so it looks like this:

```python
            elif self.interface == 'I2C':
                for _ in range(2):
                    self._co2 = self.get_measurement()
                    if self._co2:
                        return  # success - no errors
                    time.sleep(1)
```

Magnum-Pl, 2017-08-12:
I'm still getting the same error in the log, but not as often and it took a few minutes for it to show up.

**2017-08-11 23:31:07,325 - mycodo.sensor_6 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.**

[mycodo.log 2.zip](https://github.com/kizniche/Mycodo/files/1219476/mycodo.log.2.zip)

kizniche, 2017-08-12:
Strange. I suspect it may be the time.sleep()s being too short in read_register() (line 201) and write_register() (line 205). I also think if the test script had try/except tests it would also error the same as this module.

That's all for me, I'm going to sleep. Feel free to experiment to see if you can reduce or get rid of the errors that occasionally occur.

Magnum-Pl, 2017-08-12:
Ok, I'll let you know if I have any luck getting rid of the error.

Thanks for the help.

closed 2017-08-12 by Magnum-Pl

kizniche, 2017-08-13:
I did a double-take just now because I've been scouring the code looking for what could be causing the intermittent errors, made a few commits, and was going to ask if you could test the code, then I looked back at your last comment and say you edited it to say you found the issue.

Could you please test my latest [mh_z16.py](https://github.com/kizniche/Mycodo/blob/d33a4de9cc02c8419d366b8896cd9694086e3a2c/mycodo/sensors/mh_z16.py) code to see if it runs without errors? I changed the timers to what you said works and removed the 3 tries to acquire a measurement. If this works well, I'll add it to the next release. Thanks.

Magnum-Pl, 2017-08-13:
I've been running the new code for 30 minutes with zero errors. 

I'm comparing it to the K30 sensor and the data is accurate.

I'll let you know if any errors pop up.

Should I reopen this until you get it into the next release, or leave it closed?

Thanks again

kizniche, 2017-08-13:
Thanks for testing. Let's leave it closed unless you come across any issue. I'd let it run a few days before confirming everything is working, for good measure.

kizniche, 2017-08-16:
How's the sensor module been working?

Magnum-Pl, 2017-08-16:
No issues so far. 

I'm going to switch over to UART when I get home and let that run for a few days.

Magnum-Pl, 2017-08-21:
The MH-Z16 has been running great through UART. 

No errors at all.

kizniche, 2017-08-21:
Great! Thanks for testing. If there are any other sensors you have, now or in the future, let me know and I'll try to support them.

Magnum-Pl, 2017-08-21:
Sounds great, I appreciate it. 

I just order a 433mhz transceiver and a PH module. 

I'll try to find some code and I'll let you know when I get them.

samannazz, 2022-06-15:
Hello Magnum-PI, i just get stuck with the MH-Z16 CO2 sensor while measuring via I2C pins.  

I run the [sandBox electronics code for python](https://github.com/SandboxElectronics/NDIR/tree/master/NDIR_RasPi_Python) but it caused an error **sensor not found**, so I used **i2cdetect -y 1** to determine if mh-z16 use I2C but it's not connected. I have another mh-z16 same response with this.  
Now I want to use uart, but I can't find your code in Mycode/mycode, sensor folder is not available please guide me where can i find it.  
As this matter is urgent, I would appreciate a reply as soon as possible. thanks

kizniche, 2022-06-15:
The [MH-Z16 Mycodo input](https://kizniche.github.io/Mycodo/Supported-Inputs/#winsen-mh-z16) supports both I2C and UART. Type "MH-Z16" in the Input search box on the Mycodo Input page and you should see both I2C and UART inputs.

samannazz, 2022-06-16:
okay, so i got your code but its very confusing if i download the repository https://github.com/kizniche/Mycodo.git and run the mh_z16.py code under Mycode/mycode/inputs/mh_z16.py it says **ModuleNotFoundError: No module named 'mycodo'** so i cut the mh_z16.py code and paste it into same directory where abstract_base_controller.py is and remove mycode directory refrence but this abstract_base_controller.py have other file where it import classes so till know i am unable to run your code ... is there any simple methode to run this code?