kizniche/Mycodo #283: 500 internal server error

dabbingsolutions, 2017-08-09:
## Mycodo Issue Report:
  - Specific Mycodo Version:5.1.0
 
#### Problem Description
Please list: after install can not open mycodo

  - what were you trying to do install mycodo
  - specific setup details that are involved 
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs 
#### Creating Mycodo database if it doesn't exist
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 16:30:46--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 16:30:53 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 16:31:54--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 16:31:55 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 16:32:55--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 16:32:56 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 16:33:56--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 16:33:57 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 16:34:57--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 16:34:58 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.0.24/, or whatever your Pi's

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1 new sd card install os and change config file 
 2. step 2...install mycodo
 3. 500 internal server error trying to open mycodo
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-08-09:
Only if you haven't rebooted yet, can you issue the following command and paste the output?

```
sudo /etc/init.d/apache2 restart
```

kizniche, 2017-08-09:
Also this command:

```
wget --no-check-certificate -p https://localhost/ -O /dev/null

```

kizniche, 2017-08-09:
Also, please attach the entire setup.log found at ~/Mycodo/install/setup.log

dabbingsolutions, 2017-08-09:
sudo /etc/init.d/apache2 restart
[....] Restarting apache2 (via systemctl): apache2.serviceWarning: Unit file of apache2.service changed on disk, 'systemctl daemon-reload' recommended.
. ok 
wget --no-check-certificate -p https://localhost/ -O /dev/null
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 17:03:20--  https://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-09 17:03:25 ERROR 500: Internal Server Error.

~/Mycodo/install/setup.log
bash: /home/pi/Mycodo/install/setup.log: Permission denied

kizniche, 2017-08-09:
You shouldn't be trying to execute the setup.log. I just need to see what part of the install failed that isn't allowing the web server to start.

kizniche, 2017-08-09:
Alternatively, do this and let me know the error:

```
sudo /etc/init.d/apache2 stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/start_flask_ui.py -d
```

Then in another terminal issue this command:

```
wget --no-check-certificate -p https://localhost/ -O /dev/null
```

Back in the first terminal look for the error

dabbingsolutions, 2017-08-09:
wget --no-check-certificate -p https://localhost/ -O /dev/null
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 17:10:52--  https://localhost/

Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:443... failed: Connection refused.
Connecting to localhost (localhost)|127.0.0.1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/login [following]
--2017-08-09 17:10:53--  https://localhost/login
Connecting to localhost (localhost)|127.0.0.1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/create_admin [following]
--2017-08-09 17:10:53--  https://localhost/create_admin
Connecting to localhost (localhost)|127.0.0.1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 200 OK
Length: 4336 (4.2K) [text/html]
Saving to: ‘/dev/null’

/dev/null           100%[=====================>]   4.23K  --.-KB/s   in 0s     

2017-08-09 17:10:53 (10.9 MB/s) - ‘/dev/null’ saved [4336/4336]

Loading robots.txt; please ignore errors.
--2017-08-09 17:10:53--  https://localhost/robots.txt
Connecting to localhost (localhost)|127.0.0.1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/create_admin [following]
--2017-08-09 17:10:54--  https://localhost/create_admin
Connecting to localhost (localhost)|127.0.0.1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 200 OK
Length: 4336 (4.2K) [text/html]
Saving to: ‘/dev/null’

/dev/null           100%[=====================>]   4.23K  --.-KB/s   in 0s     

2017-08-09 17:10:54 (11.7 MB/s) - ‘/dev/null’ saved [4336/4336]

FINISHED --2017-08-09 17:10:54--
Total wall clock time: 1.4s
Downloaded: 2 files, 8.5K in 0.001s (11.3 MB/s)
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/start_flask_ui.py -d
2017-08-09 17:10:18,596  * Running on https://0.0.0.0:443/ (Press CTRL+C to quit)
2017-08-09 17:10:18,603  * Restarting with stat
2017-08-09 17:10:23,573  * Debugger is active!
2017-08-09 17:10:23,574  * Debugger PIN: 363-949-420
2017-08-09 17:10:53,241 127.0.0.1 - - [09/Aug/2017 17:10:53] "GET / HTTP/1.1" 302 -
2017-08-09 17:10:53,459 127.0.0.1 - - [09/Aug/2017 17:10:53] "GET /login HTTP/1.1" 302 -
2017-08-09 17:10:53,831 127.0.0.1 - - [09/Aug/2017 17:10:53] "GET /create_admin HTTP/1.1" 200 -
2017-08-09 17:10:54,051 127.0.0.1 - - [09/Aug/2017 17:10:54] "GET /robots.txt HTTP/1.1" 302 -
2017-08-09 17:10:54,323 127.0.0.1 - - [09/Aug/2017 17:10:54] "GET /create_admin HTTP/1.1" 200 -

kizniche, 2017-08-09:
Can you now do this:

Ctrl+c the terminal with the running start_flask_ui.py

then execute:

```
sudo /etc/init.d/apache2 restart
wget --no-check-certificate -p https://localhost/ -O /dev/null
```

dabbingsolutions, 2017-08-09:
sudo /etc/init.d/apache2 restart
[....] Restarting apache2 (via systemctl): apache2.serviceWarning: Unit file of apache2.service changed on disk, 'systemctl daemon-reload' recommended.
. ok 
pi@raspberrypi:~ $ wget --no-check-certificate -p https://localhost/ -O /dev/null
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-09 17:17:07--  https://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:443... connected.
WARNING: The certificate of ‘localhost’ is not trusted.
WARNING: The certificate of ‘localhost’ hasn't got a known issuer.
The certificate's owner does not match hostname ‘localhost’
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/login [following]
--2017-08-09 17:17:12--  https://localhost/login
Reusing existing connection to [localhost]:443.
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/create_admin [following]
--2017-08-09 17:17:12--  https://localhost/create_admin
Reusing existing connection to [localhost]:443.
HTTP request sent, awaiting response... 200 OK
Length: 4336 (4.2K) [text/html]
Saving to: ‘/dev/null’

/dev/null                100%[==================================>]   4.23K  --.-KB/s   in 0s     

2017-08-09 17:17:13 (10.0 MB/s) - ‘/dev/null’ saved [4336/4336]

Loading robots.txt; please ignore errors.
--2017-08-09 17:17:13--  https://localhost/robots.txt
Reusing existing connection to [localhost]:443.
HTTP request sent, awaiting response... 302 FOUND
Location: https://localhost/create_admin [following]
--2017-08-09 17:17:13--  https://localhost/create_admin
Reusing existing connection to [localhost]:443.
HTTP request sent, awaiting response... 200 OK
Length: 4336 (4.2K) [text/html]
Saving to: ‘/dev/null’

/dev/null                100%[==================================>]   4.23K  --.-KB/s   in 0s     

2017-08-09 17:17:13 (11.5 MB/s) - ‘/dev/null’ saved [4336/4336]

FINISHED --2017-08-09 17:17:13--
Total wall clock time: 6.2s
Downloaded: 2 files, 8.5K in 0.001s (10.7 MB/s)

kizniche, 2017-08-09:
Well, I'm baffled. Everything is working now, but I still have no idea why the setup script fails.

kizniche, 2017-08-09:
Can you paste the last 10 0lines or so of the log at: /var/log/apache2/error.log

dabbingsolutions, 2017-08-09:
/var/log/apache2/error.logbash: /var/log/apache2/error.log: Permission denied

kizniche, 2017-08-09:
```
sudo tail -n 100 /var/log/apache2/error.log
```

dabbingsolutions, 2017-08-09:
[Wed Aug 09 17:22:05.952652 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/session.py", line 214, in _assert_active
[Wed Aug 09 17:22:05.952778 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     % self._rollback_exception
[Wed Aug 09 17:22:05.952870 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822] InvalidRequestError: This Session's transaction has been rolled back due to a previous exception during flush. To begin a new transaction with this Session, first issue Session.rollback(). Original exception was: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'INSERT INTO users (name, password_hash, email, role, theme) VALUES (?, ?, ?, ?, ?)'] [parameters: (u'admin', '$2b$12$GPSxUb.elYtoARE53XsZjOUy2wtula32V7C.zbmQakdlFZPL6yPSC', u'roger@dabbingsolutions.com', 1, 'slate')]

also i just created an admin acc and got the 500 internal server error

kizniche, 2017-08-09:
Is that all there is that was output from the log? That seems too small.

kizniche, 2017-08-09:
try executing this:

```
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize
```

dabbingsolutions, 2017-08-09:
sudo tail -n 100 /var/log/apache2/error.log
[Wed Aug 09 17:17:12.611635 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,611] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET name=? WHERE roles.id = ?'] [parameters: ('Guest', 4)]
[Wed Aug 09 17:17:12.626706 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,626] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_stats=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.641680 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,641] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_camera=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.657901 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,657] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_logs=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.672913 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,672] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_settings=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.687933 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,687] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET edit_controllers=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.703004 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,702] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET edit_users=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:17:12.718148 2017] [wsgi:error] [pid 2378:tid 1962927152] [2017-08-09 17:17:12,717] ERROR in __init__: Failed to save <mycodo.databases.models.role.Role object at 0x6bcf0890> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET edit_settings=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Wed Aug 09 17:22:05.908518 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822] [2017-08-09 17:22:05,880] ERROR in app: Exception on /create_admin [POST]
[Wed Aug 09 17:22:05.908631 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822] Traceback (most recent call last):
[Wed Aug 09 17:22:05.908653 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Wed Aug 09 17:22:05.908673 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     response = self.full_dispatch_request()
[Wed Aug 09 17:22:05.908692 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Wed Aug 09 17:22:05.908712 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     rv = self.handle_user_exception(e)
[Wed Aug 09 17:22:05.908731 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Wed Aug 09 17:22:05.908751 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     reraise(exc_type, exc_value, tb)
[Wed Aug 09 17:22:05.908769 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Wed Aug 09 17:22:05.908788 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     rv = self.dispatch_request()
[Wed Aug 09 17:22:05.908807 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Wed Aug 09 17:22:05.908826 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     return self.view_functions[rule.endpoint](**req.view_args)
[Wed Aug 09 17:22:05.908845 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/mycodo/mycodo_flask/authentication_routes.py", line 112, in create_admin
[Wed Aug 09 17:22:05.908863 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]     err=except_msg), "error")
[Wed Aug 09 17:22:05.908882 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_babel/__init__.py", line 552, in gettext
[... cut ...]
[Wed Aug 09 17:22:05.952870 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:40822] InvalidRequestError: This Session's transaction has been rolled back due to a previous exception during flush. To begin a new transaction with this Session, first issue Session.rollback(). Original exception was: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'INSERT INTO users (name, password_hash, email, role, theme) VALUES (?, ?, ?, ?, ?)'] [parameters: (u'admin', '$2b$12$GPSxUb.elYtoARE53XsZjOUy2wtula32V7C.zbmQakdlFZPL6yPSC', u'roger@dabbingsolutions.com', 1, 'slate')]

dabbingsolutions, 2017-08-09:
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_commands.sh initialize

#### Compiling mycodo_wrapper

#### Creating users and directories
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
The user `mycodo' is already a member of `dialout'.

#### Setting permissions

kizniche, 2017-08-09:
Everything should *now* be working. I think I have an idea as to what the issue is in the setup. Thank you very much for the assistance. I'll report back shortly before closing this issue if I think I've fixed it.

kizniche, 2017-08-09:
One last thing. Can you execute this and pate the output:

```
ls -la ~/Mycodo/databases
```

dabbingsolutions, 2017-08-09:
ls -la ~/Mycodo/databases
total 92
drwxrwxr-x 3 mycodo mycodo  4096 Aug  9 17:29 .
drwxrwxr-x 9 mycodo mycodo  4096 Aug  9 16:23 ..
drwxrwxr-x 3 mycodo mycodo  4096 Aug  7 18:15 alembic
-rw-rw-r-- 1 mycodo mycodo  1446 Aug  7 18:15 alembic.ini
-rw-rw-r-- 1 mycodo mycodo    24 Aug  9 16:30 flask_secret_key
-rw-rw-r-- 1 mycodo mycodo 59392 Aug  9 17:29 mycodo.db
-rw-rw-r-- 1 mycodo mycodo   494 Aug  9 17:01 statistics.csv
-rw-rw-r-- 1 mycodo mycodo    12 Aug  9 16:36 statistics.id

Thanks for the software and the awesome support

kizniche, 2017-08-09:
I think I found a solution to the setup issue. This has been plaguing this software for a few weeks now. I'm going to push a patch that will be a minor release version and close this issue. Thanks, and you're welcome :)

kizniche, 2017-08-09:
If you refresh the page, the settings are still saved?

dabbingsolutions, 2017-08-09:
yes they are after manual refresh

kizniche, 2017-08-09:
Can you paste the output of this:

```
sudo tail -n 40 /var/log/apache2/error.log
```

kizniche, 2017-08-09:
Oh I think you need to restart the daemon...

```
sudo service mycodo restart
```

kizniche, 2017-08-09:
It *may*not restart properly. You can check in the Config (gear icon) -> Mycod Logs -> Daemon Log from the web interface. If it doesnt' restart properly, just restart the Pi with:

```
sudo shutdown now -r
```

dabbingsolutions, 2017-08-09:
sudo tail -n 40 /var/log/apache2/error.log
[Wed Aug 09 17:32:26.027992 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Wed Aug 09 17:32:26.028048 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     response = self.full_dispatch_request()
[Wed Aug 09 17:32:26.028066 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Wed Aug 09 17:32:26.028083 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     rv = self.handle_user_exception(e)
[Wed Aug 09 17:32:26.028100 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Wed Aug 09 17:32:26.028117 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     reraise(exc_type, exc_value, tb)
[Wed Aug 09 17:32:26.028134 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Wed Aug 09 17:32:26.028151 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     rv = self.dispatch_request()
[Wed Aug 09 17:32:26.028168 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Wed Aug 09 17:32:26.028185 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     return self.view_functions[rule.endpoint](**req.view_args)
[Wed Aug 09 17:32:26.028202 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Wed Aug 09 17:32:26.028219 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     return func(*args, **kwargs)
[Wed Aug 09 17:32:26.028235 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 733, in page_output
[Wed Aug 09 17:32:26.028252 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     flaskutils.relay_add(form_add_relay)
[Wed Aug 09 17:32:26.028268 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/mycodo/flaskutils.py", line 1745, in relay_add
[Wed Aug 09 17:32:26.028284 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     manipulate_relay('Add', new_relay.id)
[Wed Aug 09 17:32:26.028301 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]   File "/var/www/mycodo/mycodo/flaskutils.py", line 628, in manipulate_relay
[Wed Aug 09 17:32:26.028317 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634]     if return_values[0]:
[Wed Aug 09 17:32:26.028333 2017] [wsgi:error] [pid 2378:tid 1962927152] [remote 192.168.0.24:11634] TypeError: 'NoneType' object has no attribute '__getitem__'
[Wed Aug 09 17:32:46.934426 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478] [2017-08-09 17:32:46,932] ERROR in app: Exception on /output [POST]
[Wed Aug 09 17:32:46.934561 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478] Traceback (most recent call last):
[Wed Aug 09 17:32:46.934584 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Wed Aug 09 17:32:46.934604 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     response = self.full_dispatch_request()
[Wed Aug 09 17:32:46.934625 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Wed Aug 09 17:32:46.934644 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     rv = self.handle_user_exception(e)
[Wed Aug 09 17:32:46.934689 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Wed Aug 09 17:32:46.934708 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     reraise(exc_type, exc_value, tb)
[Wed Aug 09 17:32:46.934725 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Wed Aug 09 17:32:46.934742 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     rv = self.dispatch_request()
[Wed Aug 09 17:32:46.934759 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Wed Aug 09 17:32:46.934776 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     return self.view_functions[rule.endpoint](**req.view_args)
[Wed Aug 09 17:32:46.934793 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[... cut ...]
[Wed Aug 09 17:32:46.934877 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     manipulate_relay('Add', new_relay.id)
[Wed Aug 09 17:32:46.934894 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]   File "/var/www/mycodo/mycodo/flaskutils.py", line 628, in manipulate_relay
[Wed Aug 09 17:32:46.934911 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478]     if return_values[0]:
[Wed Aug 09 17:32:46.934926 2017] [wsgi:error] [pid 2378:tid 1937761328] [remote 192.168.0.24:15478] TypeError: 'NoneType' object has no attribute '__getitem__'

kizniche, 2017-08-09:
Yeah, the daemon is the issue, It needed to be restarted after we messed with the database. Try the daemon restart and if that doesn't work, just restart the Pi.

dabbingsolutions, 2017-08-09:
ok i will restart and start creating timers

kizniche, 2017-08-10:
This issue should now be resolved. If this same issue persists, please open, or if a new issue arises, create a new issue. Thanks, again.

closed 2017-08-10 by kizniche

dabbingsolutions, 2017-08-10:
restarted the the pi and tested the relays they are working created timers and activated it but it will no turn the relay on

kizniche, 2017-08-10:
Can you create a new issue to track it?