kizniche/Mycodo #284: timers not working

dabbingsolutions, 2017-08-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list: restarted the pi and tested the relays they are working fine created a timer and activated it and it doesnt control the relay also cant make changes to the timer settings only del and create a new one 

  - what were you trying to do create a timer to control relay
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-08-10:
Can you list the steps to reproduce the issue?

dabbingsolutions, 2017-08-10:
created the timer and activated the timer the relay didnt come on only if you go to the output page and manually turn it on

kizniche, 2017-08-10:
What type of relay? What relay settings? What type of timer? What timer settings?

dabbingsolutions, 2017-08-10:
unable to change timer settings only can del it and create a new one 

a 2.0a ssr 
ssr pin2 on trigger high state off current draw 1.oa
the time settings once a timer has been created 
start time 1300 end time 0655

 right now its 1815 and if the timer is activated it should turn the ssr on and it does not but it will work manually

dabbingsolutions, 2017-08-10:
from the output page

kizniche, 2017-08-10:
I found the timer issue. I'm working on a fix and should have a release pushed shortly. Thanks for the help.

dabbingsolutions, 2017-08-10:
ok i will try the upgrade and recreate the timers

dabbingsolutions, 2017-08-10:
HTTP request sent, awaiting response... 200 OK
Length: 4324 (4.2K) [text/html]
Saving to: '/dev/null'

     0K ....                                                  100% 5.74M=0.001s

2017-08-09 18:48:07 (5.74 MB/s) - '/dev/null' saved [4324/4324]

Loading robots.txt; please ignore errors.
--2017-08-09 18:48:07--  https://localhost/robots.txt
Reusing existing connection to [localhost]:443.
HTTP request sent, awaiting response... 200 OK
Length: 27 [text/plain]
Saving to: '/dev/null'

     0K                                                       100%  335K=0s

2017-08-09 18:48:07 (335 KB/s) - '/dev/null' saved [27/27]

FINISHED --2017-08-09 18:48:07--
Total wall clock time: 13s
Downloaded: 2 files, 4.2K in 0.001s (5.21 MB/s)
#### Successfully connected to http://localhost

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Done.

Wed Aug  9 18:48:17 MDT 2017
Upgrade completed successfully without errors.

created timers and timers do not turn the relays on you have to go to the output page and click turn on

kizniche, 2017-08-10:
You deleted the previous timer and created a new one?

I tested and it is working for me.

dabbingsolutions, 2017-08-10:
yes i del the timer then ran the update will try it again

kizniche, 2017-08-10:
Please screenshot your Timer page with the applicable timer expanded, and attach it to a comment, please.

dabbingsolutions, 2017-08-10:
[image]
[image]

dabbingsolutions, 2017-08-10:
[image]

kizniche, 2017-08-10:
Can you confirm there are no errors in the daemon log and that clicking ON from the Output page turns the relay on (not clicking OFF). 

If there are no errors and ON is what's used to turn it on, try rebooting the Pi.

dabbingsolutions, 2017-08-10:
I did del the relay and timer the created them again. the timer will not turn on the relay go to the output page and turn it on if you deactivate the timer it will turn the relay off 

yes it is the on button that is turning them on 
[image]

I will reboot the pi

dabbingsolutions, 2017-08-10:
reboot pi and timer does not turn on the relay 
[image]

dabbingsolutions, 2017-08-10:
manually turn the relay on but if I deactivate timer it will turn the relay off

kizniche, 2017-08-10:
What about the daemon log?

dabbingsolutions, 2017-08-10:
date
Wed Aug  9 19:50:27 MDT 2017
whats the command for the demon log?

kizniche, 2017-08-10:
You can view all the logs related to Mycodo under Config (gear icon) -> Mycodo Logs

dabbingsolutions, 2017-08-10:
Display last 
30
 lines from   Login Log   HTTP Log   Daemon Log KeepUp Log   Backup Log   Restore Log   Upgrade Log
Last 30 lines of /var/log/mycodo/mycodo.log:
2017-08-09 19:38:13,293 - mycodo.daemon - INFO - All Sensor controllers stopped
2017-08-09 19:38:13,294 - mycodo.daemon - INFO - All PID controllers stopped
2017-08-09 19:38:13,403 - mycodo.output - INFO - Output controller deactivated in 108.6 ms
2017-08-09 19:38:13,405 - mycodo.daemon - INFO - Mycodo terminated in 0.416 seconds

2017-08-09 19:38:34,308 - mycodo.daemon - INFO - Mycodo daemon v5.1.2 starting
2017-08-09 19:38:34,761 - mycodo.daemon - INFO - Anonymous statistics enabled
2017-08-09 19:38:34,804 - mycodo.daemon - INFO - Starting rpyc server
2017-08-09 19:38:35,145 - mycodo.output - INFO - Relay 1 setup on pin 2 and turned OFF (OFF=LOW)
2017-08-09 19:38:35,146 - mycodo.output - INFO - Relay 2 setup on pin 17 and turned OFF (OFF=LOW)
2017-08-09 19:38:35,147 - mycodo.output - INFO - Relay 3 setup on pin 3 and turned OFF (OFF=LOW)
2017-08-09 19:38:35,148 - mycodo.output - INFO - Relay 4 setup on pin 4 and turned OFF (OFF=LOW)
2017-08-09 19:38:35,150 - mycodo.output - INFO - Output controller activated in 111.1 ms
2017-08-09 19:38:35,288 - mycodo.timer_1 - INFO - Activated in 96.6 ms
2017-08-09 19:38:35,427 - mycodo.timer_2 - INFO - Activated in 88.7 ms
2017-08-09 19:38:35,434 - mycodo.daemon - INFO - All activated timer controllers started
2017-08-09 19:38:35,435 - mycodo.daemon - INFO - All activated sensor controllers started
2017-08-09 19:38:35,436 - mycodo.daemon - INFO - All activated PID controllers started
2017-08-09 19:38:35,437 - mycodo.daemon - INFO - All activated LCD controllers started
2017-08-09 19:38:35,442 - mycodo.daemon - INFO - Mycodo daemon v5.1.2 started in 1.118 seconds
2017-08-09 19:38:35,455 - mycodo.daemon - INFO - 41.908 MB ram in use
2017-08-09 19:39:42,446 - mycodo.timer_1 - INFO - Deactivated in 138.3 ms
2017-08-09 19:39:47,093 - mycodo.timer_2 - INFO - Deactivated in 148.3 ms
2017-08-09 19:39:51,606 - mycodo.timer_1 - INFO - Activated in 101.1 ms
2017-08-09 19:39:56,977 - mycodo.timer_2 - INFO - Activated in 95.1 ms
2017-08-09 20:05:44,702 - mycodo.timer_1 - INFO - Deactivated in 95.1 ms
2017-08-09 20:05:49,194 - mycodo.timer_2 - INFO - Deactivated in 114.8 ms
2017-08-09 20:08:35,746 - mycodo.timer_1 - INFO - Activated in 92.7 ms
2017-08-09 20:08:40,180 - mycodo.timer_2 - INFO - Activated in 95.5 ms

kizniche, 2017-08-10:
I have no idea what the issue is, as I cannot replicate the problem. Try restarting the daemon in debug mode and seeing if you can find an issue in the daemon log.

```
sudo service mycodo stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
```

dabbingsolutions, 2017-08-10:
pi@raspberrypi:~ $ sudo service mycodo stop
pi@raspberrypi:~ $ sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
pi@raspberrypi:~ $

kizniche, 2017-08-10:
I'd suspect you have a hardware issue. Try using different pins.

dabbingsolutions, 2017-08-10:
ok will do

dabbingsolutions, 2017-08-10:
but i think if its hardware it wouldnt work at all

kizniche, 2017-08-10:
That makes sense but I'd like to test the easy stuff first.

kizniche, 2017-08-10:
What I'd really like to see is the daemon log (in debug mode) when the daemon starts, before these lines start to print:

```
2017-08-09 20:20:03,918 - mycodo.timer_1 - DEBUG - Relay 43d209f9-f69a-4ccd-b347-29f40909f66f should be on, but is off. Turning on.
```

dabbingsolutions, 2017-08-10:
i understand i moved to pins 19 and 26 same outcome 
timer will not turn on the relay but you can go to the output page and turn the relay on if you deactivate the timer it will turn off the relay

kizniche, 2017-08-10:
Do other timers work? I suggest making a simple duration on/off timer to test, 30 seconds on, 30 seconds off.

dabbingsolutions, 2017-08-10:
ok just a min let me see if i can find it in the log

dabbingsolutions, 2017-08-10:
no timers work

kizniche, 2017-08-10:
Here is another suggestion to test:

Edit ~/Mycodo/mycodo/controller_timer.py and edit line 132 so it is changed from:

```python
                        modulate_relay = threading.Thread(
                            target=self.control.relay_on_off,
                            args=(self.relay_id,
                                  self.state,))
                        modulate_relay.start()
```

to

```python
                        modulate_relay = threading.Thread(
                            target=self.control.relay_on_off,
                            args=(self.relay_id,
                                  self.state,
                                  0,))
                        modulate_relay.start()
```

Save, then restart the daemon (`sudo service mycodo restart`). If the daemon is currently running in debug mode, you must shut it down a different way (`~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_client.py -t` then start with `sudo service mycodo start`).

kizniche, 2017-08-10:
You installed Mycodo on top of a fresh install of Raspbian, correct?

kizniche, 2017-08-10:
And what version Pi are you using?

kizniche, 2017-08-10:
Okay. If a duration timer isn't working, then the above code edit will 99.9% chance not affect anything.

dabbingsolutions, 2017-08-10:
pi 3 b just got them yes installed noobs then mycodo trying the edit

kizniche, 2017-08-10:
There is your issue, most likely. Noobs is not raspbian.

dabbingsolutions, 2017-08-10:
going back to the deamon pulled up 3000 files

kizniche, 2017-08-10:
Please install Mycodo over Raspbian and see if it resolves your issue.

dabbingsolutions, 2017-08-10:
ok will try raspbian direct install

kizniche, 2017-08-10:
Well, it may be raspbian, but I've never tested Mycodo on a Noobs install, only Raspbian. I'm not sure of the difference, but at this point, my only suggestion will be to try to reinstall Raspbian and Mycodo from scratch and see if the issue persists. It could be a software issue picked up along the wy iof installing wither.

dabbingsolutions, 2017-08-10:
i will download it and install n a different sd card

dabbingsolutions, 2017-08-10:
Ok murphy has fought me every step of the way finally have raspbian installed and mycodo the install ran fine. Tried changing the settings for the relay and get Error: Modify Relay: 'NoneType' object has no attribute '__getitem__' each time but the settings are being saved. Tested relays come on and off manually . Created the timers and same issue they will not turn the relays on. Here is the last 3000 lines of the deamon
 2017-08-10 17:13:13,170 - mycodo.daemon - INFO - Mycodo daemon v5.1.2 starting
2017-08-10 17:13:13,315 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////home/pi/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such table: camera [SQL: u'SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.camera_type AS camera_camera_type, camera.library AS camera_library, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.height AS camera_height, camera.width AS camera_width, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.relay_id AS camera_relay_id, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number \nFROM camera']'
2017-08-10 17:13:13,317 - mycodo.daemon - ERROR - Could not read camera table: (sqlite3.OperationalError) no such table: camera [SQL: u'SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.camera_type AS camera_camera_type, camera.library AS camera_library, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.height AS camera_height, camera.width AS camera_width, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.relay_id AS camera_relay_id, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number \nFROM camera']
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 594, in refresh_daemon_camera_settings
    Camera, entry='all')
  File "/var/www/mycodo/mycodo/utils/database.py", line 56, in db_retrieve_table_daemon
    return_table = return_table.all()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2613, in all
    return list(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2761, in __iter__
    return self._execute_and_instances(context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2776, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 914, in execute
    return meth(self, multiparams, params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/sql/elements.py", line 323, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1010, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
    context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1341, in _handle_dbapi_exception
    exc_info
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
    context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 450, in do_execute
    cursor.execute(statement, parameters)
[... cut ...]
2017-08-10 17:33:56,511 - mycodo.daemon - INFO - All activated PID controllers started
2017-08-10 17:33:56,512 - mycodo.daemon - INFO - All activated LCD controllers started
2017-08-10 17:33:56,513 - mycodo.daemon - INFO - Mycodo daemon v5.1.2 started in 0.900 seconds
2017-08-10 17:33:56,526 - mycodo.daemon - INFO - 41.376 MB ram in use
2017-08-10 17:40:59,255 - mycodo.timer_1 - INFO - Activated in 88.6 ms
2017-08-10 17:41:04,046 - mycodo.timer_2 - INFO - Activated in 99.7 ms
2017-08-10 17:41:42,192 - mycodo.output - WARNING - Relay 4 (flower 2) is already on.

kizniche, 2017-08-11:
Can you please attach the entire setup.log? You may have to add it to a zip archive to attach here.

dabbingsolutions, 2017-08-11:
i do not see the setup log i have login/http/deamon/keep up/backup/restore/upgrade logs

kizniche, 2017-08-11:
It is located at ~/Mycodo/install/setup.log

Drag and drop it into the github comment box and it will attach it. You may have to be root to see it, so changing its permissions may be necessary:

```
sudo chown pi.pi ~/Mycodo/install/setup.log
```

replacing pi with your user

dabbingsolutions, 2017-08-11:
user created was admin 
pi@raspberrypi:~ $ sudo chown pi.admin ~/Mycodo/install/setup.log???????

kizniche, 2017-08-11:
This command will identify what user you are:

```
whoami
```

dabbingsolutions, 2017-08-11:
pi@raspberrypi:~ $ whoami
pi
pi@raspberrypi:~ $ sudo chown pi.pi ~/Mycodo/install/setup.log
pi@raspberrypi:~ $

kizniche, 2017-08-11:
Yo have to drag and drop it into the comment box to attach or click "selecting them" link at the bottom of the box.

dabbingsolutions, 2017-08-11:
### Mycodo installation beginning at 08-10-2017 16:56:36

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap currently set to 100 MB. Changing to 512 MB and restarting
Stopping dphys-swapfile (via systemctl): dphys-swapfile.service.
Starting dphys-swapfile (via systemctl): dphys-swapfile.service.

#### Updating apt repositories
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
Hit http://archive.raspberrypi.org jessie/main armhf Packages
Hit http://mirrordirector.raspbian.org jessie/contrib armhf Packages
Hit http://mirrordirector.raspbian.org jessie/non-free armhf Packages
Hit http://archive.raspberrypi.org jessie/ui armhf Packages
Hit http://mirrordirector.raspbian.org jessie/rpi armhf Packages
Ign http://archive.raspberrypi.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/contrib Translation-en
Ign http://archive.raspberrypi.org jessie/main Translation-en
Ign http://mirrordirector.raspbian.org jessie/main Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/main Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/non-free Translation-en
Ign http://archive.raspberrypi.org jessie/ui Translation-en
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en_GB
Ign http://mirrordirector.raspbian.org jessie/rpi Translation-en
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  python-colorama python-distlib python-html5lib python-wheel
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  python-pip*
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 481 kB disk space will be freed.
(Reading database ... 
(Reading database ... 5%
(Reading database ... 10%
(Reading database ... 15%
(Reading database ... 20%
(Reading database ... 25%
(Reading database ... 30%
(Reading database ... 35%
(Reading database ... 40%
(Reading database ... 45%
(Reading database ... 50%
(Reading database ... 55%
(Reading database ... 60%
(Reading database ... 65%
(Reading database ... 70%
(Reading database ... 75%
(Reading database ... 80%
(Reading database ... 85%
(Reading database ... 90%
(Reading database ... 95%
(Reading database ... 100%
(Reading database ... 115456 files and directories currently installed.)
Removing python-pip (1.5.6-5) ...
Processing triggers for man-db (2.7.5-1~bpo8+1) ...

#### Installing prerequisite apt packages and update pip
Hit http://archive.raspberrypi.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie InRelease
Hit http://mirrordirector.raspbian.org jessie/main armhf Packages
[... 859 lines cut ...]
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
WARNING: combining -O with -r or -p will mean that all downloaded content
will be placed in the single file you specified.

--2017-08-10 17:12:05--  http://localhost/
Resolving localhost (localhost)... ::1, 127.0.0.1
Connecting to localhost (localhost)|::1|:80... connected.
HTTP request sent, awaiting response... 500 Internal Server Error
2017-08-10 17:12:06 ERROR 500: Internal Server Error.

#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.0.24/, or whatever your Pi's
IP address is, to create an admin user and log in.

kizniche, 2017-08-11:
Please attach it next time instead of pasting the text. That many lines in comments makes navigating this thread unmanageable.

kizniche, 2017-08-11:
I'm going to test a fresh install myself. I should be back within an hour with some new info.

dabbingsolutions, 2017-08-11:
ok

kizniche, 2017-08-11:
Here's what I found:

1. An issue during the install. I fixed it and now the install should go completely smoothly.
2. Timers work out of the box

I just released v5.1.3, however this only includes the install fix.

This is what I did to demonstrate timers work, using the most minimal setup:

1. Install Mycodo and create admin user
2. Go to ```Output -> Devices``` and add 2 ```Relay: Wired```
3. Set the first relay's ```BCM Pin``` to 20, then ```Save```
4. Set the second relay's ```BCM Pin``` to 21, then ```Save```
5. Go to ```Mode -> Timers```
6. Click ```New Timer```, enter a name and select the ```Timer Type``` of  ```Relay (ON/OFF)```
7. Select the second radio button (middle), set ```Start Time``` to ```12:00``` and ```End Time``` to ```6:00```, then click ```Save```
8. Select the third radio button (bottom), set ```On (sec)``` to ```20``` and ```Off (sec)``` to ```20```, then click ```Save```
9. Click ```Activate``` for first timer
10. Click ```Activate``` for second timer
11. Go to the ```Output -> Devices``` page and see the relays turn on (and on/off cycle in the case of the second relay.

dabbingsolutions, 2017-08-11:
Ok I will install a clean copy of rasbian and mycodo

dabbingsolutions, 2017-08-11:
good news its up and running time to add the sensors ect

closed 2017-08-11 by dabbingsolutions

kizniche, 2017-08-11:
I see you closed the issue. Does this mean timers are now working for you?

dabbingsolutions, 2017-08-11:
Yes they are