kizniche/Mycodo #286: 500 error after upgrade

Ericktz, 2017-08-16:
```
[Wed Aug 16 00:20:52.861333 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750] mod_wsgi (pid=18485): Target WSGI script '/var/www/mycodo/mycodo_flask.wsgi' cannot be loaded as Python module.
[Wed Aug 16 00:20:52.861452 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750] mod_wsgi (pid=18485): Exception occurred processing WSGI script '/var/www/mycodo/mycodo_flask.wsgi'.
[Wed Aug 16 00:20:52.861618 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750] Traceback (most recent call last):
[Wed Aug 16 00:20:52.861696 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/mycodo_flask.wsgi", line 8, in <module>
[Wed Aug 16 00:20:52.861810 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     from mycodo.start_flask_ui import app as application
[Wed Aug 16 00:20:52.861851 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/mycodo/start_flask_ui.py", line 11, in <module>
[Wed Aug 16 00:20:52.861918 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     app = create_app()  # required by the wsgi config and main()
[Wed Aug 16 00:20:52.861956 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/mycodo/mycodo_flask/app.py", line 67, in create_app
[Wed Aug 16 00:20:52.862019 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     register_extensions(app)
[Wed Aug 16 00:20:52.862057 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/mycodo/mycodo_flask/app.py", line 106, in register_extensions
[Wed Aug 16 00:20:52.862118 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     populate_db()
[Wed Aug 16 00:20:52.862154 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/mycodo/databases/models/__init__.py", line 138, in populate_db
[Wed Aug 16 00:20:52.862219 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     if not Misc.query.count():
[Wed Aug 16 00:20:52.862255 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2980, in count
[Wed Aug 16 00:20:52.862323 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     """
[Wed Aug 16 00:20:52.862359 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2749, in scalar
[Wed Aug 16 00:20:52.862422 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     raise sa_exc.ArgumentError(
[Wed Aug 16 00:20:52.862458 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2718, in one
[Wed Aug 16 00:20:52.862521 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     def all(self):
[Wed Aug 16 00:20:52.862557 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2761, in __iter__
[Wed Aug 16 00:20:52.862619 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     (note this may consist of multiple result rows if join-loaded
[Wed Aug 16 00:20:52.862698 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2776, in _execute_and_instances
[Wed Aug 16 00:20:52.862761 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     ret = list(self[0:1])
[Wed Aug 16 00:20:52.862797 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 914, in execute
[Wed Aug 16 00:20:52.862863 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     In the case that a plain SQL string is passed, and the underlying
[Wed Aug 16 00:20:52.862900 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/elements.py", line 323, in _execute_on_connection
[Wed Aug 16 00:20:52.862964 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     """
[Wed Aug 16 00:20:52.863000 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1010, in _execute_clauseelement
[Wed Aug 16 00:20:52.863063 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     """Execute a sql.ClauseElement object."""
[Wed Aug 16 00:20:52.863100 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1146, in _execute_context
[Wed Aug 16 00:20:52.863161 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     evt_handled = False
[Wed Aug 16 00:20:52.863197 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1341, in _handle_dbapi_exception
[Wed Aug 16 00:20:52.863259 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     else:
[Wed Aug 16 00:20:52.863295 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 202, in raise_from_cause
[Wed Aug 16 00:20:52.863359 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     cause = exc_value if exc_value is not exception else None
[Wed Aug 16 00:20:52.863395 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1139, in _execute_context
[Wed Aug 16 00:20:52.863457 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     if self._echo:
[Wed Aug 16 00:20:52.863492 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]   File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 450, in do_execute
[Wed Aug 16 00:20:52.863556 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750]     def do_begin(self, dbapi_connection):
[Wed Aug 16 00:20:52.863735 2017] [wsgi:error] [pid 18485] [remote 192.168.20.101:5750] OperationalError: (sqlite3.OperationalError) no such column: misc.max_amps [SQL: u'SELECT count(*) AS count_1 \\nFROM (SELECT misc.id AS misc_id, misc.dismiss_notification AS misc_dismiss_notification, misc.force_https AS misc_force_https, misc.hide_alert_info AS misc_hide_alert_info, misc.hide_alert_success AS misc_hide_alert_success, misc.hide_alert_warning AS misc_hide_alert_warning, misc.hide_tooltips AS misc_hide_tooltips, misc.language AS misc_language, misc.login_message AS misc_login_message, misc.max_amps AS misc_max_amps, misc.relay_usage_cost AS misc_relay_usage_cost, misc.relay_usage_currency AS misc_relay_usage_currency, misc.relay_usage_dayofmonth AS misc_relay_usage_dayofmonth, misc.relay_usage_volts AS misc_relay_usage_volts, misc.relay_usage_report_gen AS misc_relay_usage_report_gen, misc.relay_usage_report_span AS misc_relay_usage_report_span, misc.relay_usage_report_day AS misc_relay_usage_report_day, misc.relay_usage_report_hour AS misc_relay_usage_report_hour, misc.stats_opt_out AS misc_stats_opt_out \\nFROM misc) AS anon_1']

```

ocurr after reboot , after upgrade 5.1.6

kizniche, 2017-08-16:
What version did you upgrade from? Can you please attach the upgrade log?

Ericktz, 2017-08-16:
Apparently it was an error caused by an incomplete update.

`
#### Upgrade to v5.1.6 initiated 2017-08-11_23-53-00 ####

#### Beginning Upgrade: Stage 1 of 2 ####
Stopping the Mycodo daemon...Done.
Downloading latest Mycodo version to /home/pi/mycodo-5.1.6.tar.gz...Done.
Creating /tmp/Mycodo-5.1.6...Done.
Extracting /home/pi/mycodo-5.1.6.tar.gz to /tmp/Mycodo-5.1.6...Done.
The archive directory /tmp/Mycodo-5.1.6/old exists. Removing...Done.
Removing /home/pi/mycodo-5.1.6.tar.gz...Done.
Copying /tmp/Mycodo-5.1.6/.upgrade status file to /tmp/Mycodo-5.1.6...Done.
Moving env directory...Done.
Copying databases from /home/pi/Mycodo/databases/ to /tmp/Mycodo-5.1.6/databases...Done.
Copying flask_secret_key from /home/pi/Mycodo/databases/ to /tmp/Mycodo-5.1.6/databases...Done.
Copying statistics ID...Done.
Copying SSL certificates...Done.
Moving cameras directory...Done.
#### Stage 1 of 2 Complete ####
#### Continuing Upgrade: Stage 2 of 2 ####

Moving old Mycodo from /home/pi/Mycodo to /var/Mycodo-backups/Mycodo-backup-2017-08-11_23-53-00-5.0.49...Done.
Moving new Mycodo from /tmp/Mycodo-5.1.6 to /home/pi/Mycodo...Done.

#### Compiling mycodo_wrapper

#### Creating users and directories
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
The user `mycodo' is already a member of `dialout'.

#### Setting permissions

#### Setting permissions

Running post-upgrade script...

#### Removing statistics file
rm: cannot remove '/home/pi/Mycodo/databases/statistics.csv': No such file or directory

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.
#### Virtualenv already exists, skipping creation

#### Installing gpiod
Archive:  pigpio.zip
   creating: PIGPIO/
  inflating: PIGPIO/setup.py         
  inflating: PIGPIO/command.h        
  inflating: PIGPIO/pigpiod.c        
  inflating: PIGPIO/pigpiod_if2.3    
  inflating: PIGPIO/pigs.c           
  inflating: PIGPIO/x_pigpiod_if.c   
  inflating: PIGPIO/pigpiod_if2.c    
  inflating: PIGPIO/pigpiod.1        
  inflating: PIGPIO/README           
  inflating: PIGPIO/command.c        
  inflating: PIGPIO/pigpio.h         
  inflating: PIGPIO/custom.cext      
  inflating: PIGPIO/pigpiod_if.h     
  inflating: PIGPIO/Makefile         
  inflating: PIGPIO/pig2vcd.c        
  inflating: PIGPIO/pigpio.py        
  inflating: PIGPIO/pigs.1           
  inflating: PIGPIO/pigpio.c         
  inflating: PIGPIO/x_pipe           
  inflating: PIGPIO/pigpiod_if.c     
[... 319 lines cut ...]
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin: 
Fetched 1747 kB in 10s (165 kB/s)
(Reading database ... 154349 files and directories currently installed.)
Preparing to unpack .../apache2_2.4.10-10+deb8u10_armhf.deb ...
Unpacking apache2 (2.4.10-10+deb8u10) over (2.4.10-10+deb8u9) ...
Preparing to unpack .../apache2-bin_2.4.10-10+deb8u10_armhf.deb ...
Unpacking apache2-bin (2.4.10-10+deb8u10) over (2.4.10-10+deb8u9) ...
Preparing to unpack .../apache2-data_2.4.10-10+deb8u10_all.deb ...
Unpacking apache2-data (2.4.10-10+deb8u10) over (2.4.10-10+deb8u9) ...
Preparing to unpack .../wget_1.16-1+deb8u2_armhf.deb ...
Unpacking wget (1.16-1+deb8u2) over (1.16-1+deb8u1) ...
Processing triggers for systemd (215-17+deb8u5) ...
Processing triggers for man-db (2.7.0.2-5) ...
Processing triggers for install-info (5.2.0.dfsg.1-6) ...
Setting up apache2-bin (2.4.10-10+deb8u10) ...
Setting up apache2-data (2.4.10-10+deb8u10) ...
Setting up apache2 (2.4.10-10+deb8u10) ...`

Was fixed with a manual update:

`sudo ~/Mycodo/mycodo/scripts/upgrade_post.sh
`

closed 2017-08-17 by Ericktz