kizniche/Mycodo #287: AM2315 negatives values in graph view

Ericktz, 2017-08-17:
[image]
version 5.1.6

in live graph page, the values of measurements apparently seem wrong, in some cases two or more consecutively values are high values of relative humidity(1000-2000%), and in temperature shows negative values (-3000º).... 

in live stats, all apparently seems ok.... 

[image]
[image]

kizniche, 2017-08-17:
My AM2315 sensor with Mycodo 5.1.6 isn't displaying this behavior. Try deleting the sensor and adding it again and see if the issue persists.

kizniche, 2017-08-17:
Additionally, try rebooting your Pi.

kizniche, 2017-10-12:
This may be a sensor issue. Do you have a second sensor that you can use to verify the behavior? Have you ran other code (non-Mycodo) and does it display the same results?

closed 2017-10-12 by kizniche