kizniche/Mycodo #288: I cant upgrade 5.0.49 for latest 5.1.6

molexuse, 2017-08-17:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.0.49
#### Problem Description
Try to Upgrade to Latest version: 5.1.6
 
### Errors
Message
- Success: The upgrade has started. The daemon will be stopped during the upgrade.
But not upgrade.
I deleted file  ~/Mycodo/.upgrade but nothing happens, try it again.
Upgrade log is old
Trying to upgrade system files and run upgrade but nothigh happens.

kizniche, 2017-08-17:
Please attach the excerpt from the last upgrade attempt from the upgrade log.

molexuse, 2017-08-17:
[mycodoupgrade.log.txt](https://github.com/kizniche/Mycodo/files/1232461/mycodoupgrade.log.txt)

kizniche, 2017-08-17:
How long did you give the upgrade before doing anything to the Pi?

molexuse, 2017-08-18:
5-15 min.
Shutdown and restart command in web interface also don`t work.
In console work. 
Maybe me start upgrade  from the console?

kizniche, 2017-08-18:
I understand there is an issue. If you please answer the questions I ask, I can better understand what happened and potentially fix the issue.

kizniche, 2017-08-18:
Try this from the command line:

```sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_post.sh```

molexuse, 2017-08-18:
sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh
with this command upgrade is successful!

kizniche, 2017-08-18:
Good to hear.
Not sure what happened during your upgrade, but the upgrade log was corrupt at the end, so I assume it was abnormal.

closed 2017-08-18 by kizniche