kizniche/Mycodo #289: Relay on  Trigger

RynFlutsch, 2017-08-27:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.1.6
#### Problem Description
Please list:
I can change relay 'on trigger' to off
tried to save but its useless he isnt doing it.

 old relays have the right setting. updated to 5.1.6 now its a problem for new relays

kizniche, 2017-08-28:
I don't currently have access to me development Pis to test. Is there an error of any kind returned or showing up in the Daemon or HTTP logs?

RynFlutsch, 2017-08-28:
apache2 error.log
`
[Mon Aug 28 17:31:31.902511 2017] [ssl:warn] [pid 578:tid 3069456912] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Mon Aug 28 17:31:31.950222 2017] [wsgi:warn] [pid 578:tid 3069456912] mod_wsgi: Compiled for Python/2.7.8.
[Mon Aug 28 17:31:31.950347 2017] [wsgi:warn] [pid 578:tid 3069456912] mod_wsgi: Runtime using Python/2.7.9.
[Mon Aug 28 17:31:32.052115 2017] [mpm_event:notice] [pid 578:tid 3069456912] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Mon Aug 28 17:31:32.052415 2017] [core:notice] [pid 578:tid 3069456912] AH00094: Command line: '/usr/sbin/apache2'
error.log (END)
`

RynFlutsch, 2017-08-28:
tried with fresh install but he refreshes the live page endless now(white screen)
same error on apache2 with fresh instal
apache2/error.log
`[Mon Aug 28 17:44:20.733461 2017] [ssl:warn] [pid 508:tid 3069153808] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Mon Aug 28 17:44:21.958677 2017] [ssl:warn] [pid 580:tid 3069153808] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Mon Aug 28 17:44:22.026385 2017] [wsgi:warn] [pid 580:tid 3069153808] mod_wsgi: Compiled for Python/2.7.8.
[Mon Aug 28 17:44:22.026527 2017] [wsgi:warn] [pid 580:tid 3069153808] mod_wsgi: Runtime using Python/2.7.9.
[Mon Aug 28 17:44:22.152452 2017] [mpm_event:notice] [pid 580:tid 3069153808] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Mon Aug 28 17:44:22.152689 2017] [core:notice] [pid 580:tid 3069153808] AH00094: Command line: '/usr/sbin/apache2'`
mycodo.log
`2017-08-28 17:44:58,304 - mycodo.daemon - INFO - Mycodo daemon v5.1.6 starting
2017-08-28 17:44:59,857 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////root/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such table: camera [SQL: u'SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.camera_type AS camera_camera_type, camera.library AS camera_library, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.height AS camera_height, camera.width AS camera_width, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.relay_id AS camera_relay_id, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number \nFROM camera']'
2017-08-28 17:44:59,863 - mycodo.daemon - ERROR - Could not read camera table: (sqlite3.OperationalError) no such table: camera [SQL: u'SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.camera_type AS camera_camera_type, camera.library AS camera_library, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.height AS camera_height, camera.width AS camera_width, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_white_balance, camera.relay_id AS camera_relay_id, camera.cmd_pre_camera AS camera_cmd_pre_camera, camera.cmd_post_camera AS camera_cmd_post_camera, camera.stream_started AS camera_stream_started, camera.timelapse_started AS camera_timelapse_started, camera.timelapse_paused AS camera_timelapse_paused, camera.timelapse_start_time AS camera_timelapse_start_time, camera.timelapse_end_time AS camera_timelapse_end_time, camera.timelapse_interval AS camera_timelapse_interval, camera.timelapse_next_capture AS camera_timelapse_next_capture, camera.timelapse_capture_number AS camera_timelapse_capture_number \nFROM camera']
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 594, in refresh_daemon_camera_settings
    Camera, entry='all')
  File "/var/www/mycodo/mycodo/utils/database.py", line 56, in db_retrieve_table_daemon
    return_table = return_table.all()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2724, in all
    return list(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2876, in __iter__
    return self._execute_and_instances(context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2899, in _execute_and_instances`

kizniche, 2017-08-28:
What version did you upgrade from?

kizniche, 2017-08-29:
A fix was included in the latest release.

closed 2017-08-29 by kizniche

RynFlutsch, 2017-08-29:
Not Sure i think 5.0.46 to 5.1.6
Upgraded 'on trigger' is working, thank you.

Same issue now with 'Start State'
I can't change it successfully from 'on' to off.

kizniche, 2017-08-29:
I'll look into it later today. Thanks.

kizniche, 2017-08-29:
Just released v5.1.8 that incorporates this fix.

RynFlutsch, 2017-08-30:
i upgraded to 5.1.8,
start state is set right, and saved, but it doesn't apply to the relays. Reboot, delete relays and configure new did not helped. Any idea ?

RynFlutsch, 2017-08-30:
fresh install (sd card)
first configuration try of the first relay
[image]

kizniche, 2017-08-30:
What button did you press to yield that error?
Are there any errors in the HTTP or Daemon logs?

RynFlutsch, 2017-08-30:
change bcm than on trigger to off and save

kizniche, 2017-08-30:
Did you mean "Start State" to Off?
Is your relay actually on when low?

kizniche, 2017-08-30:
I'll be able to investigate this later when I'm home with my development Pis.

RynFlutsch, 2017-08-30:
no changed only On Trigger
below is your standard setup without any changes
[image]

RynFlutsch, 2017-08-30:
add 2 wired relays, click add output device:
[image]

kizniche, 2017-08-30:
You said you changed "On Trigger" to "Off", but the only options for that are "High" and "Low"

kizniche, 2017-08-30:
Are there any errors in the Daemon log?

RynFlutsch, 2017-08-30:
my bad "low" is right.
i can check in arround 10h not at the pi

kizniche, 2017-08-30:
You can access the logs from the web interface.

Go to Config (gear icon) -> Mycodo Logs.

RynFlutsch, 2017-08-30:
i am physically not there and the network is offline i am back in arround 10h to access the logs

kizniche, 2017-08-30:
I couldn't reproduce any of the errors you experienced on any of my Pis.

> fresh install (sd card)

Does this mean you start with a fresh write of Raspbian, and the first thing done is install Mycodo?

kizniche, 2017-08-31:
The latest commit just mentioned only handles preventing the 500 error. I'm still not sure what's causing your error.

RynFlutsch, 2017-08-31:
> Does this mean you start with a fresh write of Raspbian, and the first thing done is install Mycodo?

Yes fresh but No, i know your installation manual ;)

Checked on my raspi 3B upgraded from 5.0.46 to 5.1.8 without problems
no errors there.

Raspberry Zero W mycodo.log
`2017-08-31 10:21:16,962 - mycodo.sensor_2 - ERROR - All measurements returned failed CRC
2017-08-31 10:21:18,966 - mycodo.sensor_2 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-08-31 10:21:28,095 - mycodo.sensor_2 - ERROR - All measurements returned failed CRC`
I am not sure whats up, but maybe 1 or 2 relais got damaged with high inrush current on cold start

RynFlutsch, 2017-08-31:
It looks like that error comes from the am2315

kizniche, 2017-08-31:
Sensors wouldn't affect the web interface. Additionally, the web interface should work even if your relays don't. You can have essentially unconnected pins configured as relays and it will work. I'd suggest start with a fresh install of Raspbian and Mycodo.

RynFlutsch, 2017-09-07:
I reproduced the error with another Raspberry. Clean installation:
5.1.8 without errors( no localhost connection at installation) Tried to add 8 Relays:

Apache:
`[Wed Sep 06 18:43:16.934082 2017] [ssl:warn] [pid 574:tid 1995812864] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Wed Sep 06 18:43:17.049476 2017] [ssl:warn] [pid 617:tid 1995812864] AH01909: 127.0.0.1:443:0 server certificate does NOT include an ID which matches the server name
[Wed Sep 06 18:43:17.083693 2017] [wsgi:warn] [pid 617:tid 1995812864] mod_wsgi: Compiled for Python/2.7.8.
[Wed Sep 06 18:43:17.083768 2017] [wsgi:warn] [pid 617:tid 1995812864] mod_wsgi: Runtime using Python/2.7.9.
[Wed Sep 06 18:43:17.091063 2017] [mpm_event:notice] [pid 617:tid 1995812864] AH00489: Apache/2.4.10 (Raspbian) OpenSSL/1.0.1t mod_wsgi/4.3.0 Python/2.7.9 configured -- resuming normal operations
[Wed Sep 06 18:43:17.091203 2017] [core:notice] [pid 617:tid 1995812864] AH00094: Command line: '/usr/sbin/apache2'
[Wed Sep 06 18:58:18.206251 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798] [2017-09-06 18:58:18,185] ERROR in app: Exception on /output [POST]
[Wed Sep 06 18:58:18.206335 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798] Traceback (most recent call last):
[Wed Sep 06 18:58:18.206349 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
[Wed Sep 06 18:58:18.206362 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     response = self.full_dispatch_request()
[Wed Sep 06 18:58:18.206374 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
[Wed Sep 06 18:58:18.206386 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     rv = self.handle_user_exception(e)
[Wed Sep 06 18:58:18.206398 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
[Wed Sep 06 18:58:18.206410 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     reraise(exc_type, exc_value, tb)
[Wed Sep 06 18:58:18.206421 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
[Wed Sep 06 18:58:18.206433 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     rv = self.dispatch_request()
[Wed Sep 06 18:58:18.206444 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
[Wed Sep 06 18:58:18.206457 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     return self.view_functions[rule.endpoint](**req.view_args)
[Wed Sep 06 18:58:18.206468 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
[Wed Sep 06 18:58:18.206481 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     return func(*args, **kwargs)
[Wed Sep 06 18:58:18.206492 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 733, in page_output
[Wed Sep 06 18:58:18.206503 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     flaskutils.relay_add(form_add_relay)
[Wed Sep 06 18:58:18.206515 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/mycodo/flaskutils.py", line 1747, in relay_add
[Wed Sep 06 18:58:18.206526 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     manipulate_relay('Add', new_relay.id)
[Wed Sep 06 18:58:18.206537 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]   File "/var/www/mycodo/mycodo/flaskutils.py", line 630, in manipulate_relay
[Wed Sep 06 18:58:18.206549 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798]     if return_values[0]:
[Wed Sep 06 18:58:18.206579 2017] [wsgi:error] [pid 634:tid 1962714160] [remote 192.168.188.20:39798] TypeError: 'NoneType' object has no attribute '__getitem__'
`
Mycodo.log
`2017-09-06 18:43:22,724 - mycodo.daemon - INFO - Mycodo daemon v5.1.8 starting
2017-09-06 18:43:22,914 - mycodo.databases.utils - ERROR - Error raised in session_scope.  Session will be rolled back: db_uri='sqlite:////root/Mycodo/mycodo/../databases/mycodo.db', error='(sqlite3.OperationalError) no such table: camera [SQL: u'SELECT camera.id AS camera_id, camera.unique_id AS camera_unique_id, camera.name AS camera_name, camera.camera_type AS camera_camera_type, camera.library AS camera_library, camera.opencv_device AS camera_opencv_device, camera.hflip AS camera_hflip, camera.vflip AS camera_vflip, camera.rotation AS camera_rotation, camera.height AS camera_height, camera.width AS camera_width, camera.brightness AS camera_brightness, camera.contrast AS camera_contrast, camera.exposure AS camera_exposure, camera.gain AS camera_gain, camera.hue AS camera_hue, camera.saturation AS camera_saturation, camera.white_balance AS camera_ [... cut ...]
OperationalError: (sqlite3.OperationalError) no such table: alembic_version [SQL: u'SELECT alembic_version.version_num AS alembic_version_version_num \nFROM alembic_version\n LIMIT ? OFFSET ?'] [parameters: (1, 0)]
2017-09-06 18:50:11,156 - mycodo.sensor_1 - INFO - Activated in 110.9 ms
2017-09-06 18:58:18,177 - mycodo.daemon - ERROR - Could not set up relay: 'NoneType' object has no attribute 'relay_setup'
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 694, in relay_setup
    return self.controller['Relay'].relay_setup(action, relay_id)
AttributeError: 'NoneType' object has no attribute 'relay_setup'
`

kizniche, 2017-09-07:
All the "no such table" errors indicate the database doesn't exist. If visiting the web interface for the first time allowed you to create an admin user, the database was created.

Always keep track of the time you are doing things. If these errors were before the database was created, do not consider them valid (because errors are expected if the database doesn't exist).

Please provide information about your raspberry pi and a link to the exact Raspbian image you used.

RynFlutsch, 2017-09-19:
Weird, cause DB was created during installation.
Reboot same bug, next day no errors, till now ...

Now all is running ;)

Looks like my error was like [https://github.com/kizniche/Mycodo/issues/291](url)

Raspberry Pi 3B v 1.2
[http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Jessie).7z](url)

Yes i know its not original Raspbian Lite but based on and i tried first with Raspbian.

Mycodo include some algorithm to correct Runtime-delays on longtime logging?