kizniche/Mycodo #290: BME280 sensor

viktorarmani55, 2017-09-06:
## Mycodo Issue Report:
5.1.8
I've used the new Raspbian stretch and Jessie with the latest Mycodo version 
 
#### Problem Description
Please list:
  - I'm trying to setup numerous BME280 sensors using a TCA9548A multiplexer
  - I've followed the instructions supplied by yourself and Adafruit
 
### Errors
  - There seems to be no response from the sensors (Qty 5),
  - I've restarted, shutdown, reconnected wires, and the same result i.e. show "0" values (see images attached)
 
### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Starting up Mycodo 5.1.8
 2. Go to Input tab - ensure the settings are correct (as suggested by Kyle)
 3. Go to Data tab - live measurements
 4. No data output from sensor.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-09-07:
Let's start basic and see if you can get a reading with only one sensor connected directly to the Pi and using the Adafruit module.

Install the python module

```
sudo pip install -e git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
```

Run the example script, which should return the sensor values

```
git clone https://github.com/adafruit/Adafruit_Python_BME280
cd Adafruit_Python_BME280
python Adafruit_BME280_Example.py
```

viktorarmani55, 2017-09-07:
Hi Kyle, I tried the following and got the following message (see below).  I'll send a screen shot.
pi@raspberrypi:~ $ sudo pip install -e git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
Obtaining adafruit-bme280 from git+git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
  Cloning git://github.com/adafruit/Adafruit_Python_BME280.git to ./src/adafruit-bme280
Installing collected packages: adafruit-bme280
  Running setup.py develop for adafruit-bme280
Successfully installed adafruit-bme280
pi@raspberrypi:~ $ git clone https://github.com/adafruit/Adafruit_Python_BME280
Cloning into 'Adafruit_Python_BME280'...
remote: Counting objects: 77, done.
remote: Total 77 (delta 0), reused 0 (delta 0), pack-reused 77
Unpacking objects: 100% (77/77), done.
Checking connectivity... done.
pi@raspberrypi:~ $ cd Adafruit_Python_BME280
pi@raspberrypi:~/Adafruit_Python_BME280 $ python Adafruit_BME280_Example.py
Traceback (most recent call last):
  File "Adafruit_BME280_Example.py", line 3, in <module>
    sensor = BME280(t_mode=BME280_OSAMPLE_8, p_mode=BME280_OSAMPLE_8, h_mode=BME280_OSAMPLE_8)
  File "/home/pi/Adafruit_Python_BME280/Adafruit_BME280.py", line 127, in __init__
    import Adafruit_GPIO.I2C as I2C
ImportError: No module named Adafruit_GPIO.I2C
pi@raspberrypi:~/Adafruit_Python_BME280 $

kizniche, 2017-09-07:
Better yet, let's do it in the Mycodo Python environment so we have the Adafruit_GPIO module available.

```
~/Mycodo/env/bin/python Adafruit_BME280_Example.py
```

viktorarmani55, 2017-09-07:
What is the best way to send you the error message and/or screen shots.

kizniche, 2017-09-07:
I prefer just copying the error text. If you put three backticks (```) before and after the text, it will preserve the formatting.

viktorarmani55, 2017-09-07:
```pi@raspberrypi:~/Adafruit_Python_BME280 $ ~/Mycodo/env/bin/python Adafruit_BME280_Example.py
Traceback (most recent call last):
  File "Adafruit_BME280_Example.py", line 3, in <module>
    sensor = BME280(t_mode=BME280_OSAMPLE_8, p_mode=BME280_OSAMPLE_8, h_mode=BME280_OSAMPLE_8)
  File "/home/pi/Adafruit_Python_BME280/Adafruit_BME280.py", line 136, in __init__
    self._load_calibration()
  File "/home/pi/Adafruit_Python_BME280/Adafruit_BME280.py", line 147, in _load_calibration
    self.dig_T1 = self._device.readU16LE(BME280_REGISTER_DIG_T1)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 185, in readU16LE
    return self.readU16(register, little_endian=True)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 164, in readU16
    result = self._bus.read_word_data(self._address,register) & 0xFFFF
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 173, in read_word_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
IOError: [Errno 5] Input/output error```

viktorarmani55, 2017-09-07:
```pi@raspberrypi:~ $ sudo ~/Mycodo/env/bin/pip install -e git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
Obtaining adafruit-bme280 from git+git://github.com/adafruit/Adafruit_Python_BME280.git#egg=adafruit-bme280
  Updating ./Mycodo/env/src/adafruit-bme280 clone
Installing collected packages: adafruit-bme280
  Found existing installation: Adafruit-BME280 1.0
    Uninstalling Adafruit-BME280-1.0:
      Successfully uninstalled Adafruit-BME280-1.0
  Running setup.py develop for adafruit-bme280
Successfully installed adafruit-bme280
pi@raspberrypi:~ $ git clone https://github.com/adafruit/Adafruit_Python_BME280
fatal: destination path 'Adafruit_Python_BME280' already exists and is not an empty directory.
pi@raspberrypi:~ $ cd Adafruit_Python_BME280
pi@raspberrypi:~/Adafruit_Python_BME280 $ ~/Mycodo/env/bin/python Adafruit_BME280_Example.py
Traceback (most recent call last):
  File "Adafruit_BME280_Example.py", line 3, in <module>
    sensor = BME280(t_mode=BME280_OSAMPLE_8, p_mode=BME280_OSAMPLE_8, h_mode=BME280_OSAMPLE_8)
  File "/home/pi/Adafruit_Python_BME280/Adafruit_BME280.py", line 136, in __init__
    self._load_calibration()
  File "/home/pi/Adafruit_Python_BME280/Adafruit_BME280.py", line 147, in _load_calibration
    self.dig_T1 = self._device.readU16LE(BME280_REGISTER_DIG_T1)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 185, in readU16LE
    return self.readU16(register, little_endian=True)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 164, in readU16
    result = self._bus.read_word_data(self._address,register) & 0xFFFF
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 173, in read_word_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
IOError: [Errno 5] Input/output error
```

kizniche, 2017-09-07:
It doesn't appear to be able to communicate with your sensor.

Is your sensor recognized?

```
sudo i2cdetect -y 1
```

Have not, have you enabled I2C with ```raspi-config```?

viktorarmani55, 2017-09-07:
it seems to be detecting the multiplexer but not the sensor for some reason.
```pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --   '''

kizniche, 2017-09-07:
Do you have the sensor connected directly to the Pi?

viktorarmani55, 2017-09-07:
I connected the sensor directly to the pi and it seems to have detected it. 
```pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- 77 ```

viktorarmani55, 2017-09-07:
In the first image, i had the sensor connected to the multiplexer, which was connected to the pi.
The second image, the sensor is connected directly to the pi i.e. no multiplexer.

kizniche, 2017-09-07:
Does ```~/Mycodo/env/bin/python Adafruit_BME280_Example.py``` return measurements now?

viktorarmani55, 2017-09-07:
yes it does.
```pi@raspberrypi:~/Adafruit_Python_BME280 $ ~/Mycodo/env/bin/python Adafruit_BME280_Example.py
Temp      = 21.549 deg C
Pressure  = 1016.25 hPa
Humidity  = 34.53 %
```

viktorarmani55, 2017-09-07:
its actually quite accurate. It is registering almost the same value as my calibrated Vaisala sensors.

kizniche, 2017-09-07:
On the ```Input``` page of the Mycodo web interface, enable one BME280 sensor with the multiplexer disabled and see if you have measurements on the ```Data -> Live Measurements``` page.

viktorarmani55, 2017-09-07:
Wasnt able to login.  Waiting for 10 minutes....lol

kizniche, 2017-09-07:
You should be able to reset the counter by the following command

```
sudo touch ~/Mycodo/mycodo_flask.wsgi
```

viktorarmani55, 2017-09-07:
im unable to login for some reason.

kizniche, 2017-09-07:
You can delete the database and start over

```
sudo service mycodo stop
sudo rm ~/Mycodo/database/mycodo.db
sudo touch ~/Mycodo/mycodo_flask.wsgi
wget --quiet --no-check-certificate -p http://localhost/ -O /dev/null && sudo service mycodo start
```

You should then be prompted to create a new admin user/password when you visit the web interface.

viktorarmani55, 2017-09-07:
```pi@raspberrypi:~ $ sudo rm ~/Mycodo/database/mycodo.db
rm: cannot remove ‘/home/pi/Mycodo/database/mycodo.db’: No such file or directory
```

kizniche, 2017-09-07:
Typo. should be ```databases```

kizniche, 2017-09-07:
I have to go for now, but will respond when I can.

viktorarmani55, 2017-09-07:
it works. numbers are good
BME280 Sensor (1)
BME280 Sensor, 2.0 second interval
Measurement | Timestamp
22.98 C (73.36 °F) | 2017/9/7 15:12:42
32.09 % | 2017/9/7 15:12:42
5.43 C (41.77 °F) DP | 2017/9/7 15:12:42
101636 Pa | 2017/9/7 15:12:42
-25.91 m | 2017/9/7 15:12:42

kizniche, 2017-09-07:
Good. Let's try with just one multiplexer connected and one sensor connected to the multiplexer. Make sure the multiplexer I2C address and channel are correct before activating the sensor.

viktorarmani55, 2017-09-07:
ok. ill do that now.

viktorarmani55, 2017-09-07:
it doesn't seem to be responding.  It has stopped at time stamp 7:04:40 (AUS time).  Checked live graph, and it too was not responding.

viktorarmani55, 2017-09-07:
I tried ```sudo i2cdetect -y 1``` and it only seems to be detecting the multiplexer

kizniche, 2017-09-08:
Restart the daemon in debug mode and see what appears in the daemon log (Gear Icon -> Mycodo Logs)

```
sudo service mycodo stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
```

kizniche, 2017-09-08:
It is normal that only the multiplexer shows up in an address scan, as only the multiplexer is connected. How the multiplexer works is the channel is selected and when the address of the sensor is communicated with, the multiplexer handles the exchange. This is why multiple sensors with the same address can be used. Each time the sensor needs to be read, the channel is changed before communication.

Debug mode will show more log lines that have to do with switching the multiplexer channel and sensor communication.

viktorarmani55, 2017-09-08:
this is what i got ```2017-09-08 13:38:13,153 - mycodo.output - DEBUG - Initializing Outputs
2017-09-08 13:38:13,244 - mycodo.output - DEBUG - Outputs Initialized
2017-09-08 13:38:13,246 - mycodo.daemon - DEBUG - Starting all activated timer controllers
2017-09-08 13:38:13,246 - mycodo.output - INFO - Output controller activated in 102.6 ms
2017-09-08 13:38:13,247 - mycodo.daemon - INFO - All activated timer controllers started
2017-09-08 13:38:13,248 - mycodo.daemon - DEBUG - Starting all activated sensor controllers
2017-09-08 13:38:13,496 - mycodo.daemon - ERROR - Could not activate Sensor controller with ID 1: [Errno 5] Input/output error
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 452, in controller_activate
    ready, cont_id)
  File "/home/pi/Mycodo/mycodo/controller_sensor.py", line 287, in __init__
    self.i2c_bus)
  File "/home/pi/Mycodo/mycodo/sensors/bme280.py", line 91, in __init__
    self._load_calibration()
  File "/home/pi/Mycodo/mycodo/sensors/bme280.py", line 197, in _load_calibration
    self.dig_T1 = self._device.readU16LE(BME280_REGISTER_DIG_T1)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 185, in readU16LE
    return self.readU16(register, little_endian=True)
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 164, in readU16
    result = self._bus.read_word_data(self._address,register) & 0xFFFF
  File "/home/pi/Mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 173, in read_word_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
IOError: [Errno 5] Input/output error
2017-09-08 13:38:13,506 - mycodo.daemon - INFO - All activated sensor controllers started
2017-09-08 13:38:13,507 - mycodo.daemon - DEBUG - Starting all activated PID controllers
2017-09-08 13:38:13,508 - mycodo.daemon - INFO - All activated PID controllers started
2017-09-08 13:38:13,508 - mycodo.daemon - DEBUG - Starting all activated LCD controllers
2017-09-08 13:38:13,508 - mycodo.daemon - INFO - All activated LCD controllers started
2017-09-08 13:38:13,509 - mycodo.daemon - INFO - Mycodo daemon v5.1.8 started in 2.164 seconds
2017-09-08 13:38:13,518 - mycodo.daemon - INFO - 56.208 MB ram in use```

kizniche, 2017-09-08:
I just pushed a commit that may fix the issue I discovered from the above error, and released v5.1.9. Go ahead and upgrade from the upgrade menu and see if it fixes things. Make sure to stop the daemon from running in debug mode before upgrading:

```
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_client.py -t
```

viktorarmani55, 2017-09-08:
no luck. should i do a complete restart.
```Appending to crontab:
-----------------------------------------------
@reboot /var/www/mycodo/mycodo/scripts/restart_daemon.py --continuouscheck &
-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove

#### Compiling mycodo_wrapper

#### Creating users and directories
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
The user `mycodo' is already a member of `dialout'.

#### Setting permissions

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service.

#### Creating Mycodo database if it doesn't exist
#### Successfully connected to http://localhost

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Done.

Fri Sep  8 14:30:18 AEST 2017
Upgrade completed successfully without errors.```

kizniche, 2017-09-08:
What do you mean by 'no luck'? Can you copy the pertinent daemon log excerpts after the upgrade?

viktorarmani55, 2017-09-08:
```Appending to crontab:
-----------------------------------------------
@reboot /var/www/mycodo/mycodo/scripts/restart_daemon.py --continuouscheck &
-----------------------------------------------
To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove

#### Compiling mycodo_wrapper

#### Creating users and directories
useradd: user 'mycodo' already exists
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `adm'.
The user `mycodo' is already a member of `video'.
The user `mycodo' is already a member of `dialout'.

#### Setting permissions

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service.

#### Creating Mycodo database if it doesn't exist
#### Successfully connected to http://localhost

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Done.

Fri Sep  8 14:30:18 AEST 2017
Upgrade completed successfully without errors.```

viktorarmani55, 2017-09-08:
```2017-09-08 14:39:38,965 - mycodo.daemon - INFO - 49.00 MB RAM in use
2017-09-08 14:43:39,287 - mycodo.daemon - WARNING - Sensor controller with ID 1 not found
2017-09-08 14:48:00,799 - mycodo.stats - ERROR - Could not send anonymous usage statistics: Connection timed out (expected if there's no internet or the server is down)
2017-09-08 15:10:30,712 - mycodo.sensor_1 - ERROR - SensorController raised an exception: Timeout waiting to acquire lock for /var/lock/mycodo_multiplexer_0x70.pid
2017-09-08 15:10:30,713 - mycodo.sensor_1 - ERROR - [Locking bus-1 0x70] Waited 60 seconds. Breaking lock to acquire /var/lock/mycodo_multiplexer_0x70.pid
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_sensor.py", line 799, in setup_lock
    self.lock[lockfile].acquire(timeout=60)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/lockfile/linklockfile.py", line 46, in acquire
    self.path)
LockTimeout: Timeout waiting to acquire lock for /var/lock/mycodo_multiplexer_0x70.pid
2017-09-08 15:10:30,719 - mycodo.daemon - ERROR - Could not activate Sensor controller with ID 1: [Errno 5] Input/output error
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 452, in controller_activate
    ready, cont_id)
  File "/var/www/mycodo/mycodo/controller_sensor.py", line 289, in __init__
    self.i2c_bus)
  File "/var/www/mycodo/mycodo/sensors/bme280.py", line 30, in __init__
    self.sensor = BME280(address=address, busnum=bus)
  File "/home/pi/Mycodo/env/src/adafruit-bme280/Adafruit_BME280.py", line 136, in __init__
    self._load_calibration()
  File "/home/pi/Mycodo/env/src/adafruit-bme280/Adafruit_BME280.py", line 147, in _load_calibration
    self.dig_T1 = self._device.readU16LE(BME280_REGISTER_DIG_T1)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 185, in readU16LE
    return self.readU16(register, little_endian=True)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 164, in readU16
    result = self._bus.read_word_data(self._address,register) & 0xFFFF
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 173, in read_word_data
    ioctl(self._device.fileno(), I2C_RDWR, request)
IOError: [Errno 5] Input/output error```

kizniche, 2017-09-08:
I changed the BME280 module code, so attempt to read with just the sensor connected by itself, before enabling the multiplexer, as done before.

kizniche, 2017-09-08:
It also doesn't appear that your multiplexers are powered in this image.
[image]

kizniche, 2017-09-08:
Actually, I see the lower multiplexer is powered by the top rails. Are you sure the power rails are connected down the entire breadboard. Some breadboards have a break in the middle (between your multiplexers) that allow different power sources to be used on each half of the breadboard. If this is the case, your lower half of the breadboard is not powered and the sensor you have connected to the multiplexer is not powered. I would use a multimeter to confirm you are powering both your sensor and multiplexer.

kizniche, 2017-09-08:
Also, can you provide some more log lines before this line you last sent:

```
2017-09-08 14:43:39,287 - mycodo.daemon - WARNING - Sensor controller with ID 1 not found
```

Ideally, I would like to see from when the daemon starts (or sensor controller is activated) until the end of the error. The line above indicates an error occurred before this line (the sensor controller can't be found because there was an error starting it).

viktorarmani55, 2017-09-10:
Good Morning Kyle.  The sensors are working now, some of the terminals didn't have good connections.  Sorry for the hassle.  I'm testing out the remainder of my sensors, the temp, humidity and dewpoints seems to be ok, however, some of the pressures and elevation are slightly off.  As I wont be using them, I'm not that concerned.  Thanks for all your help.

kizniche, 2017-09-11:
Glad to hear. Thanks for the bug report. Even though it was a hardware issue, we still got some bugs fixed related to using the multiplexer with certain sensors (previously I never considered the initialization of a sensor module to require changing the multiplexer channel. The BME280 is one of a few sensors that needed that to work properly).

What are you using the sensors for, if you don't mind me asking?

closed 2017-09-11 by kizniche

viktorarmani55, 2017-09-12:
Hi Kyle,

I'm testing out a Greenwall module, looking at lux, moisture, temperature,
humidity, CO2, water consumption etc, for a variety of plant species, and
the potential to be installed within buildings.

Regards,
Vik