kizniche/Mycodo #291: Error when trying to add a relay

awdaniel, 2017-09-12:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.1.9
 
#### Problem Description
Just installed mycodo on the orignal pi model A, added a sensor fine but when I try to add a relay I get the error below.
 
### Errors
Error 500: Internal Server Error

Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/usr/local/lib/python2.7/dist-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 733, in page_output
    flaskutils.relay_add(form_add_relay)
  File "/var/www/mycodo/mycodo/flaskutils.py", line 1753, in relay_add
    manipulate_relay('Add', new_relay.id)
  File "/var/www/mycodo/mycodo/flaskutils.py", line 630, in manipulate_relay
    if len(return_values) > 1:
TypeError: object of type 'NoneType' has no len()

### Steps to Reproduce the issue:
How can this issue be reproduced?
Just installed Mycodo on the orignal pi model A, using the method described in the documentation. I then added a sensor fine but when I try to add a relay I get the above error.
 
### Additional Notes
Looks great the software so thanks :)

awdaniel, 2017-09-12:
Ok so the relay is appearing, so it seemed it did work but I got the error when I tried to add it.

awdaniel, 2017-09-12:
When I try to modify the Relay settings another error message is appearing at the top of the screen in a brow/orange box: "Error: Modify Relay: object of type 'NoneType' has no len()"

kizniche, 2017-09-13:
Sounds like your daemon is not running. Click the mycodo version number in the top left corner to see the daemon status.

awdaniel, 2017-09-13:
Ok so its working now! Thanks. Before the relay would always stay inactive now it activates and turns green! :)

kizniche, 2017-09-14:
Good to hear. The daemon actually controls the relays, so if it's not operating properly, usually nothing will work properly. The web UI merely requests the daemon perform actions.

closed 2017-09-14 by kizniche