kizniche/Mycodo #292: Can't get DS18B20 to work with Mycodo

Model76, 2017-09-22:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.2.0
 
#### Problem Description
I am trying to connect a DS18B20 sensor to Mycodo, but unfortunately get an error. I have read the instructions, but can't get it to work. This may very well be a user error. However, I was able to get an ADS1115 working with Mycodo without any trouble.

The sensor I have is one of those water sealed ones with 3 wires - red, black, and yellow. It looks exactly like this:
http://www.flytron.com/936-thickbox_default/waterproof-ds18b20-digital-temperature-sensor-.jpg

I have tried a bunch of different things, such as 2 different DS18B20 sensors, 3.3v and 5v on the red wire, and several different GPIO pins for the yellow wire.

Right now I have the red wire connected to 5v on pin 4, yellow to GPIO18 on pin 12, and black to ground on pin 14. I also have a 4.7K Ohm resistor between the yellow and red wires.

For the pin numbering and function naming, I am going by this chart:
https://cdn.sparkfun.com/assets/learn_tutorials/4/2/4/header_pinout.jpg

### Errors
 See "Steps to reproduce..."

### Steps to Reproduce the issue:
 1. Connect the the sensor as described above. You can restart the system or not, it doesn't make a difference.
 2. Go to the web interface for Mycodo and and click the "Input" tab.
 3. Select "1" in the fist box, and "Temperature: DS18B20" in the second box, then click "Add Device". For me, this results in:
 `
    Success: DS18B20 Sensor with ID 1 (e04ec2bf-af44-4f4e-ade5-16bae25920d3) successfully added
    Success: Add Sensor
`
 4. Click "Activate". This results in the following error: 
`Error: Cannot activate sensor without the GPIO/I2C Address/Port to communicate with it set.`
I am not sure what that message means. Was I supposed to set a GPIO/I2C Address/Port somewhere?

If I click the plus sign next to the sensor, the "Device ID" drop down field only shows "None Detected".

I hope you can help, and thank you for making Mycodo!

kizniche, 2017-09-22:
In the Mycodo manual I link to this guide (below) to setting up 1-wire support and connecting the DS18B20 to the correct pin. 1-wire must be enabled in ```raspiconfig```. Once 1-wire support is enabled (and a reboot) and the sensor is connected to the correct 1-wire pin, Mycodo will detect it (or them) and allow you to select it in the drop down box.

https://learn.adafruit.com/adafruits-raspberry-pi-lesson-11-ds18b20-temperature-sensing

closed 2017-10-12 by kizniche

snickers2k, 2022-04-22:
for everyone else reading this, i got weird results by using the sparkfun waterproof-ds18b20 with ow-shell (jumping around from 20°C to 90°C).
while using the w1thermsensor the result is just fine.