kizniche/Mycodo #293: TSL2591 Crashing Daemon?

Jhutterer, 2017-09-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.2.0
 
#### Problem Description
Please list: TSL2591 after about 12-24 hours seems to crash Daemon. 

  - what were you trying to do
  -Trying to run the TSL2591 and BMP180 on the same bus.
I have also tested this with only the TSL2591 on its own and it still crashes the daemon eventually.
 
### Errors
  - List any errors you encountered.
2017-09-25 23:52:22,182 - mycodo.sensors.tsl2591_1_41 - ERROR - TSL2591Sensor raised an exception when taking a reading: 24
2017-09-25 23:52:27,593 - mycodo.sensors.bmp180 - ERROR - BMP180Sensor raised an exception when taking a reading: [Errno 24] Too many open files: '/dev/i2c-1'
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/bmp180.py", line 99, in read
  File "/var/www/mycodo/mycodo/sensors/bmp180.py", line 88, in get_measurement
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_BMP/BMP085.py", line 67, in __init__
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 64, in get_i2c_device
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_GPIO/I2C.py", line 97, in __init__
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 97, in __init__
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/Adafruit_PureIO/smbus.py", line 122, in open
IOError: [Errno 24] Too many open files: '/dev/i2c-1'
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Run the TSL2591 sensor more than 24 hours.
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-09-26:
Thanks for the bug report. Can you please answer a few questions so I know a little more about your system?

1. What version Pi is it?
2. Is the operating system Raspbian?
3. If Raspbian, was Mycodo installed after a fresh install of the operating system?
4. Are there any other modifications to the operating system not outlined in the Mycodo install procedures?

kizniche, 2017-09-26:
Can you test the following code (two different test files). Make sure the Mycodo daemon isn't running (```sudo service mycodo stop```) for these tests to be accurate. Test each by itself, that is, test the first and see if it crashes after a few days, then stop it and test the second by itself for a few days.

Run these with the command:

```sudo ~/Mycodo/env/bin/python ./test_tsl2591_01.py```

test_tsl2591_01.py:

```python
# coding=utf-8
import traceback
import tsl2591

tsl = tsl2591.Tsl2591()
while True:
    try:
        full, ir = tsl.get_full_luminosity()
        lux = tsl.calculate_lux(full, ir)
        print("Lux: {}, Full: {}, IR: {}".format(lux, full, ir))
        time.sleep(10)
    except Exception:
        print(traceback.format_exc())
```

test_tsl2591_02.py:

```python
# coding=utf-8
import traceback
import tsl2591

while True:
    try:
        tsl = tsl2591.Tsl2591()
        full, ir = tsl.get_full_luminosity()
        lux = tsl.calculate_lux(full, ir)
        print("Lux: {}, Full: {}, IR: {}".format(lux, full, ir))
        time.sleep(10)
    except Exception:
        print(traceback.format_exc())
```

Jhutterer, 2017-09-27:
Appreciate your quick response!

Question 1 : Raspberry PI 2 Model B
Question 2 : yes
Question 3 : yes
Question 4 : no

Must include `import time ` for the scripts to run. 
I have started testing script one I will let you know results.

kizniche, 2017-09-27:
We can speed up the measurements to induce the issue sooner. I've changed the code to reflect a 2 second delay between reads instead of 10 seconds.

Jhutterer, 2017-10-01:
Ok, trying to the new delay.

kizniche, 2017-10-18:
How has testing been going? Any errors yet? If no errors are happening, this is odd. We can start testing our test code within the Mycodo sensor controller to have a more real-world testing scenario. Let me know any progress and if we should take that route.

kizniche, 2017-10-24:
I just committed a fix for the TSL2591 sensor that patches a memory leak (#254). I haven't released this yet, but it should be live to update to in a few days. I'm going to mark this closed. If the issue does appear again, after the upgrade to the new release, feel free to open this issue again.

closed 2017-10-24 by kizniche