kizniche/Mycodo #294: Chirp Sensor

viktorarmani55, 2017-10-06:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.1.9
 
#### Problem Description
Please list:

  - I'm trying to connect the TSL2561 & Chirp sensor to my existing muliplexer.
  - I've removed two BME280 sensors, and connected two other sensors i.e. TSL2561 & Chirp sensor.
 
### Errors
  - The TSL2561 seems to be operating fine, however, the Chirp sensor isn't recognized. I noticed the I2C address is 0x20 (default).  Would I need to change this to 0x77?
On another matter, if I was to connect additional multiplexers for more chirp and lux sensors, what would be the best approach?  
List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Input & Connect the TSL2561 to an existing BME280 sensor slot on multiplexer
 2. Input & Connect the Chirp sensor to an existing BME280 sensor slot on multiplexer
 3. Leave settings as default
 4. Run the live measurements, and only the TSL2561 seems to respond
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?  If I was to connect additional multiplexers for more chirp and lux sensors, what would be the best approach?

kizniche, 2017-10-12:
Can you get measurement from the Chirp sensor when you connect it directly to the I2C lines and use the manufacturer's code?

kizniche, 2017-10-12:
Are there any errors in the daemon log?

kizniche, 2017-10-26:
If you have any new information or answers to the above questions, feel free to reopen this issue.

closed 2017-10-26 by kizniche