kizniche/Mycodo #295: influxdb ramdomly failed to get measurement

reymart081678, 2017-10-10:
## Mycodo Issue Report:
  - Specific Mycodo Version:5.2.1
 
#### Problem Description
Please list:
Mycodo does not response to any connection through web and hard reset resolves it, Based on the logs the error is shown below and not sure if this is the cause.

Just want to know if there is a way mycodo can reestablish the connection if it sees this error.

2017-10-05 23:03:24,143 - mycodo.daemon - INFO - Mycodo daemon v5.2.1 starting
2017-10-05 23:03:24,592 - mycodo.daemon - INFO - Anonymous statistics enabled
2017-10-05 23:03:24,653 - mycodo.daemon - INFO - Starting rpyc server
2017-10-05 23:03:25,035 - mycodo.output - INFO - Relay 1 setup on pin 16 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,036 - mycodo.output - INFO - Relay 2 setup on pin 21 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,037 - mycodo.output - INFO - Relay 3 setup on pin 25 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,038 - mycodo.output - INFO - Relay 4 setup on pin 12 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,039 - mycodo.output - INFO - Relay 5 setup on pin 20 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,041 - mycodo.output - INFO - Relay 7 setup on pin 5 and turned OFF (OFF=LOW)
2017-10-05 23:03:25,044 - mycodo.output - INFO - Output controller activated in 106.6 ms
2017-10-05 23:03:25,162 - mycodo.timer_1 - INFO - Activated in 82.5 ms
2017-10-05 23:03:25,426 - mycodo.timer_2 - INFO - Activated in 220.4 ms
2017-10-05 23:03:25,553 - mycodo.timer_3 - INFO - Activated in 81.5 ms
2017-10-05 23:03:25,674 - mycodo.timer_4 - INFO - Activated in 81.7 ms
2017-10-05 23:03:25,824 - mycodo.timer_5 - INFO - Activated in 113.9 ms
2017-10-05 23:03:25,828 - mycodo.daemon - INFO - All activated timer controllers started
2017-10-05 23:03:26,136 - mycodo.sensor_7 - INFO - Activated in 244.6 ms
2017-10-05 23:03:26,138 - mycodo.daemon - INFO - All activated sensor controllers started
2017-10-05 23:03:26,450 - mycodo.pid_1 - INFO - Activated in 199.0 ms
2017-10-05 23:03:44,367 - mycodo.pid_2 - INFO - Activated in 172.8 ms
2017-10-05 23:03:44,369 - mycodo.daemon - INFO - All activated PID controllers started
2017-10-05 23:03:44,369 - mycodo.daemon - INFO - All activated LCD controllers started
2017-10-05 23:03:44,370 - mycodo.daemon - INFO - Mycodo daemon v5.2.1 started in 20.223 seconds
2017-10-05 23:03:44,378 - mycodo.daemon - INFO - 43.88 MB RAM in use
2017-10-05 23:03:56,589 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Mycodo to reestablish connection to influxdb

kizniche, 2017-10-12:
Can you explain the issue more?

What does "Mycodo does not response to any connection through web" mean?

I only see one error. If there's only one error, there probably isn't an issue.

reymart081678, 2017-10-13:
Mycodo web interface does not show up " too long to respond" , tried ssh "connection refuse" and hard reset is the only way bring it up again..

kizniche, 2017-10-13:
> Just want to know if there is a way mycodo can reestablish the connection if it sees this error
> 2017-10-05 23:03:56,589 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect

If this error appears only once, there isn't a major issue. This isn't a blocking or crashing error, so if you only experience this error once (if the Pi just started and is starting the services, for instance), then the future queries were successful.

> Mycodo web interface does not show up " too long to respond" , tried ssh "connection refuse"

This sounds like an issue with your network connection, not Mycodo. If there are sensor measurements being acquired during this period you cannot connect, Mycodo is still working, you are just unable to connect.

kizniche, 2017-10-13:
Because you cannot ssh to your Pi, and since Mycodo is not associated with ssh at all, this is a strong indication that your issue is a networking issue and not an issue related to Mycodo.

closed 2017-10-24 by kizniche