kizniche/Mycodo #296: Adding Async Graphs Do Nothing

pukkita, 2017-10-13:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.2.4 (2017-10-05)

 
#### Problem Description

Adding Live Graphs works fine, added a DHT22 and RPi sensors.

When trying to add an Async Graph however, nothing happens.

### Errors
- Nothing on any logs. Neither mycodo's, nor apache's.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Install Raspberry (NOOBS Lite 2.4) and apt-update, upgrade etc.
 2. Follow Mycodo's install procedure
 3. Add sensors
 4. Try to add Async Graphs.

Tried Firefox, Safari and Chrome. Spanish Locale.

kizniche, 2017-10-13:
Can you please post a screenshot of this "add async graph" option?

pukkita, 2017-10-13:
Sure:

[image]

kizniche, 2017-10-13:
Thanks. I'll look into it when I can access my development Pi later today.

pukkita, 2017-10-13:
Impressive job BTW. Loved your blog, and of course mycodo, was going to do it manually myself, but did a search anb when I found your Mycodo project felt like Santa's paid me a visit!! thanks for sharing this project!

kizniche, 2017-10-14:
Thanks. It's been a fun project. I wish I could put as much time into it as I used to. I do enjoy hearing that other people find it useful.

This issue should now be fixed in the latest release (v5.2.5). Let me know if the issue persists after you upgrade.

closed 2017-10-14 by kizniche