kizniche/Mycodo #297: Add ADC Option to Inverse Scale

kizniche, 2017-10-13:
Currently there can be a scale associated with analog-to-digital converters, so that a meaningful value can be reported based on a voltage, For instance, a photodiode will increase the voltage when more light hits the sensor, so if the voltage range is 0 - 2 volts, a scale of 0 - 100% light can be associated with these voltage values, which will subsequently be stored in the measurement database and can be used with conditional statements or PID controllers.

The issue comes if the voltage is inversely related to the actual measured value. For the light example, the % increases with increasing voltage. For some sensors, such [this DF Robot CO2 Sensor](https://www.dfrobot.com/wiki/index.php/CO2_Sensor_SKU:SEN0159), voltage decreases as the CO2 concentration increases. Because of this, and how Mycodo was only designed to use positively-correlated voltage-measurement value ranges, it will not be able to properly report the correct condition measurements.

Therefore, an option to inverse the measurement scale may need to be implemented to account for this issue.

Theoi-Meteoroi, 2017-10-23:
This would be a good feature, however the NDIR sensors for CO2 are better. The only problem is just one Pi serial port.  I looked at the MG-811 sensor early on and found it lacking for instrument grade measurement. Just how do you calibrate the thing?   I have i2c NDIR CO2, however they use an address outside what a Pi can address, AFAIK. I use them with micro controllers (arduino).

kizniche, 2017-10-23:
I've never used that particular CO2 sensor I mention in the first post (another user requested I add support). The K-30 is the only one I have. Since the user's sensor only requires the use of an ADC, it would seem easy to add an inverse scale option to allow it to be used.

As for serial ports, I know of at least one [serial port expander](https://www.atlas-scientific.com/product_pages/components/port_expander.html) that should allow multiple sensors like the K-30 to be connected. Though, I have never used this, but since Mycodo has the ability to select the /dev for the UART, it should work with Mycodo, once the proper devices on the system are in place and working.

closed 2017-10-24 by kizniche