kizniche/Mycodo #300: Compatible LCDs?

pukkita, 2017-10-21:
According to the manual there should be compatibility notes on the README, but couldn't find those...

Will a I2C 20x4 like https://www.amazon.es/gp/product/B0723HMWHQ/ref=ox_sc_act_title_1?smid=A3EMO9TEUN7TWI&psc=1 work with mycodo?

kizniche, 2017-10-21:
That should work. I'll update the Readme/Manual today and double check.

Theoi-Meteoroi, 2017-10-23:
I have a 16x2 similar to the photo you added.  Similar I2C backpack (PCF8574 based) Since LCDs use +5v, I used the SDL 9545a mux to perform the voltage buffering.  The Pi I2C would be damaged otherwise. The GUI interface allows the 9548 mux (I had trouble with that-probably my kit, because the SDL 9545a is working fine with the code on another hardware build ) but doesn't (yet) allow selecting a root bus other than bus 1.

The only change I had to make was to hard-code the bus to '3' instead of '1' in  ~Mycodo/mycodo/controller_lcd.py

`  157             # Setup I2C bus
    158             try:
    159                 if GPIO.RPI_REVISION == 2 or GPIO.RPI_REVISION == 3:
    160                     i2c_bus_number = 3    

This is working fine, displaying the TSL2591 lux reading.  The timestamp selection isn't working ( ERROR: NO DATA ) but that is something else.

If detected properly, you get a 'Mycodo <version>' banner and the state of the LCD driver code ( STOP LCD or START LCD ) before the first readings appear.  It takes one cycle time (30 sec default for LCD) to get a refresh display of the readings selected.

I have some 20x4 LCD around here somewhere and more backpack adapters.  I'll give it a go.

kizniche, 2017-10-23:
@pukkita, I added an image of the ones I use to the [LCD Displays](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#lcd-displays) section of the manual. I haven't tested any others.

> doesn't (yet) allow selecting a root bus other than bus 1

I just pushed a commit that adds the ability to select the LCD I2C bus.

> The timestamp selection isn't working ( ERROR: NO DATA )

I wasn't able to replicate this issue. My LCDs display sensor timestamps fine.

Theoi-Meteoroi, 2017-10-23:
Copy that. Changed to the Rpi temp sensor timestamp. That one works fine. Swapped positions and that sensor reports on LCD, just fine. Must be related to the TSL2591 timestamp (readings are working, so I ain't a complaining).  

And thanks for adding the i2c bus selection! You are awesomer than some other developers I have worked with. And that be many.  I'm a system integrator that works various protocol stacks with a hardware emphasis for a living.  I hope I can return value to your project by testing, feedback and some contributions.  I've built two Mycodo platforms so far and I think I'll build several more so I can always have a development/test platform.  The first couple already have jobs to do.

I can see that part of your burden is having an adequate test setup with all the supported options, and testing them in combinations that could reproduce reported issues.   And then there is the issue of some components entering last purchase phase (SHT1x and SHT7x) and others (DHT series and AM2315) that have quirks in characteristics.  I think that somehow, some devices make the project look worse than what it really is.   The code is really pretty robust if you are careful about hardware design.  Let me know if I can be helpful in some way.

kizniche, 2017-10-23:
> Must be related to the TSL2591 timestamp

That's strange, because it should work the same for all sensors, regardless of the actual code in the sensor module, as it's pulling the timestamp from the last measurement from the influxdb round-robin database.  If it can pull a measurement (which it appears it can), it should be able to display the timestamp associated with it.

> And thanks for adding the i2c bus selection! You are awesomer than some other developers I have worked with.

Thanks. I like adding features to make it useful to the community. It also didn't hurt that it was a relatively easy addition to code. There are a lot of features I'd like to add, but I don't have as much time as I used to to work on them.

> I can see that part of your burden is having an adequate test setup with all the supported options

Yes, this is sometimes an issue. @etiology was kind enough to set up a testing environment but it wasn't comprehensive. I'm still struggling to learn the ways of testing, but it's slow going. There were some great [proposals](https://github.com/kizniche/Mycodo/issues/128#issuecomment-336615771) from him about fixing the [issues that arose](https://github.com/kizniche/Mycodo/issues/128#issuecomment-336610842) since it was initially introduced.

> components entering last purchase phase

For the most part I haven't been paying to much attention to this. Sensors are now fairly easy to add, so if even one user has an old sensor lying around, I'm inclined to add it. Now that there's a Linux Command option as an input, anyone can create a script that queries their sensor and use that as a sensor input. Though this new feature hasn't been thoroughly tested (it hasn't even been released yet).

> and others (DHT series and AM2315) that have quirks in characteristics. I think that somehow, some devices make the project look worse than what it really is.

Unfortunately a lot of issues come up because of bad circuit design. There's only so much that can be done with software, the rest is in the hands of the user. This is why I want to expand the manual with a lot more circuit diagrams.

> Let me know if I can be helpful in some way

I'm not that good with delegating software tasks, even more so since you would be doing it all for free. If you see something you would/could help with, don't hesitate to ask or jump in and make a pull request. I appreciate the help.

kizniche, 2017-10-23:
I forgot to ask the most important question regarding your LCDs not displaying a timestamp: Are there any related errors in the daemon log?

kizniche, 2017-10-24:
I'm going to close this issue, but feel free to reply. If your failed LCD timestamp issue persists, please open a new issue with details.

closed 2017-10-24 by kizniche