kizniche/Mycodo #301: PWM Drive DC motor (FAN) module

pukkita, 2017-10-21:
I would like to control DC (computer) FANs by interfacing with the typical H bridge DC module based on HG7881 to drive either two DC motors or a stepper.

Fan max draw is 0.25A, so should be well inside limits (theoretical max draw per motor = 800mA).

Could this be used to interface the FAN to a PWM output?

kizniche, 2017-10-24:
I don't have the chip to test, but theoretically it should work. Currently PWM output is only supported by the PID controllers. I'm going to expand the use to input/output conditionals and timers soon.

pukkita, 2017-10-24:
thanks for the reply!

Have yet to learn all the PID stuff and conditionals, I understand when you add such feature PIDs could be inter-related depending on conditionals?

So far have had zero success with this module and PC Fans. I used a setup really similar to http://diyprojects.eu/how-to-use-h-bridge-hg7881-with-external-power-supply-and-arduino/, though I joined both grounds without a resistor.

Haven't played too much with it however, as it acts odd: Once everything is wired up, boot the raspberry, (no software or fan running), after some minutes, the module led starts flashing and the module heats up a lot.

 PC Fans have additionaly circuitry dunno if it has something to do. 

On that same page I linked previously, user Marius wrote:

> Hello.
> I’ve got into a problem using l9110. I’ve expected to use this driver for 12volt motorised ball valves, but came into a problem: when the voltage is above 5,5V , control input shorted to ground and no load connected, l9110 start to use about 70mA current per channel without any reason. Chip heats up, but doesnt fail. With lower voltages everything is fine, the board consumes 20mA, no matter logic input is high or low. All 5 drivers I’ve bought behave in the same way. Is there a faulty chips or a normal behaviour of that chip?

pukkita, 2017-10-25:
Update: managed to make it work manually, is funny the fan "sings" depending on the PWM, but decided to pass on these ones, maybe if a solenoid valve requires it, or for using stepper motors.

I think these suit much better the purpose (IRF520 MOS Module) http://www.dx.com/p/keyes-mos-driving-module-for-arduino-red-blue-black-297313#.WfDms4Xgy2M for fan control, it supports up to 1A w/o heatsink, and up to 5A with a proper one.

A nice addition would be integrating the reading of the hall sensor into Mycodo, this will mean a second GPIO would be used, but being able to read the RPM reported by the fan would open up possibilities...

AFAIK, according to [this post](https://www.raspberrypi.org/forums/viewtopic.php?f=91&t=92229) 

> Everything I've seen suggests 3-wire PC fans (ground, power, tacho) are open-collector/open-drain tacho outputs. The tacho signal should not be pulled up by the fan itself. The specs say the tacho signal is pulled up to 12V by the motherboard.
> 
> I've just connected one to my Pi. Powering from 5V and ground (showing a bit of caution) with the tacho line direct to a gpio.
> 
> The gpio shows the expected pulse when internally pulled-up to 3V3 and no activity when internally pulled down to ground.

> 
http://abyz.co.uk/videos/fan.webm

This functionality could be added as a PWM Output (FAN w/ hall sensor) output device...

closed 2017-10-26 by kizniche