kizniche/Mycodo #303: Allow PID to use both Relay and PWM Outputs

kizniche, 2017-10-26:
Currently, a single PID can use only Relay Outputs or only PWM Outputs. I would like to allow a PID to be able to use both, for such applications as temperature regulation using an air conditioner actuated by a relay turning on for durations of time and a heater controlled by a PWM signal.

pukkita, 2017-10-26:
That would be awesome!

kizniche, 2017-10-26:
Okay, I had some time after work while waiting to go home that I knocked out what I believe to be all that's needed for PIDs to be able to use any type of output. I'll test when I get home, but be advised if you're pulling from the latest commit form master that it's untested and may not work.

kizniche, 2017-10-27:
There were definitely issues with the latest commit. I'm slowly fixing the issues. Here's one that made me chuckle:

[image]

kizniche, 2017-10-27:
Alright, I think I have it at a working state. I'm going to call it a night.

closed 2017-10-28 by kizniche