kizniche/Mycodo #305: Upgrade process doesn't complete

pukkita, 2017-10-29:
## Mycodo Issue Report:
  - Specific Mycodo Version: 
Happens both from
- 5.3.0 -> 5.3.1
- 5.3.1 -> 5.3.2
 
#### Problem Description
Upgrade process seems to stall once apache2 is re-launched:

[image]
 
### Errors

Tried to restart the apache service via mycodo user:

```
root@mycodo:/etc/polkit-1# su - mycodo
No directory, logging in with HOME=/
$ service apache2 restart
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'apache2.service'.
Multiple identities can be used for authentication:
 1.  ,,, (pi)
 2.  root
```

looks polkit is the reason of the apache2 service restart "stall". 

### Steps to Reproduce the issue:
1. Upgrade from mycodo 5.3.0 to 5.3.1 or 5.3.1 to 5.3.2

kizniche, 2017-10-29:
I noticed this, too, during my upgrades. This had been resolved previously, so I think it has to do with apache2, not Mycodo. I noticed my apache2 was upgraded during the "apt-get upgrade" part of the Mycodo upgrade. I may need to find another way to restart the service, perhaps "/etc/init.d/apache2 stop && /etc/init.d/apache2 start" will work better than "/etc/init.d/apache2 restart"

Also note I don't use "service" to restart apache2, I use the init.d script. Interestingly, while the script restart command was failing repeatedly, I was able to successfully execute the same command as another user using "sudo"

This doesn't make sense because the upgrade script (including apache2 restart command) is executed as root.

kizniche, 2017-10-29:
I also wanted to say that since this is an upgrade and not an install, this issue shouldn't affect Mycodo. If you reboot, everything should be working. Let me know if it's not and we can investigate further before you begin tweaking things to try to fix it.

pukkita, 2017-10-29:
Definitely it looks like apache2/polkit related.

I assume restarting apache the last command in the upgrade process??

On a side note, is mycodo dependent on apache? not very fond of it, specially for a low-power device, would use uWSGI+nginx instead, did you try that in the past?

kizniche, 2017-10-29:
Restarting apache2 is last, but behind restarting the daemon.

What is polkit?

I tried once to get wsgi+nginx to work but was unsuccessful. If you can get it working I will do performance comparisons, and if faster, will switch to nginx.

pukkita, 2017-10-29:
polkit = [Policy Kit](https://wiki.debian.org/PolicyKit)

OK, will tackle that after stabilizing my system.

kizniche, 2017-10-29:
Just a note, because I may not have been clear earlier, the upgrade process (upgrade script) is executed as root, not the mycodo user. Therefor, no commands during that process should have permission issues.

pukkita, 2017-10-29:
I upgraded from the web GUI...

kizniche, 2017-10-29:
That's what I'm referring to. The GUI upgrade process executes the upgrade scripts as root, not the user mycodo.

kizniche, 2017-10-29:
apache2 runs as the user mycodo, as per the wsgi config file, however the upgrade scripts are executed as root, so the init.d apache2 restart command is executed as root.

pukkita, 2017-10-29:
OK.

Don't have a clue why is it called as mycodo, I updated from the GUI.

kizniche, 2017-10-29:
```
kiz@tango:~$ ps auxwww|grep apache                                                                                                   10/29/17  3:58 PM
root      1131  0.0  0.7  14040  6712 ?        Ss   15:39   0:00 /usr/sbin/apache2 -k start
mycodo    1134  5.7  7.7 192936 68160 ?        Sl   15:39   1:07 /usr/sbin/apache2 -k start
www-data  1135  0.0  1.1  16140 10516 ?        S    15:39   0:00 /usr/sbin/apache2 -k start
www-data  1136  0.0  1.1  16140 10532 ?        S    15:39   0:00 /usr/sbin/apache2 -k start
www-data  1592  0.0  1.1  16140 10312 ?        S    15:42   0:00 /usr/sbin/apache2 -k start
www-data  1601  0.0  1.1  16140 10312 ?        S    15:42   0:00 /usr/sbin/apache2 -k start
www-data  1603  0.0  1.1  16140 10356 ?        S    15:42   0:00 /usr/sbin/apache2 -k start
www-data  2860  0.0  1.1  16140 10348 ?        S    15:51   0:00 /usr/sbin/apache2 -k start
www-data  2861  0.0  1.1  16140 10276 ?        S    15:51   0:00 /usr/sbin/apache2 -k start
www-data  3452  0.0  1.1  16140 10336 ?        S    15:53   0:00 /usr/sbin/apache2 -k start
www-data  3453  0.0  1.1  16140 10340 ?        S    15:53   0:00 /usr/sbin/apache2 -k start
www-data  3469  0.0  1.1  16140 10276 ?        S    15:53   0:00 /usr/sbin/apache2 -k start
```

kizniche, 2017-10-29:
Looks normal to me.

kizniche, 2017-10-29:
[Here](https://github.com/kizniche/Mycodo/blob/5a3509ea763a066042d4af7b8735d6978311e0c5/mycodo/scripts/upgrade_commands.sh#L104-L120) is the bash script to restart apache2. I really have no idea why it sometimes has issues and sometimes does not. Perhaps it's the apache2 script issue and I should be issuing:

```
/etc/init.d/apache2 stop
/etc/init.d/apache2 start
```

instead of

```
/etc/init.d/apache2 restart
```

But I have no grounds for why that would affect anything. Just throwing out ideas to test. Although this is a difficult thing to test because the issue has only presented itself during the upgrade process. Otherwise, when I execute the upgrade script as root it works fine.

kizniche, 2017-10-29:
Here is one of my development Pis that updated to the latest version without issue:

```
#### Setting permissions

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service.

#### Creating Mycodo database if it doesn't exist
#### Successfully connected to http://localhost

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Done.

Sat Oct 28 13:45:16 EDT 2017
Upgrade completed successfully without errors.
```

kizniche, 2017-10-29:
Can you paste the rest of your upgrade log? From ```#### Setting permissions``` onward.

pukkita, 2017-10-30:
Sure:

```
#### Setting permissions

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service
```

pukkita, 2017-10-30:
From 5.3.2 -> 5.3.3:

```
#### Setting permissions

#### Restarting the Mycodo web server
Restarting apache2 (via systemctl): apache2.service
```

Theoi-Meteoroi, 2017-11-16:
Unfortunately, the init system is deprecated and will experience apparent entropy over time.  Systemd has spread its vast roots and may have touched Apache.  I've seen this in a former life, as systemd was introduced into the enterprise. Man, were they not looking for it.  It was shenanigans for months as driver installations failed on the next boot.

pukkita, 2017-11-16:
systemd is way different to SysV/etc but once you get the hang of it has much more flexibility and features.

I see there's another daemon monitoring mycodo and restarting it in case of failure, didn't you try with systemd Restart-on-failure feature?

kizniche, 2017-11-18:
I don't know much about systemd, so I haven't really explored much in the way of restarting on failure to know if it would be effective.

kizniche, 2017-11-18:
I went ahead and changed in 5.4.1 the restart behavior of a[ache2 from:

```/etc/init.d/apache2 restart```

to

```service apache2 restart```

closed 2017-12-13 by kizniche