kizniche/Mycodo #306: Mycodo 5.3.2 PWM input not working

pukkita, 2017-10-29:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.3.2
 
#### Problem Description

  - Added PWM Output from scratch.

[image]

  - Added PWM input related to previous output (from scratch)

[image]

No data is being displayed by that PWM  sensor:

[image]

### Errors

No apparent errors. Nothing in the log.

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Add PWM output
 2. Add PWM input sensor
 3. Check sensor detected values

pukkita, 2017-10-29:
More info:

###gpio readall:

```
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 1 | OUT  | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```

###  pstree -p 738 (daemon and threads) 
```
python(738)-+-{python}(741)
            |-{python}(742)
            |-{python}(743)
            |-{python}(745)
            |-{python}(748)
            |-{python}(749)
            |-{python}(750)
            |-{python}(751)
            |-{python}(754)
            |-{python}(755)
            |-{python}(763)
            |-{python}(764)
            |-{python}(766)
            |-{python}(9102)
            |-{python}(9103)
            |-{python}(14959)
            `-{python}(15974)
```

###  top -b -H -n 1 -p 738 (daemon and threads) 

```
top - 16:28:01 up  5:53,  1 user,  load average: 0.22, 0.22, 0.25
Threads:  18 total,   0 running,  18 sleeping,   0 stopped,   0 zombie
%Cpu(s):  6.7 us,  0.4 sy,  0.0 ni, 92.7 id,  0.2 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem :   945512 total,   322872 free,   203688 used,   418952 buff/cache
KiB Swap:   524284 total,   524284 free,        0 used.   652628 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S %CPU %MEM     TIME+ COMMAND
  738 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.02 python
  741 root      20   0  362848  73000  13576 S  0.0  7.7   0:05.46 python
  742 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.94 python
  743 root      20   0  362848  73000  13576 S  0.0  7.7   0:06.10 python
  745 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
  748 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.03 python
  749 root      20   0  362848  73000  13576 S  0.0  7.7   0:20.05 python
  750 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
  751 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
  754 root      20   0  362848  73000  13576 S  0.0  7.7   0:06.91 python
  755 root      20   0  362848  73000  13576 S  0.0  7.7   0:15.35 python
  763 root      20   0  362848  73000  13576 S  0.0  7.7   1:58.97 python
  764 root      20   0  362848  73000  13576 S  0.0  7.7   1:05.68 python
  766 root      20   0  362848  73000  13576 S  0.0  7.7   1:17.76 python
 9102 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
 9103 root      20   0  362848  73000  13576 S  0.0  7.7   0:08.38 python
14959 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
15974 root      20   0  362848  73000  13576 S  0.0  7.7   0:00.00 python
```

kizniche, 2017-10-29:
Try using different pins and connecting those pins with a wire.

kizniche, 2017-10-29:
Without testing, I would guess using the same pin won't work, so you'll have to generate the PWM on one pin and sense the PWM signal on another pin.

pukkita, 2017-10-29:
Ouch sorry...  I thought PWM would be "inferred" from PWM being sent to output device as with the read_PWM.py script.

Tried connecting BCM 23 to 24 and sensing from 23 on PWM input, but no PWM reading.

### Sensor

[image]

### gpio readall

```
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 0 | IN   | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | IN   | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 1 | OUT  | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI |   IN | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO |   IN | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK |   IN | 0 | 23 || 24 | 1 | IN   | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | IN   | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 3---+---+------+---------+-----+-----+
```

kizniche, 2017-10-29:
Strange. I'll run some tests this afternoon. I don't have my oscilloscope with me right now so I may not be able to do much right now.

kizniche, 2017-10-29:
I just released v5.3.3 that I tested and confirmed PWM and RPM Inputs to work (sensing at the same time even) with a PWM Output (see below).

[image]

[image]

[image]

pukkita, 2017-10-30:
Still fails for me on 5.3.3:

```
2017-10-30 10:03:47,422 - mycodo.sensors.pwm - ERROR - PWMInput raised an exception when taking a reading: a float is required
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/sensors/signal_pwm.py", line 180, in read
    self._duty_cycle) = self.get_measurement()
  File "/var/www/mycodo/mycodo/sensors/signal_pwm.py", line 163, in get_measurement
    time.sleep(self.sample_time)
TypeError: a float is required
```

Same results, both using same gpio as PWM output, and connecting BCM 23 to 24 where the PWM output is.

Should an extra gpio pin be used for PWM reading?

pukkita, 2017-10-30:
Ok, I deleted the PWM output and input, rebooted and re-added them.

No error in the logs, but no measurement either.

kizniche, 2017-10-30:
I forgot to mention I added a new option "sample time" so that will need to be set (default: 2 seconds). Make sure you put a resistor between the Output PWM signal and the PWM Input.

kizniche, 2017-10-30:
Not sure what that error you're getting. I used pins BCM 11 and 9 if you want to replicate that. 11 output, 9 input. 10k resistor.

pukkita, 2017-10-30:
Yes, I noticed that and deleted then re-created the PWM input, saw by default it sets sample time to 2s.

> Make sure you put a resistor between the Output PWM signal and the PWM Input.

d'oh! 

Added the 10k resistor, still no joy nor errors in the log.

kizniche, 2017-10-30:
Just to make sure, you're turning the PWM Output on by putting in a duty cycle (50% for instance) and then selecting "Turn On"?

pukkita, 2017-10-30:
Tried BCM 11, working like a charm. I'm afraid I screwed GPIO23?

kizniche, 2017-10-30:
Perhaps. Without a resistor, you might have sunk too much current into that pin. You could have also fried your output pin as well. I should have mentioned it earlier, but always use resistors to limit current coming from and going to the GPIO pins. Glad to hear it's working... You hopefully have many more pins free to use.

pukkita, 2017-10-30:
I setup the MOS module, Fan PWM control working like a charm.

Added a RPM sensor (10k resistor between fan tacho wire and GPIO).

I tried 2 pulses per revolution but the value is reading is way off, 1200000 RPM.

Have a multimeter with RPM reading, on it reads about 3400rpm. 

Should I hack 705 as pulses per revolution or does this mean I have some parameter way off? What's the Current Draw on PWM output for? I set it up to the max this fan draws (110mA)

Just for the sake of info, I set the minimum PWM output to 35% as otherwise the fan won't spin. However PWM is read as 40% even when output PWM reads 35%, is there some error to be expected or does this mean I have something off?.
Duty cycle

kizniche, 2017-10-30:
Try 0.5 pulses per revolution.

Are you using a hardware PWM pin for the output (see Outputs section of the manual for those pins)? Duty cycle measurements will be slightly off, but 5% looks like more variance than I experienced (though I didn't thoroughly test). What frequency are you using?

kizniche, 2017-10-30:
Also try increasing your RPM sample time to be more accurate.

kizniche, 2017-10-30:
Actually, you would need to increase your pulses per revolution to bring that 1.2M RPM measurement down. I'm not sure why it's reading that high.

kizniche, 2017-10-30:
There may be some extra circuitry than just a resistor to properly measure the tach wire. I'd look around the internet for someone doing the same thing with a Pi.

pukkita, 2017-10-30:
Now I recall you said to use a voltage divider to bring down voltage, tacho voltage is 7V, so I used a voltage divider from tacho to ground with two same ohms resistors I had laying around, with a 10k resistor on their center "tap" in series with gpio RPM input pin, is this fine? voltage reads 1.7V

kizniche, 2017-10-31:
Anything over 1.3 volts is detected by the GPIO as high, so that should work. Have you used your multimeter to measure the RPM on the same signal wires you connect to the Pi?

pukkita, 2017-10-31:
Yes, it has a Frequency/RPM function. Got it like 15 years ago, http://uk.rs-online.com/web/p/digital-multimeters/3692214/

Readings are sensible, right now it's at its minimum and reads 2700RPM.

pukkita, 2017-10-31:
> There may be some extra circuitry than just a resistor to properly measure the tach wire. I'd look around the internet for someone doing the same thing with a Pi.

Maybe this can help: https://www.raspberrypi.org/forums/viewtopic.php?f=31&t=51501#p406638

pukkita, 2017-11-03:
PWM input worked for some time, then started reading 0.

Also noticed top is 1, with one of the mycodo python processes steadily taking 100% CPU. A reboot changed nothing.

Nothing out f the ordinary in the logs.

kizniche, 2017-11-03:
Are you using a sufficiently large resistor? Sounds like you may have fried your input. That, or your output isn't working properly.

kizniche, 2017-11-03:
I have a few projects to work on this weekend, but I'll try to get a fan running with a tachometer input set up some time to test myself.

pukkita, 2017-11-03:
Yes, that's what looks like I'm afraid... I used a 10k one.

However 12 PSU isn't sharing ground with Pi ground, should it? The MOS module is powered via RPi 5V, then 12V PSU is wired directly to the module.

> I have a few projects to work on this weekend, but I'll try to get a fan running with a tachometer input set up some time to test myself.

That would be great :)

pukkita, 2017-11-06:
Didn't touch a single setting, and let the system running while keeping an eye on it.

Funnily it seems the GPIO pin is fine, I'd say is something related to low PWM duties.

What happened is temperatures dropped, and the PID was commanding the minimum duty cycle.

minimum PWM is set to 35% (otherwise the fan doesn't spin), but seems when PID commands it to be in that minimum "zone", PWM is not detected. On output PWM read 35% (minimum) but on input no PWM was detected.

I changed the PID lowering the setpoint so that PWM kicks-in and now PWM is measured.

kizniche, 2017-11-06:
Doesn't the voltage of the tachometer increase as the RPM increases? If so, perhaps the voltage is too low at a low RPM for the Pi to detect.

pukkita, 2017-11-06:
Didn't check, it's PWM what I was looking at; RPM always read the same regardless of the PWM duty cycle, using 705 pulses per rev shows 3404 RPM until DC is > 90%, showing a too slight increase to 3505 sometimes; right now DC=99% and RPMs show still as 3404.

I checked voltages and PWM voltage is 1.5V, RPM voltage is 1V at minimum PWM; 

At full duty cycle, PWM V = 300mV, RPM V = 2V  at full duty cycle.

kizniche, 2017-11-06:
Anything lower than ~1.2/~1.3 volts will be registered as LOW (https://www.youtube.com/watch?v=Wr49ia3oID4), so you may want to change your voltage divider so the lower limit is higher than this to be able to measure the lower RPM.

pukkita, 2017-11-07:
Will tweak the voltage divider and test. On highest RPM, voltage from the tacho is actually higher, not lower.

Wouldn't it be possible to simply set an input which would read the PWM output  mycodo is commanding straight from mycodo internals?

The main goal of having PWM as input is to record what did mycodo command to the device, so this would be a straightforward, zero additional load approach.

It will be actually the RPM measurement what will provide the actual fan-reported RPM as physical check and status  feedback, "closing the loop" for DC FAN control/monitoring.

The RPi generating the PWM and sensing it at the same time imposes an unneeded burden on the RPi, and I think is prone to induce interference/errors given the frequencies, etc; right now I'm using short cables, but once its deployed cables are going to be much longer.

kizniche, 2017-11-07:
> On highest RPM, voltage from the tacho is actually higher, not lower

Yes, I didn't say anything that contradicted this. I stated the inverse, that "the voltage is too low at a low RPM for the Pi to detect."

> Wouldn't it be possible for the PWM input to read the PWM output mycodo is commanding straight from mycodo internals?

Not sure. I never tested that. I've only tested separate pins for an output and input.

> The main goal of having PWM as input is to record what did mycodo command to the device

This is already recorded. All duty cycles produced in Mycodo are stored int he database.

> It will be actually the RPM measurement what will provide the actual fan-reported RPM as physical check and status feedback, "closing the loop" for DC FAN control/monitoring.

Agreed. The RPM measurement is more useful than measuring the PWM, as the PWM can be assumed to be generated correctly. The fan RPM is what needs to be regulated by PID, for instance.

pukkita, 2017-11-07:
> >     On highest RPM, voltage from the tacho is actually higher, not lower
> 
> Yes, I didn't say anything that contradicted this. I stated the inverse, that "the voltage is too low at a low RPM for the Pi to detect."

Ouch, my fault sorry :)

>>     The main goal of having PWM as input is to record what did mycodo command to the device
> 
> This is already recorded. All duty cycles produced in Mycodo are stored in the database.
 
That's exactly what I refer to, could it be possible to create an input which will simply "sense" by reading the PWM from such database? So that the same PWM value that appears on output device will be mirrored to the PWM input. This way we can record PWM, graph it, and tune PID accordingly.

> Agreed. The RPM measurement is more useful than measuring the PWM, as the PWM can be assumed to be generated correctly. The fan RPM is what needs to be regulated by PID, for instance.

That will require knowing the working RPM range of the fan... 

The usefulness I see on the PWM "input" (mirrored from output) and RPM inputs is that just by glancing at a PWM graph, you would instantly know: 

1. What duty cycles is the fan requiring to keep the setpoint, which will help to tune PID, 

2. Determine if the fan is properly sized for the duty (if fan is 70% of the time at 99% PWM DC will mean for better accurate setpoint tracking a more powerful one will be needed) 

3. RPM reading  will instantly signal if the fan is in fact spinning, and if RPMs "match" the duty cycle (if they doesn't, will mean a shot bearing, obstruction, failing fan/contact, etc.

pukkita, 2017-11-07:
What is stored? the actual value, or time-series?

I don't mean to store it a second time, just to create a "pseudo" input that will use that same data for display as input.

kizniche, 2017-11-07:
It's stored as the Output PWM. I'm not sure who's more confused, me or you. lol

pukkita, 2017-11-08:
heheh :)

Ok, it's stored as the output PWM. Is it possible to graph it along time on mycodo 5.3.4?

kizniche, 2017-11-08:
You should see it in the Output IDs of the graph. If this output is being used with a PID, then there should be a periodic log of the Output PWM to the database (which will show up on the log). How is your output PWM being generated (PID, Timer, manually, etc.)?

pukkita, 2017-11-08:
By PID.

kizniche, 2017-11-08:
With a PID, you can also display the "PID Output (Duty Cycle)" and this will be the same as the PWM Output (except if there's a duty cycle min/max in the PID options). If there is a min/max set, then it's often good to have both displayed on the graph (what I do).

pukkita, 2017-11-08:
How is that displayed? I thought those wonderful PID graphs are only available on the development version?

kizniche, 2017-11-08:
See the screenshot, below, of adding a new graph

[image]

pukkita, 2017-11-08:
Wow!! didn't recall seeing all those options! Superb! 👍

Theoi-Meteoroi, 2017-11-10:
I've been following this with interest - and it seems that the interfacing to the GPIO pins may not be as straight forward as it seems, depending on the fan internal design. This is a pretty good thread that illustrates my point: 

[https://electronics.stackexchange.com/questions/69613/how-to-read-rpm-from-3-wire-fan-pic18f4550](url)

I think open-collector designs are probably most common for DC fans meant for PC enclosures and such. It helps to snag a copy of the datasheet if possible.

pukkita, 2017-11-11:
Thanks Meteoroi, will try when I fix PWM output back...

In fact was about to try fixing my RPM reading issues when noticed PWM is no longer working. 

Fan is at full speed, even when on output I can see PWM changing as I change the setpoint, or even when output appears as inactive.

If I leave the fan with MOS module and power off the Pi, or unplug the gpio wire, the fan still spins, at a really low, almost about to stop, speed. This has happened since the beggining.

I replaced MOS module, same behaviour. Used a different GPIO pin, same behaviour.

gpio readall doesn't show the gpio as pwm but OUT, don't know if that's to be expected.

Only modification I made to the system was adding a DTO to change 1wire to GPIO 27 (added a DS18B20 and plan to add more)

```
dtoverlay=w1-gpio,gpiopin=27
```

Nothing relevant on mycodo log:

```
2017-11-11 22:50:23,116 - mycodo.daemon - INFO - Mycodo daemon v5.3.4 starting
2017-11-11 22:50:23,650 - mycodo.daemon - INFO - Anonymous statistics enabled
2017-11-11 22:50:23,695 - mycodo.daemon - INFO - Starting rpyc server
2017-11-11 22:50:24,037 - mycodo.output - INFO - PWM 1 setup on pin 23
2017-11-11 22:50:24,039 - mycodo.output - INFO - Output controller activated in 130.6 ms
2017-11-11 22:50:24,041 - mycodo.daemon - INFO - All activated timer controllers started
2017-11-11 22:50:24,270 - mycodo.sensor_1 - INFO - Activated in 166.0 ms
2017-11-11 22:50:24,460 - mycodo.sensor_3 - INFO - Activated in 137.6 ms
2017-11-11 22:50:24,739 - mycodo.sensor_4 - INFO - Activated in 224.8 ms
2017-11-11 22:50:24,740 - mycodo.daemon - INFO - All activated sensor controllers started
2017-11-11 22:50:25,231 - mycodo.pid_1 - INFO - Activated in 430.0 ms
2017-11-11 22:50:25,232 - mycodo.daemon - INFO - All activated PID controllers started
2017-11-11 22:50:25,903 - mycodo.lcd_1 - INFO - Activated in 576.3 ms
2017-11-11 22:50:25,904 - mycodo.daemon - INFO - All activated LCD controllers started
2017-11-11 22:50:25,904 - mycodo.daemon - INFO - Mycodo daemon v5.3.4 started in 2.784 seconds
2017-11-11 22:50:25,920 - mycodo.daemon - INFO - 49.86 MB RAM in use
2017-11-11 22:50:35,293 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2017-11-11 22:52:30,527 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2017-11-11 22:52:35,356 - mycodo.pid_1 - ERROR - Failed to read measurement from the influxdb database: Could not connect.
2017-11-11 22:52:45,387 - mycodo.influxdb - ERROR - No measurement available in the past 60 seconds.
2017-11-11 22:52:45,389 - mycodo.pid_1 - WARNING - No data returned from influxdb
```

kizniche, 2017-11-11:
Connect your PWM output via 10k resistor to a PWM Input to verify its producing a PWM signal.

pukkita, 2017-11-11:
Done, no PWM detected.

kizniche, 2017-11-11:
You may have damaged your Pi. Have another to test with?

pukkita, 2017-11-11:
Yes, I'm afraid GPIO 23 and 24 are shot... cannot understand why it has been working for weeks with GPIO 24... switched to 25 and all is well again.

pukkita, 2017-11-12:
Tried this approach:

[image]

10K pullup goes to RPi 5V positive rail, which powers the MOS module.

No difference in RPM readings, set as per 2 pulses per revolution it reads 1200000 RPM, same as using voltage divider.

kizniche, 2017-11-12:
Does the RPM change if you slow the fan with your finger?

Mycodo doesn't employ any internal resistors. You should always assume this should be done in your hardware design. I didn't want any confounding variables like an extra resistor added that may interfere with a circuit.

pukkita, 2017-11-12:
No, it doesn't change. Sometimes it displays 1250000, others 1200000, regardless of duty cycle/RPM.

kizniche, 2017-11-12:
If the RPM doesn't change, it appears to be an issue with the circuit you're using that's not allowing the Pi to interpret the signal. Can you hook up a multimeter and move the fan blades manually and measure a voltage every time the hall sensor is switched? If so, is the voltage in the detectable range of the Pi? if so, can you do the same when connected to the Pi and read a very low RPM?

pukkita, 2017-11-12:
I noticed something.

Figures I posted previously were taken while using 30kHz as output frequency. If I change the frequency to 20kHz, RPMs read are exactly half lower 600000, if I lower even more to 10kHz (annoying as there's audible frequency resonance), RPMs read are even less; DMM reads 600, RPM input 300000. 

My DMM has Hz mode too, I measured and same frequency used in mycodo output is read on the DMM.

Will test what you suggested next. Will also try to locate my scope, and probes, hope it still works, I haven't used it for years, hope is less rusty than me lol.

Theoi-Meteoroi, 2017-11-12:
This is an application note from a fan vendor that discusses some of the various fan tachometer designs and the internal circuitry. I'll build, test and document a circuit that should work with common PC case fans (open collector, non-isolated) and submit to the project.
[Comair_CM_MonitoringFanPerformance.pdf](https://github.com/kizniche/Mycodo/files/1465187/Comair_CM_MonitoringFanPerformance.pdf)

pukkita, 2017-11-13:
> Added: The more I consider the problem, I can see that direct PWM of the fan supply voltage has issues that need to be addressed to prevent premature wear, enhance reliability and prevent glitching the tachometer output. The PWM signal needs to be integrated through a low-pass filter before the power stage - unless the fan is explicitly designed for PWM drivers.

Makes all the sense. 

If using a four wire one I assume the fan can be powered directly with 12V PSU, with its ground connected to RPi ground,  fan PWM pin directly connected to the PWM output gpio,  and tacho wire interfaced to the Pi following the previous circuit using a diode and R2 pullup to RPI 3.3V?

Theoi-Meteoroi, 2017-11-16:
I've ordered a couple of PWM fans to test with however, I still want to make the 3-wire type work. I have some blower style 3-pin fans that will work dandy in an enclosure without cutting too large a hole and perhaps even support a closure flap for when a sample isn't needed.  I'm looking at PM2.5 detectors that won't last with constant, contaminated airflow.  I've already almost finished a test circuit on a spare Pi HAT.  I'll submit the tested design when I reach that milestone.

closed 2018-02-26 by kizniche