kizniche/Mycodo #310: Testing Linux command

reymart081678, 2017-11-07:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.3.3
 
#### Problem Description
Please list: 

Need to test how the linux input command work
Please provide sample and the condition statement is blank and not sure how to set it to enable the action.

[image]

### Errors

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?   Sample way to use linux input/output command

kizniche, 2017-11-07:
Available variables can be found [here](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#conditional-statement-variables).

kizniche, 2017-11-07:
At the moment there isn't a variable for the measurement acquired from the Linux Command Input, which there should be. I'll create an issue to remind me to add this feature.

closed 2017-11-07 by kizniche