kizniche/Mycodo #312: Sql Alchemy Error after upgrade

RynFlutsch, 2017-11-11:
Hello Kyle

## Mycodo Issue Report:
  - Specific Mycodo Version:
upgrade from 5.3.3 to 5.3.5

Is it possible fix this without the need of fresh a install?

Thank You

[image]

kizniche, 2017-11-11:
It looks like your database wasn't upgraded. Use the following command and then reboot.

cd ~/Mycodo/database && sudo ~/Mycodo/env/bin/alembic upgrade head

RynFlutsch, 2017-11-11:
Great fix, thank you!

kizniche, 2017-11-11:
You're welcome. Glad it was an easy fix.

Could you attach the end of your upgrade log (from when the last upgrade started to the end) so I may inspect it for what went wrong and find any potential issues with the Mycodo upgrade process?

RynFlutsch, 2017-11-11:
I think its not possible, but i can watch monday.

It startet with the missing python-fasteners.
After i added that, the sql error occurred.

kizniche, 2017-11-11:
You can access the log in the Mycodo menu: Config -> Mycodo Logs -> Upgrade Log

closed 2017-11-13 by kizniche