kizniche/Mycodo #313: Add picamera-specific image options

kizniche, 2017-11-11:
I've finally implemented the picamera options, but have no yet included all that are available. This is here for a reminder to add those options.

camera.ISO
camera.exposure_mode
camera.meter_mode
camera.awb_mode
camera.image_effect
camera.color_effects

closed 2019-10-19 by kizniche

chitters2004, 2019-10-20:
Could you please add an option to hide the last time lapse and still photos? Having the last still appear first would probably make more sense too. Oh and the video stream is still a different zoom/resolution to still images!

kizniche, 2019-10-20:
I asked for information to investigate that issue in the issue you mentioned the resolution problem. https://github.com/kizniche/Mycodo/issues/703#issuecomment-544189217