kizniche/Mycodo #315: AM2315 sensor i2c address hardcoded to 0x5c - new probes at 0x57

Theoi-Meteoroi, 2017-11-12:
## Mycodo Issue Report:
  - Specific Mycodo Version:  5.3.5
 
#### Problem Description
Please list:    

I have new AM2315 sensors that are at i2c 0x57 address. The sensor type is hardcoded for 0x5c.

  - what were you trying to do
  - specific setup details that are involved
 
### Errors

N/ A

  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:

Isolate AM2315 alone on a i2c bus

run i2cdetect to discover address used:

pi@control1:~ $ sudo i2cdetect -y 3
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- 57 -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- UU -- -- -- -- -- -- -- 
70: 70 -- -- UU -- -- -- 77               

How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

Please make the i2c address editable or at least a selection 0x57 - 0x5c

kizniche, 2017-11-12:
I'll make it an editable text field with the default at 0x57
Expect a new release with this change sometime today

closed 2017-11-12 by Theoi-Meteoroi

Theoi-Meteoroi, 2017-11-12:
Opened in error - further testing and probing with a bus-pirate sniffer revealed this 0x57 i2c address is actually a 24c32 EEPROM on a DS3231 RTC board, configured in the system.  I always use a hardware RTC with Raspi. Apologies for the red herring.
I'm resuming debug of the AM3215 ( still not working ) to find out why the wakeup commands sent by the driver are apparently not working for the AM2315 samples I have.  The data sheet is explicit about the sleep behavior although the grammar can lead to confusion.
[AM2315.pdf](https://github.com/kizniche/Mycodo/files/1465134/AM2315.pdf)

kizniche, 2017-11-12:
No problem. It wasn't a major change. I'll revert the edits.

Theoi-Meteoroi, 2017-11-12:
As I parse the datasheet closely, there are clues to why I saw my original AM2315 working with Mycodo initially and then had consistent trouble as I added sensors.  I will debug it and return when I have something useful to share.

Theoi-Meteoroi, 2017-11-16:
Well, I figured out why this sensor is unreliable with Rpi3 hardware I2C.  It is among a number of I2C devices that really hates the BCM2835 clock stretching blunder (hardware bug: https://github.com/raspberrypi/linux/issues/254 ).   The wakeup attempts fail, consistently. I checked the bitstream with a sniffer, and see that the sensor may respond once out of 20 or so tries (or not at all) but only with a single byte returned. The solution is to use a software implementation of the I2C bus. You need to add pull-up resistors (4.7k is dandy) to 3.3v and install the i2c_gpio device overlay.   Seems to work fine now, will run for a few days, but the CRC failures are gone and I get good readings, every time.  And no twiddling the power for the sensor is required.

kizniche, 2017-11-16:
Thanks for the info. If you can explain a little more in-depth the steps you took to do that, I can include it in the manual under the AM2315 section.

kizniche, 2017-11-16:
Is [this comment](https://github.com/raspberrypi/linux/issues/254#issuecomment-267474334) how the overlay is enabled in /boot/config.txt?

Theoi-Meteoroi, 2017-11-16:
I was playing with the delay ... 2 uSec is default,  AM2315 seems to like it set for 4 uSec.

These are settings I use in /boot/config.txt

dtparam=act_led_trigger=heartbeat
dtoverlay=i2c-gpio,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=4      <---- this line
dtoverlay=i2c-mux,pca9548,addr=0x70
dtoverlay=i2c-mux-pca9545a,addr=0x73
dtoverlay=i2c-rtc,ds3231
dtoverlay=pi3-disable-bt
start_x=1
gpu_mem=128

Theoi-Meteoroi, 2017-11-16:
I wanted to add - when the data sheet has SMBus or I2c_ModBus (as in this case) mentioned, the device is probably going to need SW I2C. It is sort of an indication of the age of the design.  Also mention of a maximum 100khz clock. New compliant designs clock at 400khz or better.  This app note discusses how SMBus != I2C

https://www.maximintegrated.com/en/app-notes/index.mvp/id/476

kizniche, 2017-11-22:
When the i2c-gpio overlay is working, does it merely add a new bus at /dev/i2c-2?

kizniche, 2017-11-28:
I finally had my AM2315 stop responding, so I started testing with the overlay. I found it appears on /dev/i2c-3

kizniche, 2017-11-28:
Have you had your sensor hang at all since putting it on the software I2C bus?

kizniche, 2018-06-05:
Just an update to round-off this issue. My AM2315 has been working perfectly on the software I2C.