kizniche/Mycodo #316: Suggestion: LCD multi-screen browsing

pukkita, 2017-11-12:
A nice feature that will extend greatly LCD functionality would be:

- Making possible to design more than one screen, e.g. on a 4 line LCD, have first 4 Lines as 1st screen, 5-8 as second screen, 9-13 as third, etc.

- By using a rotary encoder, or a simple momentary-contact button on a gpio pin, signal mycodo to switch screens: initially screen will display 1st screen, once button is pushed, second, etc: if there were 5 screens: 1->2->3->4->5->1->2....

- Being possible also to set screens to  "auto switch" with a definable delay/display period  would be a nice finishing touch.

kizniche, 2017-11-12:
Great idea! I would probably be inclined to introduce this with a switching period, and then perhaps at a later time add button support.

pukkita, 2017-11-12:
Yes, autoswitching will be great, no need for button/rotary encoder presses/wiring/switches.

kizniche, 2017-11-13:
This new feature dropped in nicely and is ***very*** useful. Thanks for the recommendation. I'll get this released tonight, right after I fix this last bug related to displaying a degree symbol on an LCD.

closed 2017-11-13 by kizniche