kizniche/Mycodo #319: Live graphs zoom hold

Magnum-Pl, 2017-11-13:
Is there a way you can add the ability to hold the zoom level on a Live Graph through a refresh? I don’t mean a full page refresh, just the graph refresh itself. When I’m fine tuning the PID for my CO2 regulator, I’d like to be able to set the graph refresh rate at 3 seconds to match my sensor reading rate. This would allow me to watch the graph move in real time across the screen like an EKG instead of the usual 30-60 second refresh rate. As it is now, the zoom level returns to the default level after every refresh and I spend a lot of time going back and forth between the default and the 30m zoom level.

kizniche, 2017-11-13:
I spent a lot of time working to get where it's at right now. There are a lot of issues that come with zomming I found. For instance, if it doesn't reset the start and end points, then the graph stays zoomed wherever you zoomed, and you cannot see updated data.

Is this what you're trying to accomplish (example): If you have a 30 minute x-axis and zoom to the most recent 5 minutes, and then have the most recent 5 minutes continually update and stay in that 5 minute time frame?

For what you describe what you want to do, can't you make two graphs, one 30 minutes and the other 5 minutes, with the same data displayed? That way you can see the long and short data sets without having to mess with zooming.

Magnum-Pl, 2017-11-13:
The example you described is what I’m looking for. I use two graphs like you suggested and a  gauge, but I’d love to be able to watch it continually update at whatever zoom I set it to.
I understand what you’re saying about the zooming issues though, so that’s a bummer.  

Also, can you add the option to display Fahrenheit in the Live Graph gauges?

kizniche, 2017-11-20:
Check out this (https://www.highcharts.com/stock/demo/dynamic-update/) and let me know if that's what you're trying to do. I'm working on integrating this ability.

Magnum-Pl, 2017-11-20:
Yes, that’s exactly what I had in mind.

kizniche, 2017-11-20:
In the middle of adding a few more graph options, I was going to add an option to disable the auto min/max reset, then I realized I didn't need to, because this feature already exists with the current iteration of the Navbar feature. Just check the Enable Navbar and see for yourself.

Magnum-Pl, 2017-11-21:
I found it! I get the results I was looking for by setting the X-axis to 1-5 minutes and the graph refresh rate at 1 second. Not sure why I didn’t try that before. Was that always there??

Is there a way you can add the ability to set a graph refresh rate of less than 1 second? As it is now, entering a decimal saves as “none”.

kizniche, 2017-11-21:
It's been there for about a year or more. lol

I'll see about changing the refresh period to a decimal.

Magnum-Pl, 2017-11-21:
It works great as is, but the decimal refresh rate might smooth it out a bit. 

Thanks for your help

closed 2017-11-21 by Magnum-Pl

kizniche, 2017-11-21:
Strange. I'm now testing very short refresh durations by having a Linux Command Input taking measurements several times per second and a graph with a less than 1 second refresh rate. Earlier during my tests I could swear the graph didn't reset the zoom automatically when using the Navbar,, but now it is doing it. It seems as though I have to bring that option back to reset the x-axis on refresh.

Magnum-Pl, 2017-11-21:
The Navbar and Range Selector always reset after each refresh for me. The only way I was able to get it to hold the zoom level was to manually set it in the graph options. Doing that means I can’t use the Navbar and Range Selector though, so that’s sort of a work around. It would be better if the zoom level held. 

How well does the graph work with the less than 1 second refresh rate?

kizniche, 2017-11-21:
Okay, I added the option "Enable X-Axis Reset" and then unchecked, the Navbar can be moved/zoomed and it retains it's duration after graph refreshes.

Less than one second works, but the graph tends to jump around a lot when it redraws. Though that may be a side effect of of the random data I'm using.

I'll push this update and get a new release out.

Magnum-Pl, 2017-11-21:
Will that also hold the zoom when using the range selector buttons? 

It bounces around like that with a 1 second rate when I have the zoom set at 1 minute, but if I pull the zoom back a bit, it’s not as bad

kizniche, 2017-11-21:
Not sure. I didn't test that. I just made the release so you can go try for yourself ;)

Magnum-Pl, 2017-11-21:
It’s working great. With the “enable x-axis reset” box unchecked, the zoom level does not reset after a graph refresh using either the navbar or zoom range selector. 

I’m testing a less than 1 second graph refresh rate and even though I’m now able to set this as a decimal, the refresh rate is still only 1 second. I have two graphs running on top of each other with a .1 second and 1 second refresh rate and they both move at the same speed. You can only see this at a resolution of 3 minutes or less. Were you able to get the graph to move faster than 1 second when you were testing?

kizniche, 2017-11-21:
Make sure your measurements are being acquired less than a second apart if you want the graph to update quicker than 1 second. I suspect your measurements are 1 second or longer apart. Try creating a Linux Command Input with the default command (generates random numbers) and period set to 0.25 seconds and then create a graph with a refresh period of 0.5 seconds.

kizniche, 2017-11-21:
Look at the timestamp of the data to see if it's actually shorter than a second apart.

Magnum-Pl, 2017-11-22:
I started testing with the sensor read rate at .1 seconds and the graph refresh rate at .5 seconds. The graph still refreshes at a 1 second rate. I’ve tried other sensor and refresh rate combinations with the same results. 

It seems like the graph has fresh sensor readings, it’s just not able to refresh at a rate faster than 1 second. 

I know just enough Linux to setup and run a Pi headless, but not enough to do what you suggested. 

This isn’t a huge issue for me at this point. I’m satisfied with a 1 second refresh rate. I was hoping a faster rate might smooth out some of the bouncing around.

kizniche, 2017-11-22:
I just figured it out. If only one new data point is placed on the graph, then the graph moves across the screen smoothly and without doing that spaztic redraw. If I set the Input period or Graph period that makes the graph acquire more than one point in the series, then it won't be a smooth movement. Try these two settings and see the difference:

Add a new Input of the type: Linux Command
Set the Input Period to 1 second.
Set the Graph Period to 0.5 seconds.

Add another Input of the type: Linux Command
Set the Input Period to 0.25 seconds.
Set the Graph Period to 1 second

kizniche, 2017-11-22:
Also, I have not yet been able to get the graphs to move faster than 1 frame per second.

Magnum-Pl, 2017-11-22:
Oh, that Linux Command! That makes sense. I’ve never used that before. 

I see what you mean about the smoother movement. The graph doesn’t bounce around as much with the data from my MH-Z16 CO2 sensor. I still get a funky effect during each refresh unless the refresh rate is 1 second or above though. For now, I’m going to leave the sensor and graph refresh rates at 1 second. 

It’s not easy to get the Navbar zoomed into a 1 minute resolution. Can you add 1, 5 and 10 minutes buttons to the zoom range selector? To condense all those buttons, could you use a slider that would stop at those preset zoom levels instead of the buttons?

kizniche, 2017-11-22:
I tried but I wasn't able to successfully implement a range selector drop down option. However, I did add 1min, 5min, and 15 min to the range selector buttons to better cover the shorter time periods.

Magnum-Pl, 2017-11-23:
The drop down list would’ve been nice, but the buttons will work fine too. 

Thanks for your help. These changes should allow me to tune my CO2 PID a little better and with less futzing around than I was able to before.

kizniche, 2018-02-06:
I'm just about to release v5.5.19, which includes the ability to set the minimum/maximum of any y-axis of any graph. I remembered this thread because this should enable you to get your smooth scrolling of data across the screen without the y-axes adjusting and messing up the look. You may have to enable the Navbar and enable Graph Shift, or play around with different option configurations to find what works properly. I should have v5.5.19 released in the next few days. Just wanted to give you a heads-up if you wanted to test what you were trying to do in the past.

kizniche, 2018-02-06:
Update: just released v5.5.19