kizniche/Mycodo #320: error 500 when deactivate sensors

leoneo, 2017-11-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.0, raspbian fresh install
 
#### Problem Description
Please list: Error occurs when I disable a sensor

 ### Errors
Error (Full Traceback):

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 924, in page_input
    utils_input.sensor_deactivate(form_mod_sensor)
  File "/var/www/mycodo/mycodo/mycodo_flask/utils/utils_input.py", line 392, in sensor_deactivate
    form_mod_sensor.modSensor_id.data)
  File "/var/www/mycodo/mycodo/mycodo_flask/utils/utils_input.py", line 410, in sensor_deactivate_associated_controllers
    if sensor_id in [each_lcd.line_1_sensor_id,
AttributeError: 'LCD' object has no attribute 'line_1_sensor_id'

kizniche, 2017-11-17:
This issue was fixed but hasn't been released yet. I have a bit more testing to do with 5.4.1 before release. I hope to get it out sometime this weekend.

leoneo, 2017-11-17:
ok thanks I don't have the error when I deactivate the LCD

kizniche, 2017-11-17:
When the sensor is deactivated, it is checking if there are any activated PID or LCDs using the sensor, and if so, deactivating those before the sensor is deactivated.

kizniche, 2017-11-18:
I'm going to close this but if the issue appears after your upgrade to v5.4.1, please reopen it.

closed 2017-11-18 by kizniche