kizniche/Mycodo #321: no pH data UART

leoneo, 2017-11-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.0 fresh install
 
#### Problem Description
Please list: no pH data in live page

 
### Errors
`2017-11-17` 08:44:27,968 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:44:42,998 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:44:55,949 - mycodo.influxdb - ERROR - No measurement available.
2017-11-17 08:44:55,950 - mycodo.lcd_1 - ERROR - Error: 'unicode' object does not support item assignment
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_lcd.py", line 357, in create_lcd_line
    self.lcd_string_line[display_id][i] = 'ERROR: NO DATA'
TypeError: 'unicode' object does not support item assignment
2017-11-17 08:44:56,379 - mycodo.lcd_1 - ERROR - KeyError: Unable to output to LCD.
2017-11-17 08:44:58,036 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:44:58,058 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-11-17 08:45:13,091 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:45:26,540 - mycodo.influxdb - ERROR - No measurement available.
2017-11-17 08:45:26,541 - mycodo.lcd_1 - ERROR - Error: 'unicode' object does not support item assignment
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_lcd.py", line 357, in create_lcd_line
    self.lcd_string_line[display_id][i] = 'ERROR: NO DATA'
TypeError: 'unicode' object does not support item assignment
2017-11-17 08:45:26,926 - mycodo.lcd_1 - ERROR - KeyError: Unable to output to LCD.
2017-11-17 08:45:28,123 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:45:43,156 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-17 08:45:43,157 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-11-17 08:45:57,103 - mycodo.influxdb - ERROR - No measurement available.
2017-11-17 08:45:57,105 - mycodo.lcd_1 - ERROR - Error: 'unicode' object does not support item assignment
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_lcd.py", line 357, in create_lcd_line
    self.lcd_string_line[display_id][i] = 'ERROR: NO DATA'
TypeError: 'unicode' object does not support item assignment
2017-11-17 08:45:57,488 - mycodo.lcd_1 - ERROR - KeyError: Unable to output to LCD.
2017-11-17 08:45:58,190 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.

leoneo, 2017-11-17:
First surclassement 5.0.8 to 5.4.0 no problem pH data

I think that the problem is in the input "Device Location" I have change the default setting by "/dev/ttyAMA0" but there are no pH datas

kizniche, 2017-11-18:
Can you deactivate the pH sensor, then activate it, then look at the daemon log for specific errors that occur right after the sensor is activated?

leoneo, 2017-11-18:
Hi Kyle, 
Daemon log : 

  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialposix.py", line 501, in read
    'device reports readiness to read but returned no data '
SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2017-11-18 07:51:07,346 - mycodo.sensor_4 - ERROR - StopIteration raised. Possibly could not read sensor. Ensure it's connected properly and detected.
2017-11-18 07:51:13,060 - mycodo.sensor_4 - INFO - Deactivated in 64.0 ms
2017-11-18 07:51:25,758 - mycodo.device.atlas_scientific_uart_/dev/ttyAMA0 - ERROR - instance raised an exception when initializing: Could not configure port: (5, 'Input/output error')
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/devices/atlas_scientific_uart.py", line 23, in __init__
    timeout=0)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
SerialException: Could not configure port: (5, 'Input/output error')
2017-11-18 07:51:25,765 - mycodo.device.atlas_scientific_uart_/dev/ttyAMA0 - ERROR - Opening serial
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/devices/atlas_scientific_uart.py", line 23, in __init__
    timeout=0)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialposix.py", line 272, in open
    self._reconfigure_port(force_update=True)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/serial/serialposix.py", line 326, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
SerialException: Could not configure port: (5, 'Input/output error')
2017-11-18 07:51:25,770 - mycodo.sensor_4 - INFO - Activated in 288.3 ms
2017-11-18 07:51:25,772 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.
2017-11-18 07:51:40,808 - mycodo.sensors.atlas_ph - ERROR - UART device is not set up.Check the log for errors.

kizniche, 2017-11-18:
I just made a release. There aren't any changes that specifically fix this issue, but I have changed some of the Atlas UART code. If you could upgrade and report if you get the same error, I can start to diagnose the issue. Are you able to get measurements with other code using /dev/ttyAMA0?

leoneo, 2017-11-18:
I have 2 SD cards : 1st where Mycodo 5.0.28 pH data works and the 2nd with Mycodo 5.4.0 where no pH data
with the same configuration

I have upgrade to 5.4.1, and reboot but all Mycodo don't work now "Internal Server Error"

kizniche, 2017-11-18:
You let the upgrade complete fully and the upgrade log's last line was "upgrade completed successfully"?

leoneo, 2017-11-18:
no the upgrade took a long time to load, I did not get the message of success.

kizniche, 2017-11-18:
Can you attach the upgrade log?

leoneo, 2017-11-18:
How can I ? 

Mycodo doesn't start

leoneo, 2017-11-18:
After reinstallation 

#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Daemon is not running, restarting
Sat 18 Nov 09:50:32 +04 2017

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.1.91/, or whatever your Pi's
IP address is, to create an admin user and log in.

I have always the error : 

Internal Server Error

The server encountered an internal error or misconfiguration and was unable to complete your request.

Please contact the server administrator at [no address given] to inform them of the time this error occurred, and the actions you performed just before this error.

More information about this error may be available in the server error log.

Apache/2.4.25 (Raspbian) Server at 192.168.1.91 Port 443

kizniche, 2017-11-18:
The upgrade log is located at /var/log/mycodo/mycodoupgrade.log and the HTTP log is located at /var/log/apache2/error.log
Look at the apache2 error log for the trace back of the Internal Server Error

leoneo, 2017-11-18:
[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/1484606/mycodoupgrade.log)
[error.log](https://github.com/kizniche/Mycodo/files/1484607/error.log)

kizniche, 2017-11-18:
I just released v5.4.2

You can upgrade from the command line with the following command:

sudo /bin/bash ~/Mycodo/mycodo/scripts/upgrade_mycodo_release.sh

kizniche, 2017-11-18:
From your upgrade log, it looks like at 2017-11-18 8:19 your upgrade was interrrupted.

```
[2017-11-18 08:18:20] #### Updating pip
[2017-11-18 08:18:31] Requirement already up-to-date: pip in ./Mycodo/env/lib/python2.7/site-packages
[2017-11-18 08:18:33] 
[2017-11-18 08:18:33] #### Installing pip requirements from requirements.txt
[2017-11-18 08:18:43] Requirement already up-to-date: pip in ./Mycodo/env/lib/python2.7/site-packages
[2017-11-18 08:18:59] Requirement already up-to-date: testfixtures in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 1))
[2017-11-18 08:19:01] Requirement already up-to-date: mock in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 2))
[2017-11-18 08:19:04] Requirement already up-to-date: pytest==3.2.3 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 3))
[2017-11-18 08:19:05] Requirement already up-to-date: factory_boy==2.8.1 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 4))
[2017-11-18 08:19:06] Requirement already up-to-date: future in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 5))
[2017-11-18 08:19:08] Requirement already up-to-date: alembic==0.9.6 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 6))
[2017-11-18 08:19:08] Collecting argparse==1.4.0 (from -r /home/pi/Mycodo/install/requirements.txt (line 7))
[2017-11-18 08:19:10]   Downloading argparse-1.4.0-py2.py3-none-any.whl
[2017-11-18 08:19:13] Requirement already up-to-date: bcrypt==3.1.4 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 8))
[2017-11-18 08:19:14] Requirement already up-to-date: daemonize==2.4.7 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 9))
[2017-11-18 08:19:16] Requirement already up-to-date: fasteners==0.14.1 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 10))
[2017-11-18 08:19:28] Requirement already up-to-date: Flask==0.12.2 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 11))
[2017-11-18 08:19:29] Requirement already up-to-date: Flask_Babel==0.11.2 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 12))
[2017-11-18 08:19:30] Requirement already up-to-date: Flask_InfluxDB==0.1 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 13))
[2017-11-18 08:19:31] Collecting Flask_Limiter==1.0.0 (from -r /home/pi/Mycodo/install/requirements.txt (line 14))
[2017-11-18 08:19:33]   Downloading Flask-Limiter-1.0.0.tar.gz (89kB)
[2017-11-18 08:19:42] Requirement already up-to-date: Flask_Login==0.4.0 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 15))
[2017-11-18 08:19:43] Requirement already up-to-date: Flask_SQLAlchemy==2.2 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 16))
[2017-11-18 08:19:44] Requirement already up-to-date: Flask_SSLify in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 17))
[2017-11-18 08:19:45] Requirement already up-to-date: Flask_WTF==0.14.2 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 18))
[2017-11-18 08:19:48] Requirement already up-to-date: geocoder==1.32.1 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 19))
[2017-11-18 08:19:49] Requirement already up-to-date: imutils==0.4.3 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 20))
[2017-11-18 08:19:50] Requirement already up-to-date: influxdb==4.1.1 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 21))
[2017-11-18 08:19:51] Requirement already up-to-date: pi_switch==0.4.0 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 22))
[2017-11-18 08:19:52] Requirement already up-to-date: pyserial==3.4 in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 23))
[2017-11-18 08:19:53] Requirement already up-to-date: python-dateutil in ./Mycodo/env/lib/python2.7/site-packages (from -r /home/pi/Mycodo/install/requirements.txt (line 24))
```

leoneo, 2017-11-18:
there is no change, always the same error.

[mycodoupgrade.log](https://github.com/kizniche/Mycodo/files/1484993/mycodoupgrade.log)
[error.log](https://github.com/kizniche/Mycodo/files/1484995/error.log)

kizniche, 2017-11-18:
Did you upgrade to v5.4.2?

I see the issue. Remove from ~/Mycodo/mycodo/mycodo_flask/utils/utils_remote_host.py line 7:

```
from bs4 import BeautifulSoup
```

kizniche, 2017-11-18:
Rather, the only error I see in the HTTP log is due to an unused/missing module when trying to access the Remote Admin Dashboard. That shouldn't be causing the sensor read issue.

There's nothing to indicate an issue with the pH sensor. I'll check out the code again.

leoneo, 2017-11-18:
to 5.4.3

Sorry the problem is to install, i don't kwno if there is always the issue with the pH sensor because I can not access to mycodo.

kizniche, 2017-11-18:
I compared the code in v5.3.6 and the latest and there's no difference with the pH_uart device code.

As per the Atlas Scientific manual (https://www.atlas-scientific.com/_files/code/pi_sample_code.pdf), page 3: The Raspberry Pi Foundation has failed to make a working UART on the Pi 3. Because of this no UART connected devices can run on a Raspberry Pi 3.

Have you tried I2C?

kizniche, 2017-11-18:
Also, test if you can obtain measurements using the Atlas code: https://github.com/AtlasScientific/Raspberry-Pi-sample-code

kizniche, 2017-11-18:
> Sorry the problem is to install

The install log indicated the install was successful:

```
***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.1.91/, or whatever your Pi's
IP address is, to create an admin user and log in.
```

leoneo, 2017-11-19:
I have resolve the install problem by installing bs4 (see in #322 )

In 5.4.3 There are no datas to DHT22 and pH UART but RPI temperature and BMP180 Work

It is not a problem in hardware because all work great in 5.0.28
[image]
[image]

kizniche, 2017-11-19:
Are these Mycodo I stalls on the exact same systems? That is, the only variable change is the version of Mycodo running? No two different SD cards, but rather you have a working Mycodo, then stop that instance, and start another version on the exact same system (hardware and software) start the other version of Mycodo... And there is a difference in pH measurement acquisition?

I need to exclude how you set up UART as the possible issue. So, as I asked before, on the system where UART doesn't work, does the stock Atlas Scientific UART code acquire measurements? If no, it is a UART issue, and not Mycodo.

leoneo, 2017-11-19:
Thanks, 
you let me on the way to solve, I have two different SD systems and I had forgot that in the Atlas's sample code they ask to modify two files : /boot/cmdline.txt and /etc/inittab

Now pH UART works in 5.4.3 ;)

but the DHT22 always doesn't work
I do not know why?

kizniche, 2017-11-19:
Great about the pH. I'm in the middle of diagnosing the DHT22 right now (duplicate of #324).

kizniche, 2017-11-19:
I just fixed the DHT22/11 inputs. I'll make a release in a few minutes.

closed 2017-11-19 by kizniche