kizniche/Mycodo #323: PID method Bezier curve  does not function after upgrade

reymart081678, 2017-11-18:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.3.3
 
#### Problem Description
Please list: Previous version of mycodo for PID method Bezier curve is okay and after upgrade it always complain Method has ended.
It is weird why it will end.

Setting of Bezier curve which is the same  as the previous version 5.2.2 of Mycodo.

[image]

 
### Errors
 Mycodo Log. Tried several times but it will deactivate PID. For immediate resolution disable PID Method  and use static set point.
1. Remove PID and recreate still the same.
2. Remove PID method bezier curve and recreate, still the same issue.
2. Remove both PID and Method, then re create both , result is still the same.

2017-11-18 12:43:22,007 - mycodo.sensor_7 - INFO - Activated in 214.9 ms
2017-11-18 12:43:57,978 - mycodo.pid_1 - INFO - Activated in 296.7 ms
2017-11-18 12:43:58,083 - mycodo.pid_1 - WARNING - Method has ended. Activate the PID controller to start it again.
2017-11-18 12:43:58,321 - mycodo.pid_1 - INFO - Deactivated in 341.5 ms
2017-11-18 12:44:35,392 - mycodo.pid_1 - INFO - Activated in 413.6 ms
2017-11-18 12:44:35,491 - mycodo.pid_1 - WARNING - Method has ended. Activate the PID controller to start it again.
2017-11-18 12:44:35,729 - mycodo.pid_1 - INFO - Deactivated in 334.9 ms
2017-11-18 12:47:02,512 - mycodo.pid_3 - INFO - Activated in 323.6 ms
2017-11-18 12:47:02,621 - mycodo.pid_3 - WARNING - Method has ended. Activate the PID controller to start it again.
2017-11-18 12:47:02,865 - mycodo.pid_3 - INFO - Deactivated in 352.5 ms
2017-11-18 12:48:56,210 - mycodo.pid_4 - INFO - Activated in 297.7 ms
2017-11-18 12:48:56,295 - mycodo.pid_4 - WARNING - Method has ended. Activate the PID controller to start it again.
2017-11-18 12:48:56,537 - mycodo.pid_4 - INFO - Deactivated in 325.9 ms

### Steps to Reproduce the issue:
How can this issue be reproduced?
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?
Additional issue that i noticed is that using conditional relay.
 - PID ignores the conditional relay setting but using manual method of activating and deactivating the relay ,it is activating both relays as expected in the conditional.
   Example:
   
[image]

PID setting with disabled PID method.

[image]

kizniche, 2017-11-18:
I'll check this out. Thanks for the detailed report.

kizniche, 2017-11-18:
I found a bunch of issues with the latest release 5.4.1 so I pulled it until I can get the issues fixed and v5.4.2 released. I noticed you're running 5.3.3, which is a few versions old. Hopefully when you upgrade there will be less issues than that version.

reymart081678, 2017-11-18:
I got another version for 5.4.0 on different rpi. I'll try to simulate if the same behaviour

reymart081678, 2017-11-18:
The same issue using 5.4.0, just simulated using rpi temp sensor.
[image]

PID:
[image]
Method:

For relay actuation, PID reaaly does'nt follow the condition that is set.

[image]

[image]

kizniche, 2017-11-18:
I just realized I've been chasing a ghost. What's happening with PIDs and conditionals is expected behavior. If the relay is turned on (without a duration) it will trigger the conditional. If the relay is turned on with a duration it will trigger the conditional that is configured to be triggered for that duration. Because PIDs operate by turning on relays for durations, there would need to be a conditional set up for turning the relay on for a duration in order for it to trigger.

kizniche, 2017-11-18:
I can add a third option to the "If State" that will be "On (any duration)", that way the conditionals will trigger for any on duration instead of a specific on duration.

kizniche, 2017-11-18:
I just released v5.4.2

Update when yo have a chance and let me know if any of the issues you mentioned persist.

kizniche, 2017-11-18:
v5.4.3 released that fixes an issue with Output conditional triggering

reymart081678, 2017-11-19:
Thank you. I upgraded 5.4.0 to 5.4.2
   Issue i get it similar to #322  that after upgrade i get server internal error, I followed the resolution to install bs4 and works okay after bs4 installation.
   - I checked http.log and there are issues in getting information in the influxdb but unfortunately i did not capture the log information.

- The issue of PID when using Bezier method is already resolve.
[image]

- Conditional is also already resolve. 
[image]

Upgraded to 5.4.3 and upgrade is successful but need to restart apache2 manually to bring up the webpage.

upgrade.log, Last entry

[2017-11-19 02:03:16] #### Compiling mycodo_wrapper
[2017-11-19 02:03:17] 
[2017-11-19 02:03:17] #### Creating users and directories
[2017-11-19 02:03:17] The user `mycodo' is already a member of `gpio'.
[2017-11-19 02:03:17] The user `mycodo' is already a member of `adm'.
[2017-11-19 02:03:18] The user `mycodo' is already a member of `video'.
[2017-11-19 02:03:18] The user `mycodo' is already a member of `dialout'.
[2017-11-19 02:03:18] 
[2017-11-19 02:03:18] #### Setting permissions
[2017-11-19 02:03:20] 
[2017-11-19 02:03:20] #### Restarting the Mycodo web server

http.log

[Sun Nov 19 02:17:02.771348 2017] [ssl:warn] [pid 5421:tid 1996300288] AH01909: 127.0.1.1:443:0 server certificate does NOT include an ID which matches the server name
[Sun Nov 19 02:17:02.773802 2017] [mpm_event:notice] [pid 5421:tid 1996300288] AH00489: Apache/2.4.25 (Raspbian) OpenSSL/1.0.2l mod_wsgi/4.5.11 Python/2.7 configured -- resuming normal operations
[Sun Nov 19 02:17:02.773932 2017] [core:notice] [pid 5421:tid 1996300288] AH00094: Command line: '/usr/sbin/apache2'
[Sun Nov 19 08:17:27.775793 2017] [wsgi:error] [pid 6209:tid 1964430384] /var/www/mycodo/env/lib/python2.7/site-packages/flask_limiter/extension.py:113: UserWarning: Use of the default `get_ipaddr` function is discouraged. Please refer to https://flask-limiter.readthedocs.org/#rate-limit-domain for the recommended configuration
[Sun Nov 19 08:17:27.776065 2017] [wsgi:error] [pid 6209:tid 1964430384]   " for the recommended configuration", UserWarning

reymart081678, 2017-11-19:
Update:
I think this is the issue when upgrading to 5.4.X, i tried to upgrade my 5.3.3 to 5.4.3. Daemon does not start.

pi@raspberrypi:/var/log/mycodo $ sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/my                codo_daemon.py --debug
Traceback (most recent call last):
  File "/home/pi/Mycodo/mycodo/mycodo_daemon.py", line 46, in <module>
    from mycodo.controller_input import InputController
  File "/home/pi/Mycodo/mycodo/controller_input.py", line 26, in <module>
    import fasteners
ImportError: No module named fasteners
pi@raspberrypi:/var/log/mycodo $ ls -l /var/Mycodo-backups

kizniche, 2017-11-19:
fasteners is included in the [requirements.txt](https://github.com/kizniche/Mycodo/blob/3828c25f7aad419157d5fa52fb45dd0e556b869f/install/requirements.txt#L10) and should have been upgraded. It sounds like you have corrupted files. Use the following command to reinstall all python requirements. This tip is found in the [Install Notes](https://github.com/kizniche/Mycodo#install-notes):

```
sudo ~/Mycodo/env/bin/pip install -r ~/Mycodo/install/requirements.txt --upgrade --force-reinstall --no-deps
```

reymart081678, 2017-11-19:
retried and all are okay now. running 5.4.3

kizniche, 2017-11-19:
Great! Thanks for testing.

closed 2017-11-19 by kizniche