kizniche/Mycodo #324: DHT22 and PWM inputs issues on 5.4.2

pukkita, 2017-11-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.2
 
#### Problem Description
Some inputs are not working: DHT22, PWM / RPM input.
 
[image]

### Errors
#### `mycodo.log`

```
2017-11-19 10:21:17,737 - mycodo.daemon - INFO - Mycodo daemon v5.4.3 started in 2.134 seconds
2017-11-19 10:21:17,744 - mycodo.daemon - INFO - 49.11 MB RAM in use
2017-11-19 10:21:22,939 - mycodo.inputs.signal_pwm - ERROR - SignalPWMInput raised an exception when taking a reading: unsupported operand type(s) for +: 'NoneType' and 'float'
2017-11-19 10:21:23,812 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
2017-11-19 10:21:31,954 - mycodo.inputs.signal_pwm - ERROR - SignalPWMInput raised an exception when taking a reading: 'NoneType' object has no attribute 'send'
2017-11-19 10:23:24,092 - mycodo.influxdb - ERROR - No measurement available.
2017-11-19 10:23:24,157 - mycodo.influxdb - ERROR - No measurement available.
2017-11-19 10:23:24,195 - mycodo.influxdb - ERROR - No measurement available.
2017-11-19 10:23:29,188 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
2017-11-19 10:23:35,953 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
2017-11-19 10:23:35,954 - mycodo.input_6 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2017-11-19 10:23:51,029 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable

```
---

#### `upgrade.log`

```
[2017-11-19 09:50:48] Successfully installed adafruit-bme280 argparse-1.4.0 tsl2591
[2017-11-19 09:50:49] 
[2017-11-19 09:50:49] #### Ensuring compatible version of influxdb is installed ####
[2017-11-19 09:50:49] Correct version of InfluxDB currently installed
[2017-11-19 09:50:49] 
[2017-11-19 09:50:49] #### Upgrading database with alembic
[2017-11-19 09:50:50] 
[2017-11-19 09:50:50] #### Enabling mycodo startup script
[2017-11-19 09:50:51] 
[2017-11-19 09:50:51] #### Compiling Translations
[2017-11-19 09:50:53] 
[2017-11-19 09:50:53] #### Updating Mycodo crontab entry
[2017-11-19 09:50:53] Removing crontab entry ...
[2017-11-19 09:50:53] Appending to crontab:
[2017-11-19 09:50:53] -----------------------------------------------
[2017-11-19 09:50:53] @reboot /usr/local/bin/pigpiod -s 1 &
[2017-11-19 09:50:54] -----------------------------------------------
[2017-11-19 09:50:54] To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh mycodo --remove
[2017-11-19 09:50:54] 
[2017-11-19 09:50:54] 
[2017-11-19 09:50:54] #### Updating Mycodo restart monitor crontab entry
[2017-11-19 09:50:54] Removing crontab entry ...
[2017-11-19 09:50:54] Appending to crontab:
[2017-11-19 09:50:54] -----------------------------------------------
[2017-11-19 09:50:54] @reboot /var/www/mycodo/mycodo/scripts/restart_daemon.py --continuouscheck &
[2017-11-19 09:50:54] -----------------------------------------------
[2017-11-19 09:50:54] To remove previously added crontab entry, run: /home/pi/Mycodo/install/crontab.sh restart_daemon --remove
[2017-11-19 09:50:54] 
[2017-11-19 09:50:54] 
[2017-11-19 09:50:54] #### Compiling mycodo_wrapper
[2017-11-19 09:50:54] 
[2017-11-19 09:50:54] #### Creating users and directories
[2017-11-19 09:50:54] The user `mycodo' is already a member of `gpio'.
[2017-11-19 09:50:54] The user `mycodo' is already a member of `adm'.
[2017-11-19 09:50:55] The user `mycodo' is already a member of `video'.
[2017-11-19 09:50:55] The user `mycodo' is already a member of `dialout'.
[2017-11-19 09:50:55] 
[2017-11-19 09:50:55] #### Setting permissions
[2017-11-19 09:50:56] 
[... 18 lines cut ...]
[Sun Nov 19 09:48:35.683710 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,683] ERROR in __init__: Failed to save <Role(id=3, name='Monitor')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET name=? WHERE roles.id = ?'] [parameters: ('Monitor', 3)]
[Sun Nov 19 09:48:35.709746 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,709] ERROR in __init__: Failed to save <Role(id=3, name='Monitor')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_camera=? WHERE roles.id = ?'] [parameters: (1, 3)]
[Sun Nov 19 09:48:35.735772 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,735] ERROR in __init__: Failed to save <Role(id=3, name='Monitor')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_settings=? WHERE roles.id = ?'] [parameters: (1, 3)]
[Sun Nov 19 09:48:35.761629 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,761] ERROR in __init__: Failed to save <Role(id=3, name='Monitor')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET edit_users=? WHERE roles.id = ?'] [parameters: (0, 3)]
[Sun Nov 19 09:48:35.788830 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,788] ERROR in __init__: Failed to save <Role(id=4, name='Guest')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET name=? WHERE roles.id = ?'] [parameters: ('Guest', 4)]
[Sun Nov 19 09:48:35.816290 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,815] ERROR in __init__: Failed to save <Role(id=4, name='Guest')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_camera=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Sun Nov 19 09:48:35.843670 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,843] ERROR in __init__: Failed to save <Role(id=4, name='Guest')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET view_settings=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Sun Nov 19 09:48:35.869685 2017] [wsgi:error] [pid 10021:tid 1964295216] [2017-11-19 09:48:35,869] ERROR in __init__: Failed to save <Role(id=4, name='Guest')> due to error: (sqlite3.OperationalError) attempt to write a readonly database [SQL: u'UPDATE roles SET edit_users=? WHERE roles.id = ?'] [parameters: (0, 4)]
[Sun Nov 19 09:48:44.695946 2017] [wsgi:error] [pid 10021:tid 1929376816] [remote 192.168.10.150:61047] 2017-11-19 09:48:44,695 URL for 'daemon_active' raised and error: Connection refused. Is the daemon running?
useradd: user 'mycodo' already exists
rm: cannot remove '/home/pi/Mycodo/databases/statistics.csv': No such file or directory
pigpiod: no process found
Cloning into '/home/pi/Mycodo/install/wiringPi'...

```

### Steps to Reproduce the issue:
How can this issue be reproduced? Upgrade to mycodo 5.4.2
 
### Additional Notes
I checked reading the DHT22 manually, it works fine.

I tried deactivating, saving then reactivating the input to no avail.

I tried deleting, then re-adding the input, no difference. Only input that us being readcorrectly is a DS18B20.

pukkita, 2017-11-19:
Wow! Multi-Screen LCD is implemented! 👍 

Cannot wait to fix inputs!

kizniche, 2017-11-19:
Edit each of the inputs that are giving you errors (in Mycodo/mycodo/inputs/ edit dht22.py, signal_pwm.py, and signal_rpm.py) and change all instances of "logger.error" to "logger.exception", save, and restart the daemon to get a more detailed error. For example, in dht22.py, edit line 221 so that section changes from:

```python
        except Exception as e:
            self.logger.error(
                "{cls} raised an exception when taking a reading: "
                "{err}".format(cls=type(self).__name__, err=e))
```

to

```python
        except Exception as e:
            self.logger.exception(
                "{cls} raised an exception when taking a reading: "
                "{err}".format(cls=type(self).__name__, err=e))
```

However, so this for all instances of "logger.error" in each file.

kizniche, 2017-11-19:
I'll conduct my own tests, but that will speed up the diagnostics.

kizniche, 2017-11-19:
I just tested PWM, and it is working for me.

How I tested:
Set up output on PWM hardware pin 12 (BCM), 2000 Hz.
Set up PWM input on pin 16 (BCM).
Connected a wire with a 10k Ohm resistor in-line between pins 16 and 12.
Set PWM output to 50% duty cycle.
Activate PWM Input.

kizniche, 2017-11-19:
Okay, spoke too soon. I acquired a measurement, then it errored.

kizniche, 2017-11-19:
I was able to isolate the PWM Input issue, but I could not for the RPM Input. The RPM Input was working fine for me. Perhaps the error in the PWM Input caused an issue with pigpio (which the RPM Input also uses) that affected the RPM Input.

kizniche, 2017-11-19:
Okay. I just got the RPM error to be induced. Investigating now.

kizniche, 2017-11-19:
Fixed RPM, now only working on DHT22.

leoneo, 2017-11-19:
I have change dht22.py file like you want

daemon log : 

2017-11-19 19:44:58,259 - mycodo.lcd_1 - INFO - Activated in 1139.8 ms
2017-11-19 19:44:58,261 - mycodo.daemon - INFO - All activated LCD controllers started
2017-11-19 19:44:58,263 - mycodo.daemon - INFO - Mycodo daemon v5.4.3 started in 13.808 seconds
2017-11-19 19:44:58,274 - mycodo.daemon - INFO - 48.54 MB RAM in use
2017-11-19 19:45:01,448 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/dht22.py", line 217, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'NoneType' object is not iterable
2017-11-19 19:45:15,643 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/dht22.py", line 217, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'NoneType' object is not iterable
2017-11-19 19:45:30,730 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/dht22.py", line 217, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'NoneType' object is not iterable
2017-11-19 19:45:30,732 - mycodo.input_3 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2017-11-19 19:45:45,748 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/dht22.py", line 217, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'NoneType' object is not iterable
2017-11-19 19:46:00,733 - mycodo.inputs.dht22 - ERROR - DHT22Sensor raised an exception when taking a reading: 'NoneType' object is not iterable
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/dht22.py", line 217, in read
    self._dew_point, self._humidity, self._temperature = self.get_measurement()
TypeError: 'NoneType' object is not iterable

kizniche, 2017-11-19:
Thanks. I'm hooking my DHT22 up right now so I'll be able to test my code edits.

kizniche, 2017-11-19:
I think I finally got all the inputs working. I also found an issue where Duty Cycle, Frequency, Pulse Width, and RPM were never added to Graphs, for both Inputs and PID setpoints.

[image]

closed 2017-11-19 by kizniche

kizniche, 2017-11-20:
It seems as though the DHT22 input is more stable than it was before. I haven't had to rely on a transistor to power cycle it once all day. I wonder if my refactoring fixed that intermittent issue that would allow it to become unresponsive. I'll report back after a day or two of it running continually.

kizniche, 2017-11-20:
11 hours continually running and not one missed measurement!