kizniche/Mycodo #325: Error on capture still image

cohe4ko, 2017-11-20:
## Mycodo Issue Report:
  5.4.4
 
#### Problem Description

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 135, in page_camera
    camera_record('photo', mod_camera)
  File "/var/www/mycodo/mycodo/devices/camera.py", line 115, in camera_record
    if not img_orig:
ValueError: The truth value of an array with more than one element is ambiguous. Use a.any() or a.all()

kizniche, 2017-11-20:
Try changing lines 110 to 117 of camera.py to this:

```python
        if record_type in ['photo', 'timelapse']:
            edited = False
            retval, img_orig = cap.read()
            cap.release()

            if not retval:
                logger.error("Could not access camera")
                return
```

restart apache with ```sudo service apache2 restart``` and see if that works.

kizniche, 2017-11-20:
Another potential solution to test is:

```python
        if record_type in ['photo', 'timelapse']:
            edited = False
            _, img_orig = cap.read()
            cap.release()

            if any(img_orig):
                logger.error("Could not access camera")
                return
```

Though I'm not sure why img_orig would be an array. In all my tests, it is a single image. I've never experienced your error before.

cohe4ko, 2017-12-06:
if record_type in ['photo', 'timelapse']:
            edited = False
            retval, img_orig = cap.read()
            cap.release()

            if not retval:
                logger.error("Could not access camera")
                return 
this one works.

kizniche, 2017-12-06:
It actually captured an image?

cohe4ko, 2017-12-06:
Yes. Everything is ok

kizniche, 2017-12-06:
Thanks for testing. That's extremely strange, because my camera still ceases to work with opencv (which at one time it did work). I'll have to revisit my investigation.

closed 2017-12-20 by kizniche