kizniche/Mycodo #327: [suggestion]  Earth Periodic Terms Emulation

pukkita, 2017-11-22:
It was already suggested on the ANCHOR ; it will be great to have a sort of PID/Method to emulate Earth Periodic conditions, Like sun - daytime periods, moon phases, tides, seasons, etc... 

It could be used to recreate conditions close to nature, e.g. on day 0 we set it to "initiate program" with a selected (virtual or real) climate profile, to start at two weeks before spring, and the system will automatically control and change setpoints for daylight, moonlight, temperatures along the day/night, humidity, etc.

This could be a sort of "master control" which would be dynamically changing setpoints, keeping track of the day it's at, etc.

Just a crazy idea :)

kizniche, 2017-11-22:
This is a big feature that I've thought a lot about how to implement. I've already [converted a sunrise/sunset calculator to python](https://github.com/kizniche/Mycodo/blob/3a742330b8ef9c5b262e98ec951326aa0798153b/mycodo/utils/sunriseset.py). The issue comes that once I start designing a framework, it becomes so large and complicated I put it off until I have a long period to work on it. I haven't begun to look for packages that could do a lot of the leg work, but that could be fruitful enough to get this feature rolling again.

pukkita, 2017-11-24:
> I haven't begun to look for packages that could do a lot of the leg work, but that could be fruitful enough to get this feature rolling again.

Will try to help in that area looking for ready software components :)

kizniche, 2017-12-06:
What features do you desire in a sunrise/set method?

What comes to my mind is being able to set:

1. From what minimum (before sunrise) to what maximum (after sunrise). (i.e min/max PWN duty cycle)
2. At what rate to rise/set, such as "rise at 2% per minute until hit max", "lower at 10% per minute until hit min"
3. Sunrise/sunset offset. For example, start rising 20 minutes before sunrise (to mimic twilight).
4. Time offset. That is, if it's January 1, and you want it to mimic the sunrise/set of March 1, then you would need an offset of 60+ days.

This would be a basic implementation that would mimic the sunrise and sunset of a specific latitude/longitude and date (adjusted to the offset).

Comments/ideas?

pukkita, 2017-12-07:
Hello Kyle!

Sorry for not coming back with the potential components, I found some python modules for lunar phase calculation, but didn't have the time to analyze if they were zombie projects or not, will get back on this.

Re: sunrise/sunset: your approach looks nice and dandy!

closed 2020-07-23 by kizniche