kizniche/Mycodo #328: Am2315 problem

RynFlutsch, 2017-11-22:
## Mycodo Issue Report:
  - Specific Mycodo Version:
Upgraded to
5.4.4
 since than there is a problem with the sensor.

[image]
[image]

Timers can be set with seconds ?
Is it possible to take a midpoint of 2-X sensors (same type) as condition?

kizniche, 2017-11-22:
It looks like your sensor is returning all failed CRC checks. Have you physically unplugged and plugged your sensor back in, then reactivated it in Mycodo?

Can you elaborate of the 2 sensor idea?

kizniche, 2017-11-22:
I was able to replicate the issue and will be looking into it.

kizniche, 2017-11-22:
I just pushed an update. I'll make a new release for users to upgrade to soon. Thanks for the bug report.

RynFlutsch, 2017-11-22:
If Mean of sensor 1-5 is X than turn on Relais X.
Or Get Mean of Sensor 1-5 for PID Set Value

kizniche, 2017-11-30:
I have an idea. Let me know if this is a suitable solution.

A new Input type, "Multiple Input Manipulation" (or another descriptive name), that will periodically take the latest stored value from several activated inputs and do math on those values and store the resulting value.

For instance, you can select two inputs and make it average them. Then this new Input can be used with a PID or conditional to carry out actions.

This could also be a way I can easily implement Sensor Verification. That is, select two sensors for the new Input, and if the measurements deviate more than 10% (or whatever value is set), then the measurement is not saved (no saving means PIDs will stop operating and Conditionals will not trigger). If the sensor values are within 10% of each other, then it will save the average of them (and allow the PID or Conditionals to operate).

kizniche, 2017-11-30:
You could also do some interesting things, such as create 2 Multiple Input Manipulation Inputs, each one will average two measurements (such as from two humidity sensors at the back of a chamber, and two humidity sensors at the front of a chamber). Then, another Multiple Input Manipulation Input can be created that will employ sensor verification and hold those two new Inputs to being equal or less than 10% deviation. If a deviation occurs, it won't allow the PID or Conditional to operate (and send an email notification for good measure that something is wrong).

RynFlutsch, 2017-12-01:
Sounds great but i understand this right, its only an idea, not implemented yet ;) ?
Yeah iam Looking for DIY Greenhouse controlling.
i.e. multiple AM2315 and Chirp (I2C Soil moisture sensor )
I am at the beginning, i will start soon with the PID stuff.
For the first steps i will operate with 2 heatmatts+ pump+ maybe 12V fans @ aquarium.

Btw. Create Timespan Timer with off - setting at the first creation is buggy. After creation it's on and is needed to switch to off again.
Maybe its a handy feature to dublicate Timers, or put mutiple Outputs in one Timer?

kizniche, 2017-12-06:
I just implemented the Math Controller feature with the latest release, yesterday. It only includes one module so far, the ability to average multiple input measurements.

As for affecting multiple outputs, you can create two new Output Conditionals, one will be:

```
If Output A turns ON, Turn Output B ON
```

and the other will be

```
If Output A turns OFF, Turn Output B OFF
```

This is how I link Outputs.

kizniche, 2017-12-06:
Also, I found that using the software I2C has alleviated all my AM2315 issues. I haven't had it skip a beat since doing so. You can read about how to do this [here](https://github.com/kizniche/Mycodo/blob/8738590ea5e8b34d10e4a0d4c6c9a480885916a8/mycodo-manual.md#notes).

closed 2017-12-13 by kizniche