kizniche/Mycodo #329: Pi Zero Install Error

jacopienaar, 2017-11-23:
## Mycodo Issue Report:
  - Specific Mycodo Version: Latest
 
#### Problem Description
Install aborted during installation of latest version of Mycodo on Pi Zero W. I do not know enough to resolve this issue myself..

 I have Mycodo running perfectly on Pi3 and tried swopping that SD card to Pi Zero and it would start up. Perhaps a possibility with a  compatibility issue with Pi Zero?
 
### Errors

#Start of error . . . . 

  Running setup.py install for pi-switch: started
    Running setup.py install for pi-switch: still running...
    Running setup.py install for pi-switch: finished with status 'error'
    Complete output from command /home/pi/Mycodo/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uHk2CM/pi-switch/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-8uWhn_-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python2.7/pi-switch:
    running install
    running build
    running build_py
    creating build
    creating build/lib.linux-armv6l-2.7
    creating build/lib.linux-armv6l-2.7/pi_switch
    copying pi_switch/__init__.py -> build/lib.linux-armv6l-2.7/pi_switch
    running build_ext
    building '_pi_switch' extension
    creating build/temp.linux-armv6l-2.7
    creating build/temp.linux-armv6l-2.7/pi_switch
    creating build/temp.linux-armv6l-2.7/pi_switch/wrapper
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -fno-strict-aliasing -D_FORTIFY_SOURCE=2 -g -fstack-protector-strong -Wformat -Werror=format-security -fPIC -I/usr/include/python2.7 -c pi_switch/wrapper/pi_switch.cpp -o build/temp.linux-armv6l-2.7/pi_switch/wrapper/pi_switch.o -Wno-write-strings
    cc1plus: warning: command line option ‘-Wstrict-prototypes’ is valid for C/ObjC but not for C++
    arm-linux-gnueabihf-gcc: internal compiler error: Killed (program cc1plus)
    Please submit a full bug report,
    with preprocessed source if appropriate.
    See <file:///usr/share/doc/gcc-4.9/README.Bugs> for instructions.
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 4
    
    ----------------------------------------
Command "/home/pi/Mycodo/env/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-uHk2CM/pi-switch/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-8uWhn_-record/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/Mycodo/env/include/site/python2.7/pi-switch" failed with error code 1 in /tmp/pip-build-uHk2CM/pi-switch/

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Tried to install it twice - same issue occurred 
 
### Additional Notes
Thanks for developing this amazing little program! I grow oyster mushrooms at home and this has enabled me to control my environmentals and resulted in better and bigger flushes!

kizniche, 2017-11-23:
Have you tried using a new SD card? Not just swapping with one Raspbian/Mycodo is already installed on, but a new card and a fresh install of Raspbian. I suspect your SD card may be damaged, causing file corruption.

Oysters are delicious. Do you have any photos you can share?

kizniche, 2017-11-29:
Mycodo is comprised of a backend (daemon) and frontend (web user interface). The daemon is always running in the background, however if no controllers (inputs/outputs/PIDs/etc.) have been added, it's virtually doing nothing.

Use the ```top``` command to see what process(es) are utilizing your CPU.

kizniche, 2017-11-29:
Also, I don't have a Pi Zero to test/optimize Mycodo, so I can't do my own testing with this to discover an issue.

kizniche, 2017-12-06:
Have you had a chance to run ```top``` to determine what the CPU hogs are? I'd be interested to see the output.

jacopienaar, 2017-12-07:
The SD card used is brand new but I have read that people with similar CPU usage issues on Pi Zero have had it resolved by using a different card. I will only be able to try this in January as I am about to leave for the holidays. Then I will see if I can install Mycodo directly onto the Zero and also if this resolves the CPU usage issue

The top two commands sucking CPU life is both Mycodo - I'll attach a screenshot in the next post. 

There is unfortunately not much to show at the moment as my cakes are all spent and I am packing all away until I get back. I've attached some pics of a few weeks back - species is Pleurotus Pulmonarius
[image]
[image]

jacopienaar, 2017-12-07:
[image]

kizniche, 2017-12-07:
Those are some nice mushrooms!

Your screenshot has the first line of ```top``` cut off that has the CPU load, but from the CPU usage, it appears normal... Only a little over 50% CPU usage.

kizniche, 2017-12-08:
Using a GUI will have a considerable load on the OS, though. For a Pi Zero, I would suggest turning the GUI off and running it headless. All but one of my 4 Pis run headless (Pi 1, 2, and 3), with the other Pi 3 onlt running X because it has a 7-inch touchscreen attached to the front of the [electrical box](https://hackaday.io/project/11997-mycodo-environmental-regulation-system/log/42201-power-output-and-mounting-options) all the components are stored in.

fry42, 2018-03-09:
Hi!
I recently installed on a zero w and thought I'd share my findings, maybe it helps:
- Installed raspbian stretch lite Nov17 (=no GUI)
- updated to the most recent packages (apt-get update/upgrade)
- installed Mycodo
The install script returned a success after 1min30, but not much was installed. Investigating the logs showed that one of the sources servers had a hickup and reported a internal server error.
Running the script a second time resulted in a working installation, after about 30min.

So the only bug would be that the install script didn't catch the failed installation attempt.
The CPU load was around 50% and the Interface felt a little slow & sluggish.
My conclusion was that the 1GHz single core CPU wasn't quite up to the task. Switching to a quad-core 1,2GHz Pi3 improved things :)

kizniche, 2018-03-09:
Thanks for the feedback. What part did the install fail on? Could you paste the relevant part of the setup.log?

fry42, 2018-03-09:
Sorry, I don't have the logs anymore...
It was right at the beginning, when your setup script calls for updating the sources lists.

If it helps I could try the same setup again, my zero is currently not in use...

kizniche, 2018-03-09:
Don't worry about it. Install issues related to internet connectivity are rare. Most of the time the issue can be solved by rerunning the setup command. I put a good effort into making the setup.sh able to be executed multiple times without issue (where before this would be problematic). If I see more github issues appear, I'll investigate it.

kizniche, 2018-03-09:
BTW, there's a log generated when the setup.sh is ran, located in ~/Mycodo/install/setup.log

zsole2, 2018-03-10:
With the new minimal install, my Pi Zeros run quite great, the idle time is up to 80-90% before adding anything. Then as the functionality increases, the load and CPU usage increases proportionally. My latest hunch is that the most strain is coming from data presentation (Dashboard), rather than the effector functionality (Inputs/Outputs/PIDs), but will need to test this hypothesis more systematically. If it's true, a push for the CLI interface (#17) would be in order.

closed 2018-04-06 by kizniche