kizniche/Mycodo #330: BME280 Sensor error

viktorarmani55, 2017-11-24:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.7
 
#### Problem Description
Please list:

  - I disconnected the BME Sensors for a new greenwall experiment
  - I reconnected the sensors and the BME280 sensors don't work, however the lux sensor works
  - I've checked the SDA and SCL connections, and they seem to be fine.
  - I tried adding the BME280 sensors one by one, and the same error occurred (input/output error)
 
### Errors
  - Input/output error.  Checked connections and they are fine.
  - I'll send the pictures via gmail

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Connect the BME280 sensor to the multiplexer power, ground, SDA, SCL
 2. In the settings, I2C address 0x77, Multiplexer(Mx) I2C address 0x70, and select the correct Mx channel
 3. Input/output error. 
 
### Additional Notes
The connected sensors were working fine, however, I've noticed them recently not working.  I'll send pictures of the problem.

kizniche, 2017-11-24:
Were the sensors working with this version, or were they working in a previous version and after the upgrade stopped working?

If so, what version did you upgrade from?

kizniche, 2017-11-24:
Can you run the daemon in debug mode with one sensor activated and look for any errors?

```
sudo service mycodo stop
sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_daemon.py -d
```

The daemon can then be stopped with the client terminate command:

```sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_client.py -t```

viktorarmani55, 2017-11-24:
Hi Kyle,

I've placed a new SD card with the latest version of the Mycodo i.e. 5.4.7,
as I thought it would be best for a new experiment (done this several days
ago).  It was working with this version, however, I didn't try all the
channels, only 3 or 4.
My old SD cards have been replaced with other projects, so I won't be able
to go back to an older version.  Even the older version seemed to
be working, however, only trialed 3-4 sensors, not all the multiplexer
channels.

Regards,
Vik

kizniche, 2017-11-24:
If you need a working system right away, you can always download an earlier version from the [release page](https://github.com/kizniche/Mycodo/releases).

As for the issue, I'm unsure if it's a multiplexer or sensor issue, so the debug log would help a lot. I'm away (Thanksgiving) from home and my development Pi so I can't test myself.

kizniche, 2017-11-24:
So with just one sensor it is acquiring measurements? What happens when you enable two sensors one different multiplexer channels and run the daemon in debug mode for a few minutes?

kizniche, 2017-11-24:
I'm off to sleep. I'll pick up this issue again tomorrow.

viktorarmani55, 2017-11-24:
Hi Kyle,

What I've noticed is that the Mx channels do not correspond to the
multiplexer numbers e.g. SD1 SC1, SD2 SC2.  They seemed to be jumbled up.
So to fix the problem, I enable 1 sensors, then one by one change the Mx
channel until an error message doesn't pop up and it is green
(successful).  I've managed to do this for 5 sensors now, however, I'm
having an issue with my last two sensors.

Regards,
Vik

kizniche, 2017-11-24:
Are you using the TCA9548A breakout board made by Adafruit?

viktorarmani55, 2017-11-24:
Hi Kyle,

Yes, it's the tca9548a from adafruit.

Regards,
Vik

viktorarmani55, 2017-11-24:
BTW, happy thanksgiving.

kizniche, 2017-11-28:
Thanks. Happy Thanksgiving to you as well.

I haven't had a chance to test my multiplexer. I'll be getting home today but might not get to test for another day or so. So, your multiplexer appears to be working atypically and a few sensors don't work... Do they work when you connect them straight to the I2C lines, without the multiplexer?

viktorarmani55, 2017-12-01:
Hi Kyle,

I've tested each sensor individually, and they work without any issues.
I've attached an image of a sample reading.

Regards,
Vik

viktorarmani55, 2017-12-05:
Hi Kyle,
Have you had a chance to test out the multiplexer and bme280 sensors. 
Regards,
Vik

kizniche, 2017-12-05:
I just tested my multiplexer and I'm experiencing the same disjointed channels as you. Here is what I was able to determine using my AM2315 (I don't have a BME280), channel-wise:

```
Mux Channel		Mycodo Channel
0			1
1			3
2			4
3			8
4			X
5			X
6			X
7			X
```

X = Could not detect the AM2315 on any channel 0 - 8.

I was testing with the command below to change channels (bus 3, channel 4 below) with the [newest test script that includes the bus option](https://github.com/kizniche/Mycodo/blob/5c8680d7cafd99706fa1657816e705d79e6bad02/mycodo/tests/manual_tests/test_i2c_TCA9548A.py):

```~/Mycodo/mycodo/tests/manual_tests/test_i2c_TCA9548A.py -b 3 -a 70 -c 4```

Then, see if the sensor appeared with ```i2cdetect -y 3``` like below:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- 5c -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: 70 -- -- -- -- -- -- --    
```

Do these "new" channels align with what you found?

kizniche, 2017-12-05:
I'm working on a fix. I should have more to report soon.

kizniche, 2017-12-05:
I found a fix that corrects all the multiplexer channel issues. I have no idea how this happened, but my apologies. I'll have a fix pushed and a release made soon.

Thanks for the bug report.

kizniche, 2017-12-05:
I just released [v5.4.12](https://github.com/kizniche/Mycodo/releases/tag/v5.4.12). Let me know if your issues are resolved after updating. Thanks.

closed 2017-12-05 by kizniche

kizniche, 2017-12-05:
I had to pull back the release because I found a database issue. I'll try to have a fix and new release out soon.

kizniche, 2017-12-05:
I finally figured out the database upgrade issue during Mycodo upgrades, though if you have issues with your database after the upgrade, you may have to delete your database and create a new one (by visiting the web UI after deleting it) to be clear if the issue. I'm not even sure how the issue arose, but my testing has found a fix. So, I released the latest code as v5.4.14. Good luck with the upgrade and let me know if it fixes your multiplexer issue.

viktorarmani55, 2017-12-05:
Thanks Kyle. I'll check it soon, and let you know.

viktorarmani55, 2017-12-06:
All is working fine.  Thanks for your help.  Much appreciated.

kizniche, 2017-12-06:
you're welcome. Thanks for the bug report. It's hard to test everything myself, moreso for the sensors I don't own.