kizniche/Mycodo #331: Installation problems, "Could not connect to http://localhost"

pewi81, 2017-11-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: Latest
 
#### Problem Description
Please list: See below
 
#### Restarting the Mycodo web server
#### Creating Mycodo database if it doesn't exist
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...

  - what were you trying to do
Install the latest Mycodo 
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Install Mycodo following the instructions
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

pewi81, 2017-11-26:
Problem solved itself after several tries!

closed 2017-11-26 by pewi81

kizniche, 2017-11-26:
This is an issue with the web server not starting fast, so I added 5 repeats with 60-second pauses.

kizniche, 2017-11-26:
I'll change it to read:

```Waiting 60 seconds then trying again (up to 5 times)...```

so it's clearer that it's not stuck in an infinite loop.