kizniche/Mycodo #332: 500 Internal Server Error

pewi81, 2017-11-26:
## Mycodo Issue Report:
  - Specific Mycodo Version: Latest
 
#### Problem Description
Please list: 500 Internal Server Error as I try to reach the web interface
  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)
`pi@raspberrypi:~/Mycodo/install $ sudo apt-get install -y jq
Reading package lists... Done
Building dependency tree       
Reading state information... Done
jq is already the newest version (1.5+dfsg-1.3).
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 129 not upgraded.
pi@raspberrypi:~/Mycodo/install $ cd ~
pi@raspberrypi:~ $ curl -s https://api.github.com/repos/kizniche/Mycodo/releases/latest | \
> jq -r '.tarball_url' | wget -i - -O mycodo-latest.tar.gz
--2017-11-26 20:51:34--  https://api.github.com/repos/kizniche/Mycodo/tarball/v5.4.8
Resolving api.github.com (api.github.com)... 192.30.253.116, 192.30.253.117
Connecting to api.github.com (api.github.com)|192.30.253.116|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/kizniche/Mycodo/legacy.tar.gz/v5.4.8 [following]
--2017-11-26 20:51:34--  https://codeload.github.com/kizniche/Mycodo/legacy.tar.gz/v5.4.8
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7131693 (6.8M) [application/x-gzip]
Saving to: ‘mycodo-latest.tar.gz’

mycodo-latest.tar.gz             100%[=========================================================>]   6.80M  1.03MB/s    in 6.6s    

2017-11-26 20:51:42 (1.04 MB/s) - ‘mycodo-latest.tar.gz’ saved [7131693/7131693]

FINISHED --2017-11-26 20:51:42--
Total wall clock time: 8.5s
Downloaded: 1 files, 6.8M in 6.6s (1.04 MB/s)
pi@raspberrypi:~ $ mkdir Mycodo
mkdir: cannot create directory ‘Mycodo’: File exists
pi@raspberrypi:~ $ tar xzf mycodo-latest.tar.gz -C Mycodo --strip-components=1
tar: .github: Cannot utime: Operation not permitted
tar: .github: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: databases/alembic/versions: Cannot utime: Operation not permitted
tar: databases/alembic/versions: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: databases/alembic: Cannot utime: Operation not permitted
tar: databases/alembic: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: databases: Cannot utime: Operation not permitted
tar: databases: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: install: Cannot utime: Operation not permitted
tar: install: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: manual_images: Cannot utime: Operation not permitted
tar: manual_images: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/databases/models: Cannot utime: Operation not permitted
tar: mycodo/databases/models: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/databases: Cannot utime: Operation not permitted
tar: mycodo/databases: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/devices: Cannot utime: Operation not permitted
tar: mycodo/devices: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/inputs: Cannot utime: Operation not permitted
tar: mycodo/inputs: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/mycodo_flask/camera: Cannot utime: Operation not permitted
tar: mycodo/mycodo_flask/camera: Cannot change mode to rwxr-xr-x: Operation not permitted
tar: mycodo/mycodo_flask/forms: Cannot utime: Operation not permitted
tar: mycodo/mycodo_flask/forms: Cannot change mode to rwxr-xr-x: Operation not permitted
[... 587 lines cut ...]
#### Could not connect to http://localhost. Waiting 60 seconds then trying again...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Sun 26 Nov 21:08:43 CET 2017

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.1.104/, or whatever your Pi's
IP address is, to create an admin user and log in.`

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

pewi81, 2017-11-26:
Code looks terrible but I tried to add it as code without success!

kizniche, 2017-11-26:
[restart flask in debug mode](https://github.com/kizniche/Mycodo/wiki/Diagnosing-Issues#debug-mode-1) then try to visit the web interface to see the specific error.

I installed the latest Mycodo on the latest Raspbian on a Pi 2 an hour ago, without issues, so I think it is a pip package that didn't compile correctly, bad filesystem/corrupt files, or something.

kizniche, 2017-11-26:
Also, multi-line code should be encased in three backtics (\`\`\`)

kizniche, 2017-11-26:
I see you already had the Mycodo directory, and a bunch of errors after that:

```
pi@raspberrypi:~ $ mkdir Mycodo
mkdir: cannot create directory ‘Mycodo’: File exists
...
```

pewi81, 2017-11-26:
The debug thing didn't work!

kizniche, 2017-11-26:
Please elaborate

pewi81, 2017-11-26:
The commands returns this in the terminal:

```pi@raspberrypi:~ $ sudo /etc/init.d/apache2 stop
[ ok ] Stopping apache2 (via systemctl): apache2.service.
pi@raspberrypi:~ $ sudo ~/Mycodo/env/bin/python ~/Mycodo/mycodo/mycodo_flask/start_flask_ui.py --debug
/home/pi/Mycodo/env/bin/python: can't open file '/home/pi/Mycodo/mycodo/mycodo_flask/start_flask_ui.py': [Errno 2] No such file or directory```

kizniche, 2017-11-26:
See my previous comment about you already having a Mycodo directory and the errors after that. Your issues probably stem from that.

kizniche, 2017-11-26:
Also, there was a typo in the command. I've updated the wiki to reflect the proper command.

pewi81, 2017-11-27:
Since I couldn't figure a way round this I erased the Mycodo dir and made a fresh install!

closed 2017-11-27 by pewi81