kizniche/Mycodo #336: wireless / remote sensors / GPIO pins

eap904, 2017-11-30:
Apologies if this is not the correct place to pose this.

Trying to find more information about wireless monitoring / control. 

I see there is an option for remote GPIO, so it would be nice to get pi zero, esp8266 etc and have it in the grow room separate from the main pi 3, but have it's sensors and all others used on the same instance of mycodo for ease of accessing data. Low enough powered so that it can run off of battery packs if needed. 

Also, I noticed in the relays, there is now a wireless relay option... does this work with the ubiquitous 'smart plugs' which are controllable via wifi? It looks to need a 433Mhz transmitter of some type. 

Thank you

kizniche, 2017-11-30:
There isn't currently the ability to obtain wireless measurements, but wireless relays can be used with the 433MHz protocols. These are the typical 433MHz wireless switching outlets you can buy cheap on Amazon or eBay.

The easiest way to add a remote sensor is to get some wireless protocol working with an ATMega, for instance, that transmits and have your Pi use a receiver that will return a measurement when a script is executed on the Pi. Mycodo can then use the "Linux Command" Input to execute this script and store/use the returned value.

There are so many ways to create a wireless sensor setup that I haven't really decided on one I want to integrate into the system yet.

Suggestions are welcome.

eap904, 2017-12-02:
yes, the options are dizzying.. so 'Linux Command' is an option when adding an input? 

Have you looked into remote GPIO? Not sure what mycodo currently uses to communicate with the pins... GPIO Zero?

https://www.raspberrypi.org/magpi/remote-control-gpio-raspberry-pi-gpio-zero/

kizniche, 2017-12-02:
Yes, it's currently an option that can be used. I have not looked to closely at remote GPIO. I basically build everything I need unless a well-formed library already exists. I haven't needed anything beyond RPi.GPIO yet for manipulating the GPIOs.

Theoi-Meteoroi, 2017-12-26:
I've looked at this quite a bit - wanting to have sensor modules without a interconnect rats nest.  The sensor would typically need a micro-controller anyway and there are many with 2.4Ghz wireless radios, the ESP-8266 comes to mind. Lousy for battery power but has a strong radio.  Anyway, if you host the readings on one of these simple web servers configured as AP, you could fetch the reading with 'curl' in a linux command with Mycodo. I've implemented this before with another project.  Another choice could be mqtt - but you would need to write the glue code for the mycodo side to pass your mqtt updates to a linux cli output.   

The deal is most sensors are fairly sensitive to noise, of all things . Long i2c cables and that sort will not work reliably and will induce noise or bouncing ground levels.  If you decouple the sensor electronics from the rather noisy Raspi power supply, it gets much better except for connectivity to the raspberry pi.  There are some projects that might provide some of what you request - Firmata is a sort of "remote GPIO" through an attached arduino - that was the first thing I thought of when I read your question.  The problem of random USB port assignment can be fixed with a persistent udev rule.

antoinechauveau, 2018-03-23:
I'm currently setting up something similar to what @Theoi-Meteoroi described:

1. Sensors built with ESP8266. They run a simple web server that gives access to the sensor data.
Example command for integration to Mycodo: `wget -q -t 2 -T 5 -O - <sensor_ip>/humidity`

2. Controls built with TP-LINK smart plugs. They are connected directly to the wifi network (no smart hub) and can be controlled through a small shell script for on/off/measure power.
Example command in Mycodo: `hs110.sh <plug_ip> on`

eap904, 2018-03-24:
antoinechauveau

could I get more detailed info about how to implement those 2 things? I have a esp8266 board, but do not know how to set it up to run such a web server

antoinechauveau, 2018-03-24:
Here's a small tutorial I followed to get started with the webserver:
https://techtutorialsx.com/2016/10/03/esp8266-setting-a-simple-http-webserver/

Theoi-Meteoroi, 2018-03-25:
Here is an example of some code I had working with a Adafuit huzzah.  Includes a local RTC and simply connects to the AP and uses a specific MQTT server IP address.   I eventually migrated the design to ATWINC-1500 because the 8266 is kind of a power pig. The ESP32 has been buggy for me but would seem ideal with bluetooth to allow tweeking with a phone.   

For most uses, I'm partial to the Cortex M0 and ARM processors in general.   A stretch goal I have is to implement a LORA-WAN network with remote sensors.  There are excellent open-source options to implement the gateway with a Rpi3 but the hardware path requires some hacking.  

[Huzzah_MQTT_Iprobe_v0.3.ino.zip](https://github.com/kizniche/Mycodo/files/1844955/Huzzah_MQTT_Iprobe_v0.3.ino.zip)

jlb-qi, 2019-06-13:
@Theoi-Meteoroi I'm curious to find out if you ever made some progress on integrating MQTT to Mycodo. I was naively hoping that it could be integrated through the bash command input however, it would appear not. My setup is an esp8266 with a dht22 (for now) that publishes a feed which the Pi, obviously subscribes to.

closed 2019-10-07 by kizniche