kizniche/Mycodo #338: Server error

reymart081678, 2017-12-03:
## Mycodo Issue Report:
  - Specific Mycodo Version:5.4.11
 
#### Problem Description
Please list:
Try to monitor and got server error   
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

2017-12-04 06:54:18,592 - mycodo.pid_5 - ERROR - Exception while reading measurement from the influxdb database: timestamp out of range for platform time_t
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_pid.py", line 389, in get_last_measurement
    local_timestamp = str(datetime.datetime.fromtimestamp(utc_timestamp))
ValueError: timestamp out of range for platform time_t
2017-12-04 06:54:33,518 - mycodo.pid_5 - ERROR - Exception while reading measurement from the influxdb database: timestamp out of range for platform time_t
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_pid.py", line 389, in get_last_measurement
    local_timestamp = str(datetime.datetime.fromtimestamp(utc_timestamp))
ValueError: timestamp out of range for platform time_t
2017-12-04 06:54:48,540 - mycodo.pid_5 - ERROR - Exception while reading measurement from the influxdb database: timestamp out of range for platform time_t
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_pid.py", line 389, in get_last_measurement
    local_timestamp = str(datetime.datetime.fromtimestamp(utc_timestamp))
ValueError: timestamp out of range for platform time_t
2017-12-04 06:55:03,550 - mycodo.pid_5 - ERROR - Exception while reading measurement from the influxdb database: timestamp out of range for platform time_t
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_pid.py", line 389, in get_last_measurement
    local_timestamp = str(datetime.datetime.fromtimestamp(utc_timestamp))
ValueError: timestamp out of range for platform time_t

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?
Graph:
[image]

PID
[image]

kizniche, 2017-12-03:
Is your Pi's time incorrectly set?

kizniche, 2017-12-04:
I have a hunch this error is due to Mycodo requesting data from the future, meaning your time is incorrectly set.

kizniche, 2017-12-04:
In the Mycodo UI, go to the gear icon and then System information. The first data line will show the current system time.

kizniche, 2017-12-04:
Also, if the time "appears" to be correct, you may have an incorrect time zone set.

reymart081678, 2017-12-04:
I'll check it later, no issue running at lower version 5.4.3 and after
upgrade and 2 days after it started having hangup issue.

reymart081678, 2017-12-04:
Timezone and date is okay

kizniche, 2017-12-04:
Add these logging lines after line 388 of controller_pid.py, so it looks like this:

```python
                utc_dt = datetime.datetime.strptime(
                    self.last_time.split(".")[0],
                    '%Y-%m-%dT%H:%M:%S')
                self.logger.error("TEST01: {}".format(self.last_time.split(".")[0]))
                self.logger.error("TEST02: {}".format(t.time()))
                utc_timestamp = calendar.timegm(utc_dt.timetuple())
```

save, and restart the daemon. Then, once the errors start happening, copy the logging lines here. It might be good to include the data from the logging lines from before the error starts to see what a stable state looks like.

reymart081678, 2017-12-04:
Sorry for long debug info:  300 lines, result after including logging lines, server stop responding around 2230H and when i check around 2330H the server is not responding. I do not encounter this issue using 5.3.4

2017-12-04 22:26:20,102 - mycodo.output - DEBUG - 2017-12-04 22-26-20
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:26:20,108 - mycodo.output - DEBUG - Output 9 (circulationFan) turned off.
2017-12-04 22:26:34,675 - mycodo.pid_4 - ERROR - TEST01: 2017-12-04T14:26:25
2017-12-04 22:26:34,677 - mycodo.pid_4 - ERROR - TEST02: 1512397594.68
2017-12-04 22:26:34,678 - mycodo.pid_4 - DEBUG - Latest humidity: 87.72 @ 2017-12-04 22:26:25
2017-12-04 22:26:34,883 - mycodo.output - DEBUG - Output 1 (humidifier) turned off.
2017-12-04 22:26:35,015 - mycodo.output - DEBUG - Output 2 (humidifierFan) turned off.
2017-12-04 22:26:35,059 - mycodo.output - DEBUG - 2017-12-04 22-26-35
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:26:35,149 - mycodo.pid_5 - ERROR - TEST01: 2017-12-04T14:26:25
2017-12-04 22:26:35,150 - mycodo.pid_5 - ERROR - TEST02: 1512397595.15
2017-12-04 22:26:35,150 - mycodo.pid_5 - DEBUG - Latest temperature: 20.82 @ 2017-12-04 22:26:25
2017-12-04 22:26:35,171 - mycodo.output - DEBUG - Output 9 (circulationFan) turned off.
2017-12-04 22:26:49,735 - mycodo.pid_4 - ERROR - TEST01: 2017-12-04T14:26:40
2017-12-04 22:26:49,736 - mycodo.pid_4 - ERROR - TEST02: 1512397609.74
2017-12-04 22:26:49,736 - mycodo.pid_4 - DEBUG - Latest humidity: 87.72 @ 2017-12-04 22:26:40
2017-12-04 22:26:49,882 - mycodo.output - DEBUG - Output 1 (humidifier) turned off.
2017-12-04 22:26:50,024 - mycodo.output - DEBUG - Output 2 (humidifierFan) turned off.
2017-12-04 22:26:50,078 - mycodo.output - DEBUG - 2017-12-04 22-26-50
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:26:50,125 - mycodo.pid_5 - ERROR - TEST01: 2017-12-04T14:26:40
2017-12-04 22:26:50,128 - mycodo.pid_5 - ERROR - TEST02: 1512397610.13
2017-12-04 22:26:50,130 - mycodo.pid_5 - DEBUG - Latest temperature: 20.81 @ 2017-12-04 22:26:40
2017-12-04 22:26:50,152 - mycodo.output - DEBUG - Output 9 (circulationFan) turned off.
2017-12-04 22:27:04,763 - mycodo.pid_4 - ERROR - TEST01: 2017-12-04T14:26:55
2017-12-04 22:27:04,764 - mycodo.pid_4 - ERROR - TEST02: 1512397624.76
2017-12-04 22:27:04,765 - mycodo.pid_4 - DEBUG - Latest humidity: 87.76 @ 2017-12-04 22:26:55
2017-12-04 22:27:04,938 - mycodo.output - DEBUG - Output 1 (humidifier) turned off.
2017-12-04 22:27:05,009 - mycodo.output - DEBUG - Output 2 (humidifierFan) turned off.
2017-12-04 22:27:05,043 - mycodo.output - DEBUG - 2017-12-04 22-27-05
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:27:05,069 - mycodo.pid_5 - ERROR - TEST01: 2017-12-04T14:26:55
2017-12-04 22:27:05,070 - mycodo.pid_5 - ERROR - TEST02: 1512397625.07
2017-12-04 22:27:05,071 - mycodo.pid_5 - DEBUG - Latest temperature: 20.8 @ 2017-12-04 22:26:55
2017-12-04 22:27:05,085 - mycodo.output - DEBUG - Output 9 (circulationFan) turned off.
2017-12-04 22:27:19,709 - mycodo.pid_4 - ERROR - TEST01: 2017-12-04T14:27:10
2017-12-04 22:27:19,710 - mycodo.pid_4 - ERROR - TEST02: 1512397639.71
2017-12-04 22:27:19,711 - mycodo.pid_4 - DEBUG - Latest humidity: 87.75 @ 2017-12-04 22:27:10
2017-12-04 22:27:19,902 - mycodo.output - DEBUG - Output 1 (humidifier) turned off.
2017-12-04 22:27:19,973 - mycodo.output - DEBUG - Output 2 (humidifierFan) turned off.
2017-12-04 22:27:20,008 - mycodo.output - DEBUG - 2017-12-04 22-27-19
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:27:20,106 - mycodo.pid_5 - ERROR - TEST01: 2017-12-04T14:27:10
2017-12-04 22:27:20,106 - mycodo.pid_5 - ERROR - TEST02: 1512397640.11
2017-12-04 22:27:20,107 - mycodo.pid_5 - DEBUG - Latest temperature: 20.8 @ 2017-12-04 22:27:10
2017-12-04 22:27:20,123 - mycodo.output - DEBUG - Output 9 (circulationFan) turned off.
2017-12-04 22:27:32,704 - mycodo.output - DEBUG - Output 5 (exhaustFan) turned off.
2017-12-04 22:27:34,685 - mycodo.pid_4 - ERROR - TEST01: 2017-12-04T14:27:25
2017-12-04 22:27:34,686 - mycodo.pid_4 - ERROR - TEST02: 1512397654.69
2017-12-04 22:27:34,687 - mycodo.pid_4 - DEBUG - Latest humidity: 87.75 @ 2017-12-04 22:27:25
2017-12-04 22:27:34,921 - mycodo.output - DEBUG - Output 1 (humidifier) turned off.
2017-12-04 22:27:35,013 - mycodo.output - DEBUG - Output 2 (humidifierFan) turned off.
2017-12-04 22:27:35,056 - mycodo.output - DEBUG - 2017-12-04 22-27-35
[Output Conditional 5] humidityCntrlOff
If output 1 (humidifier) turns off, Then turn output 2 (humidifierFan) off.

2017-12-04 22:27:35,151 - mycodo.pid_5 - ERROR - TEST01: 2017-12-04T14:27:25
[... 206 lines cut ...]
2017-12-04 23:35:22,793 - mycodo.output - INFO - Output 1 setup on pin 16 and turned OFF (OFF=LOW)
2017-12-04 23:35:22,795 - mycodo.output - INFO - Output 2 setup on pin 21 and turned OFF (OFF=LOW)
2017-12-04 23:35:22,796 - mycodo.output - INFO - Output 3 setup on pin 25 and turned OFF (OFF=LOW)
2017-12-04 23:35:22,797 - mycodo.output - INFO - Output 4 setup on pin 12 and turned OFF (OFF=LOW)
[... cut ...]
2017-12-04 23:35:25,371 - mycodo.daemon - INFO - All activated input controllers started
2017-12-04 23:35:25,978 - mycodo.pid_4 - INFO - Activated in 531.3 ms
2017-12-04 23:35:26,348 - mycodo.pid_5 - INFO - Activated in 300.3 ms
2017-12-04 23:35:26,349 - mycodo.daemon - INFO - All activated PID controllers started
2017-12-04 23:35:26,350 - mycodo.daemon - INFO - All activated LCD controllers started
2017-12-04 23:35:26,351 - mycodo.daemon - INFO - Mycodo daemon v5.4.11 started in 4.355 seconds
2017-12-04 23:35:26,358 - mycodo.daemon - INFO - 46.75 MB RAM in use

reymart081678, 2017-12-04:
correction 5.4.3 version is the last version that is working, I'll revert it back now and observe

kizniche, 2017-12-05:
I don't see the error in the above log lines. Also, can you isolate the issue but turning everything off except the PID and the sensor associated with the PID to determine what exactly is the issue?

kizniche, 2017-12-05:
Also, you don't need to run the daemon in debug mode. I made the new log lines log as error so they would appear when the daemon is running normally.

kizniche, 2017-12-09:
I'm not really sure what's going on with your system. I haven't been able to reproduce this error.

kizniche, 2017-12-13:
If you're running the latest version (v5.4.15), go to ```Config -> System Information``` and see if the ```Database Version``` has a green status. If it's red, please paste what it says.

Theoi-Meteoroi, 2017-12-26:
I found 5.4.15 was last most stable version, in my case from fresh install.

Theoi-Meteoroi, 2018-04-10:
I think this was a local problem - as you mentioned the errors were a bad clock value.  And the Pi clock is software and unreliable without a RTC or reliable NTP source to function in the real world.  If not on a reliable network that can receive NTP broadcasts, a RTC is required for stable OS operations. The application relies on OS services, like a sane clock.  I don't think this is a bug in Mycodo.

kizniche, 2018-04-11:
This is the only occurrence I've seen of this issue, so I'm inclined to believe it's a system issue. I'm sure a lot of atypical behaviors can result from the time being off. If the OP can provide more information, or cares to continue investigating this issue, please reopen it and add some more information.

closed 2018-04-11 by kizniche