kizniche/Mycodo #339: No matching distribution found for smbus

hideouspenguinboy, 2017-12-05:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.11
 
#### Problem Description
Please list:

  - what were you trying to do:  Installation fails when trying to install smbus.
  - specific setup details that are involved: 
Raspberry Pi 3 rev B v1.2
Raspian Full, with most recent updates

Normal installation save for new Raspian config does not require file size change.
sudo /bin/bash ./setup.sh

Reformatted and reinstalled everything and tried again. Failure.
### Errors

Collecting smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41))
  Could not find a version that satisfies the requirement smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41)) (from versions: )
No matching distribution found for smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41))

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. run install
 2. failure
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue? Nope. It's pretty clear what's up.

kizniche, 2017-12-05:
Please attach your entire install log, at ~/Mycodo/install/setup.log

hideouspenguinboy, 2017-12-05:
### Mycodo installation beginning at 12-04-2017 23:09:02

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap currently set to 100 MB. Changing to 512 MB and restarting
Stopping dphys-swapfile (via systemctl): dphys-swapfile.service.
Starting dphys-swapfile (via systemctl): dphys-swapfile.service.

#### Updating apt repositories
Get:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [1                                                                                                  1.7 MB]
Get:4 http://archive.raspberrypi.org/debian stretch/main armhf Packages [127 kB]
Fetched 11.8 MB in 13s (891 kB/s)
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  python-pip*
0 upgraded, 0 newly installed, 1 to remove and 4 not upgraded.
After this operation, 671 kB disk space will be freed.
(Reading database ... 123003 files and directories currently installed.)
Removing python-pip (9.0.1-2+rpt1) ...
Processing triggers for man-db (2.7.6.1-2) ...

#### Installing prerequisite apt packages and update pip
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
gcc is already the newest version (4:6.3.0-4).
gcc set to manually installed.
git is already the newest version (1:2.11.0-3+deb9u2).
libgtk2.0-0 is already the newest version (2.24.31-2).
libgtk2.0-0 set to manually installed.
logrotate is already the newest version (3.11.0-0.1).
python-dev is already the newest version (2.7.13-2).
python-dev set to manually installed.
python-numpy is already the newest version (1:1.12.1-3).
python-numpy set to manually installed.
python-setuptools is already the newest version (33.1.1-1).
python-setuptools set to manually installed.
python-smbus is already the newest version (3.1.2-3).
wget is already the newest version (1.18-5+deb9u1).
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  apache2-bin apache2-data apache2-utils ffmpeg libapr1 libaprutil1
  libaprutil1-dbd-sqlite3 libaprutil1-ldap libavdevice57
  libboost-python1.62-dev libboost-python1.62.0 libboost1.62-dev
  libio-pty-perl libipc-run-perl liblua5.2-0 libopencv-contrib2.4v5
  libopencv-legacy2.4v5 libopencv-ml2.4v5 libopencv-photo2.4v5 libsdl2-2.0-0
  libsigsegv2 ssl-cert
Suggested packages:
  apache2-doc apache2-suexec-pristine | apache2-suexec-custom ffmpeg-doc
  gawk-doc libboost1.62-doc gccxml libboost-atomic1.62-dev
  libboost-chrono1.62-dev libboost-context1.62-dev libboost-coroutine1.62-dev
  libboost-date-time1.62-dev libboost-exception1.62-dev libboost-fiber1.62-dev
  libboost-filesystem1.62-dev libboost-graph1.62-dev
  libboost-graph-parallel1.62-dev libboost-iostreams1.62-dev
[... 567 lines cut ...]
  Downloading ratelim-0.1.6-py2.py3-none-any.whl
Collecting pytz (from influxdb==4.1.1->-r /home/pi/Mycodo/install/requirements.t                                                                                                  xt (line 21))
  Downloading pytz-2017.3-py2.py3-none-any.whl (511kB)
Collecting plumbum (from rpyc==3.4.4->-r /home/pi/Mycodo/install/requirements.tx                                                                                                  t (line 26))
  Downloading plumbum-1.6.4-py2.py3-none-any.whl (110kB)
Collecting smbus-cffi (from tsl2591->-r /home/pi/Mycodo/install/requirements.txt                                                                                                   (line 31))
  Downloading smbus-cffi-0.5.1.tar.gz
Collecting adafruit-pureio (from Adafruit_GPIO->-r /home/pi/Mycodo/install/requi                                                                                                  rements.txt (line 39))
  Downloading Adafruit_PureIO-0.2.1-py2-none-any.whl
Requirement already up-to-date: spidev in /usr/lib/python2.7/dist-packages (from                                                                                                   Adafruit_GPIO->-r /home/pi/Mycodo/install/requirements.txt (line 39))
Collecting smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (lin                                                                                                  e 41))
  Could not find a version that satisfies the requirement smbus (from MCP342x->-                                                                                                  r /home/pi/Mycodo/install/requirements.txt (line 41)) (from versions: )
No matching distribution found for smbus (from MCP342x->-r /home/pi/Mycodo/insta                                                                                                  ll/requirements.txt (line 41))

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

kizniche, 2017-12-05:
Something abnormal is happening on your system.

smbus is already installed, via python-smbus, through apt-get.

Can you try the following commands, one after the other:

```sudo ~/Mycodo/env/bin/pip install --upgrade pip```

Than copy the output of that command to a new comment here. Then, try the next command.

```sudo ~/Mycodo/env/bin/pip install -r ~/Mycodo/install/requirements.txt --upgrade --force-reinstall --no-deps```

hideouspenguinboy, 2017-12-05:
1. Requirement already up-to-date: pip in
./Mycodo/env/lib/python2.7/site-packages

2. Collecting testfixtures (from -r
/home/pi/Mycodo/install/requirements.txt (line 1))
  Using cached testfixtures-5.3.1-py2.py3-none-any.whl
Collecting mock (from -r /home/pi/Mycodo/install/requirements.txt (line 2))
  Using cached mock-2.0.0-py2.py3-none-any.whl
Collecting pytest==3.2.3 (from -r /home/pi/Mycodo/install/requirements.txt
(line 3))
  Using cached pytest-3.2.3-py2.py3-none-any.whl
Collecting factory_boy==2.8.1 (from -r
/home/pi/Mycodo/install/requirements.txt (line 4))
  Using cached factory_boy-2.8.1-py2.py3-none-any.whl
Collecting future (from -r /home/pi/Mycodo/install/requirements.txt (line
5))
  Using cached future-0.16.0.tar.gz
Collecting alembic==0.9.6 (from -r /home/pi/Mycodo/install/requirements.txt
(line 6))
  Using cached
https://www.piwheels.hostedpi.com/simple/alembic/alembic-0.9.6-py2.py3-none-any.whl
Collecting argparse==1.4.0 (from -r
/home/pi/Mycodo/install/requirements.txt (line 7))
  Using cached argparse-1.4.0-py2.py3-none-any.whl
Collecting bcrypt==3.1.4 (from -r /home/pi/Mycodo/install/requirements.txt
(line 8))
  Using cached bcrypt-3.1.4.tar.gz
Collecting daemonize==2.4.7 (from -r
/home/pi/Mycodo/install/requirements.txt (line 9))
  Using cached daemonize-2.4.7-py2.py3-none-any.whl
Collecting fasteners==0.14.1 (from -r
/home/pi/Mycodo/install/requirements.txt (line 10))
  Using cached fasteners-0.14.1-py2.py3-none-any.whl
Collecting Flask==0.12.2 (from -r /home/pi/Mycodo/install/requirements.txt
(line 11))
  Using cached Flask-0.12.2-py2.py3-none-any.whl
Collecting Flask_Babel==0.11.2 (from -r
/home/pi/Mycodo/install/requirements.txt (line 12))
  Using cached Flask_Babel-0.11.2-py2.py3-none-any.whl
Collecting Flask_InfluxDB==0.1 (from -r
/home/pi/Mycodo/install/requirements.txt (line 13))
  Using cached Flask-InfluxDB-0.1.tar.gz
Collecting Flask_Limiter==1.0.0 (from -r
/home/pi/Mycodo/install/requirements.txt (line 14))
  Using cached Flask-Limiter-1.0.0.tar.gz
Collecting Flask_Login==0.4.0 (from -r
/home/pi/Mycodo/install/requirements.txt (line 15))
  Using cached Flask_Login-0.4.0-py2.py3-none-any.whl
Collecting Flask_SQLAlchemy==2.2 (from -r
/home/pi/Mycodo/install/requirements.txt (line 16))
  Using cached Flask_SQLAlchemy-2.2-py2.py3-none-any.whl
Collecting Flask_SSLify (from -r /home/pi/Mycodo/install/requirements.txt
(line 17))
  Using cached Flask-SSLify-0.1.5.tar.gz
Collecting Flask_WTF==0.14.2 (from -r
/home/pi/Mycodo/install/requirements.txt (line 18))
  Using cached Flask_WTF-0.14.2-py2.py3-none-any.whl
Collecting geocoder==1.32.1 (from -r
/home/pi/Mycodo/install/requirements.txt (line 19))
  Using cached geocoder-1.32.1-py2.py3-none-any.whl
Collecting imutils==0.4.3 (from -r /home/pi/Mycodo/install/requirements.txt
(line 20))
  Using cached imutils-0.4.3.tar.gz
Collecting influxdb==4.1.1 (from -r
/home/pi/Mycodo/install/requirements.txt (line 21))
  Using cached influxdb-4.1.1-py2.py3-none-any.whl
Collecting pi_switch==0.4.0 (from -r
/home/pi/Mycodo/install/requirements.txt (line 22))
  Using cached pi_switch-0.4.0.tar.gz
Collecting pyserial==3.4 (from -r /home/pi/Mycodo/install/requirements.txt
[... 108 lines cut ...]
environment /home/pi/Mycodo/env
  Found existing installation: pyserial 3.2.1
    Not uninstalling pyserial at /usr/lib/python2.7/dist-packages, outside
environment /home/pi/Mycodo/env
  Found existing installation: RPi.GPIO 0.6.3
    Not uninstalling rpi.gpio at /usr/lib/python2.7/dist-packages, outside
environment /home/pi/Mycodo/env
  Running setup.py develop for tsl2591
  Running setup.py develop for adafruit-bme280
  Found existing installation: picamera 1.13
    Not uninstalling picamera at /usr/lib/python2.7/dist-packages, outside
environment /home/pi/Mycodo/env
Successfully installed Adafruit-ADS1x15-1.0.2 Adafruit-BMP-1.5.2
Adafruit-GPIO-1.0.3 Adafruit-TMP-1.6.1 Flask-0.12.2 Flask-Babel-0.11.2
Flask-InfluxDB-0.1 Flask-Limiter-1.0.0 Flask-Login-0.4.0
Flask-SQLAlchemy-2.2 Flask-SSLify-0.1.5 Flask-WTF-0.14.2 MCP342x-0.3.4
RPi.GPIO-0.6.3 SQLAlchemy-1.2.0b3 WTForms-2.1 adafruit-bme280 alembic-0.9.6
argparse-1.4.0 bcrypt-3.1.4 daemonize-2.4.7 factory-boy-2.8.1
fasteners-0.14.1 future-0.16.0 geocoder-1.32.1 imutils-0.4.3 influxdb-4.1.1
mock-2.0.0 pi-switch-0.4.0 picamera-1.13 pyserial-3.4 pytest-3.2.3
python-dateutil-2.6.1 rpyc-3.4.4 sht-sensor-17.5.5 tentacle-pi-0.6.2
testfixtures-5.3.1 tsl2591 w1thermsensor-1.0.4

It looks like it's installed right.  Just so you know this was an out of
the box install on a brand new Raspi3 and on two different SD cards.

kizniche, 2017-12-05:
Sometimes that smbus issue happens, but I have no idea why. I suspect it's either due to a corrupt file or a bug with pip. I've had the problem a few times in the >100 times I've installed Mycodo. Luckily I figured out the fix. BTW, I pulled that command from [here](https://github.com/kizniche/Mycodo#install-notes) in the install notes.

closed 2017-12-05 by kizniche

hideouspenguinboy, 2017-12-05:
Hmmm, it's still not starting up after the restart, though. I tried running the update, since that tends to fail if the install is bad and the only error it shot back was "it's the current version."

Checking whats running:
pi@raspberrypi:~ $ ps -A
  PID TTY          TIME CMD
    1 ?        00:00:02 systemd
    2 ?        00:00:00 kthreadd
    3 ?        00:00:00 ksoftirqd/0
    5 ?        00:00:00 kworker/0:0H
    7 ?        00:00:00 rcu_sched
    8 ?        00:00:00 rcu_bh
    9 ?        00:00:00 migration/0
   10 ?        00:00:00 lru-add-drain
   11 ?        00:00:00 cpuhp/0
   12 ?        00:00:00 cpuhp/1
   13 ?        00:00:00 migration/1
   14 ?        00:00:00 ksoftirqd/1
   16 ?        00:00:00 kworker/1:0H
   17 ?        00:00:00 cpuhp/2
   18 ?        00:00:00 migration/2
   19 ?        00:00:00 ksoftirqd/2
   21 ?        00:00:00 kworker/2:0H
   22 ?        00:00:00 cpuhp/3
   23 ?        00:00:00 migration/3
   24 ?        00:00:00 ksoftirqd/3
   25 ?        00:00:00 kworker/3:0
   26 ?        00:00:00 kworker/3:0H
   27 ?        00:00:00 kdevtmpfs
   28 ?        00:00:00 netns
   29 ?        00:00:00 khungtaskd
   30 ?        00:00:00 oom_reaper
   31 ?        00:00:00 writeback
   32 ?        00:00:00 kcompactd0
   33 ?        00:00:00 crypto
   34 ?        00:00:00 bioset
   35 ?        00:00:00 kblockd
   36 ?        00:00:00 watchdogd
   37 ?        00:00:00 kworker/0:1
   38 ?        00:00:00 rpciod
   39 ?        00:00:00 xprtiod
   40 ?        00:00:00 kswapd0
   41 ?        00:00:00 vmstat
   42 ?        00:00:00 nfsiod
   52 ?        00:00:00 kthrotld
   53 ?        00:00:00 bioset
   54 ?        00:00:00 bioset
   55 ?        00:00:00 bioset
   56 ?        00:00:00 bioset
   57 ?        00:00:00 bioset
   58 ?        00:00:00 bioset
   59 ?        00:00:00 bioset
   60 ?        00:00:00 bioset
   61 ?        00:00:00 bioset
   62 ?        00:00:00 bioset
   63 ?        00:00:00 bioset
   64 ?        00:00:00 bioset
   65 ?        00:00:00 bioset
   66 ?        00:00:00 bioset
   67 ?        00:00:00 bioset
   68 ?        00:00:00 bioset
   69 ?        00:00:00 bioset
   70 ?        00:00:00 bioset
   71 ?        00:00:00 bioset
   72 ?        00:00:00 bioset
   73 ?        00:00:00 bioset
   74 ?        00:00:00 bioset
   75 ?        00:00:00 bioset
   76 ?        00:00:00 bioset
   77 ?        00:00:00 iscsi_eh
   78 ?        00:00:00 dwc_otg
   79 ?        00:00:00 kworker/2:1
   80 ?        00:00:00 DWC Notificatio
   82 ?        00:00:00 VCHIQ-0
   83 ?        00:00:00 VCHIQr-0
   84 ?        00:00:00 VCHIQs-0
   85 ?        00:00:00 VCHIQka-0
   86 ?        00:00:00 SMIO
   87 ?        00:00:00 kworker/3:2
   88 ?        00:00:00 irq/92-mmc1
   90 ?        00:00:00 bioset
   91 ?        00:00:00 mmcqd/0
   92 ?        00:00:00 kworker/1:1
   94 ?        00:00:00 jbd2/mmcblk0p7-
   95 ?        00:00:00 ext4-rsv-conver
   96 ?        00:00:00 ipv6_addrconf
  107 ?        00:00:00 kworker/u8:1
  111 ?        00:00:00 kworker/2:1H
  114 ?        00:00:00 kworker/u8:2
  118 ?        00:00:00 systemd-journal
  154 ?        00:00:00 systemd-udevd
  161 ?        00:00:00 kworker/0:2
  184 ?        00:00:00 kworker/2:3
  230 ?        00:00:00 cfg80211
  235 ?        00:00:00 brcmf_wq/mmc1:0
  238 ?        00:00:00 brcmf_wdog/mmc1
  281 ?        00:00:00 systemd-timesyn
  310 ?        00:00:00 cron
  311 ?        00:00:00 avahi-daemon
  312 ?        00:00:00 rsyslogd
  313 ?        00:00:00 dbus-daemon
  322 ?        00:00:00 avahi-daemon
  329 ?        00:00:00 systemd-logind
  330 ?        00:00:00 thd
  364 ?        00:00:00 dhcpcd
  388 ?        00:00:00 sshd
  417 ?        00:00:00 lightdm
  420 ?        00:00:00 kworker/1:3
  423 ?        00:00:00 kworker/0:1H
  429 tty1     00:00:00 login
  444 ?        00:00:00 wpa_supplicant
  452 tty7     00:00:00 Xorg
  462 ?        00:00:00 apache2
  463 ?        00:00:00 apache2
  464 ?        00:00:00 apache2
  523 ?        00:00:00 kworker/3:1H
  547 ?        00:00:00 lightdm
  555 ?        00:00:00 systemd
  559 ?        00:00:00 (sd-pam)
  563 ?        00:00:00 kworker/u9:0
  564 ?        00:00:00 hci0
  565 ?        00:00:00 hci0
  566 ?        00:00:00 hciattach
  569 ?        00:00:00 kworker/u9:1
  571 ?        00:00:00 bluetoothd
  577 ?        00:00:00 bluealsa
  587 ?        00:00:00 lxsession
  602 ?        00:00:00 krfcommd
  606 ?        00:00:00 dbus-daemon
  622 tty1     00:00:00 bash
  668 ?        00:00:00 ssh-agent
  674 ?        00:00:00 gvfsd
  679 ?        00:00:00 gvfsd-fuse
  695 ?        00:00:00 openbox
  696 ?        00:00:00 lxpolkit
  700 ?        00:00:02 lxpanel
  701 ?        00:00:01 pcmanfm
  708 ?        00:00:00 ssh-agent
  718 ?        00:00:00 polkitd
  740 ?        00:00:00 menu-cached
  746 ?        00:00:00 gvfs-udisks2-vo
  749 ?        00:00:00 udisksd
  801 ?        00:00:00 gvfs-afc-volume
  806 ?        00:00:00 gvfs-goa-volume
  810 ?        00:00:00 gvfs-gphoto2-vo
  819 ?        00:00:00 gvfs-mtp-volume
  851 ?        00:00:00 gvfsd-trash
  863 ?        00:00:00 jbd2/mmcblk0p5-
  864 ?        00:00:00 ext4-rsv-conver
  874 ?        00:00:00 kworker/1:1H
  875 ?        00:00:00 kworker/u8:0
  876 ?        00:00:00 kworker/2:0
  877 ?        00:00:00 kworker/3:1
  878 ?        00:00:00 sshd
  888 ?        00:00:00 sshd
  891 pts/0    00:00:00 bash
  905 ?        00:00:00 sshd
  915 ?        00:00:00 sshd
  918 pts/1    00:00:00 bash
  933 pts/1    00:00:00 ps

hideouspenguinboy, 2017-12-05:
Also, checking crontab.... there is none. Checking systemctl to see what starts up with the raspi... no mention of Mycodo.

pi@raspberrypi:~/Mycodo/install $ systemctl
UNIT                                                        LOAD   ACTIVE SUB       DESCRIPTION
proc-sys-fs-binfmt_misc.automount                           loaded active waiting   Arbitrary Executable File
sys-devices-platform-soc-3f201000.serial-tty-ttyAMA0-hci0.device loaded active plugged   /sys/devices/platfor
sys-devices-platform-soc-3f201000.serial-tty-ttyAMA0.device loaded active plugged   /sys/devices/platform/soc
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0-mmcblk0p1.device loaded active
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0-mmcblk0p2.device loaded active
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0-mmcblk0p5.device loaded active
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0-mmcblk0p6.device loaded active
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0-mmcblk0p7.device loaded active
sys-devices-platform-soc-3f202000.sdhost-mmc_host-mmc0-mmc0:0007-block-mmcblk0.device loaded active plugged
sys-devices-platform-soc-3f300000.mmc-mmc_host-mmc1-mmc1:0001-mmc1:0001:1-net-wlan0.device loaded active plug
sys-devices-platform-soc-3f980000.usb-usb1-1\x2d1-1\x2d1.1-1\x2d1.1:1.0-net-eth0.device loaded active plugged
sys-devices-platform-soc-soc:audio-sound-card0.device       loaded active plugged   /sys/devices/platform/soc
sys-devices-virtual-block-ram0.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram1.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram10.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram11.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram12.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram13.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram14.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram15.device                      loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram2.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram3.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram4.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram5.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram6.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram7.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram8.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-block-ram9.device                       loaded active plugged   /sys/devices/virtual/bloc
sys-devices-virtual-misc-rfkill.device                      loaded active plugged   /sys/devices/virtual/misc
sys-devices-virtual-tty-ttyprintk.device                    loaded active plugged   /sys/devices/virtual/tty/
sys-module-configfs.device                                  loaded active plugged   /sys/module/configfs
sys-module-fuse.device                                      loaded active plugged   /sys/module/fuse
sys-subsystem-bluetooth-devices-hci0.device                 loaded active plugged   /sys/subsystem/bluetooth/
sys-subsystem-net-devices-eth0.device                       loaded active plugged   SMSC9512/9514 Fast Ethern
sys-subsystem-net-devices-wlan0.device                      loaded active plugged   /sys/subsystem/net/device
-.mount                                                     loaded active mounted   Root Mount
boot.mount                                                  loaded active mounted   /boot
dev-mqueue.mount                                            loaded active mounted   POSIX Message Queue File
media-pi-SETTINGS.mount                                     loaded active mounted   /media/pi/SETTINGS
run-rpc_pipefs.mount                                        loaded active mounted   RPC Pipe File System
run-user-1000-gvfs.mount                                    loaded active mounted   /run/user/1000/gvfs
run-user-1000.mount                                         loaded active mounted   /run/user/1000
sys-fs-fuse-connections.mount                               loaded active mounted   FUSE Control File System
sys-kernel-config.mount                                     loaded active mounted   Configuration File System
sys-kernel-debug.mount                                      loaded active mounted   Debug File System
systemd-ask-password-plymouth.path                          loaded active waiting   Forward Password Requests
systemd-ask-password-wall.path                              loaded active waiting   Forward Password Requests
init.scope                                                  loaded active running   System and Service Manage
[... cut ...]
apt-daily.timer                                             loaded active waiting   Daily apt download activi
systemd-tmpfiles-clean.timer                                loaded active waiting   Daily Cleanup of Temporar

LOAD   = Reflects whether the unit definition was properly loaded.
ACTIVE = The high-level unit activation state, i.e. generalization of SUB.
SUB    = The low-level unit activation state, values depend on unit type.

135 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.

kizniche, 2017-12-05:
The initial error interrupted your install. The fix merely fixed that one part. Try running the install script again to finish the install.

hideouspenguinboy, 2017-12-05:
pi@raspberrypi:~/Mycodo/install $ sudo /bin/bash ./setup.sh
### Mycodo installation beginning at 12-05-2017 13:15:19

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Get:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Get:3 http://mirrordirector.raspbian.org/raspbian stretch/main armhf Packages [11.7 MB]
Fetched 11.7 MB in 15s (751 kB/s)
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python-pip' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

#### Installing prerequisite apt packages and update pip
Hit:1 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
apache2 is already the newest version (2.4.25-3+deb9u3).
gawk is already the newest version (1:4.1.4+dfsg-1).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u2).
libapache2-mod-wsgi is already the newest version (4.5.11-1).
libav-tools is already the newest version (7:3.2.9-1~deb9u1).
libboost-python-dev is already the newest version (1.62.0.1+b4).
libffi-dev is already the newest version (3.2.1-6).
libgtk2.0-0 is already the newest version (2.24.31-2).
libi2c-dev is already the newest version (3.1.2-3).
logrotate is already the newest version (3.11.0-0.1).
moreutils is already the newest version (0.60-1).
python-dev is already the newest version (2.7.13-2).
python-numpy is already the newest version (1:1.12.1-3).
python-opencv is already the newest version (2.4.9.1+dfsg1-2).
python-setuptools is already the newest version (33.1.1-1).
python-smbus is already the newest version (3.1.2-3).
sqlite3 is already the newest version (3.16.2-5).
wget is already the newest version (1.18-5+deb9u1).
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Searching for pip
Best match: pip 9.0.1
Processing pip-9.0.1-py2.7.egg
pip 9.0.1 is already the active version in easy-install.pth
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg

#### Updating pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
[... 270 lines cut ...]
  Using cached ratelim-0.1.6-py2.py3-none-any.whl
Collecting pytz (from influxdb==4.1.1->-r /home/pi/Mycodo/install/requirements.txt (line 21))
  Using cached pytz-2017.3-py2.py3-none-any.whl
Collecting plumbum (from rpyc==3.4.4->-r /home/pi/Mycodo/install/requirements.txt (line 26))
  Using cached plumbum-1.6.4-py2.py3-none-any.whl
Collecting smbus-cffi (from tsl2591->-r /home/pi/Mycodo/install/requirements.txt (line 31))
  Using cached smbus-cffi-0.5.1.tar.gz
Collecting adafruit-pureio (from Adafruit_GPIO->-r /home/pi/Mycodo/install/requirements.txt (line 39))
  Using cached Adafruit_PureIO-0.2.1-py2-none-any.whl
Requirement already up-to-date: spidev in /usr/lib/python2.7/dist-packages (from Adafruit_GPIO->-r /home/pi/Mycodo/install/requirements.txt (line 39))
Collecting smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41))
  Could not find a version that satisfies the requirement smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41)) (from versions: )
No matching distribution found for smbus (from MCP342x->-r /home/pi/Mycodo/install/requirements.txt (line 41))

*******************************
*** ERROR: Install Aborted! ***
*******************************

An error occurred that prevented Mycodo from being installed!

Please contact the developer by submitting a bug report at
https://github.com/kizniche/Mycodo/issues with the pertinent
excerpts from the setup log located at Mycodo/install/setup.log

An error occurred. Exiting...

kizniche, 2017-12-05:
It seems to be related to [python-MCP342x Issue #2](https://github.com/stevemarple/python-MCP342x/issues/2)

kizniche, 2017-12-05:
Open ~/Mycodo/install/requirements.txt and change the following line:

```MCP342x```

to

```MCP342x == 0.3.3```

Then try the setup again.

hideouspenguinboy, 2017-12-05:
pi@raspberrypi:~/Mycodo/install $ sudo /bin/bash ./setup.sh
### Mycodo installation beginning at 12-05-2017 14:56:34

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python-pip' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

#### Installing prerequisite apt packages and update pip
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
apache2 is already the newest version (2.4.25-3+deb9u3).
gawk is already the newest version (1:4.1.4+dfsg-1).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u2).
libapache2-mod-wsgi is already the newest version (4.5.11-1).
libav-tools is already the newest version (7:3.2.9-1~deb9u1).
libboost-python-dev is already the newest version (1.62.0.1+b4).
libffi-dev is already the newest version (3.2.1-6).
libgtk2.0-0 is already the newest version (2.24.31-2).
libi2c-dev is already the newest version (3.1.2-3).
logrotate is already the newest version (3.11.0-0.1).
moreutils is already the newest version (0.60-1).
python-dev is already the newest version (2.7.13-2).
python-numpy is already the newest version (1:1.12.1-3).
python-opencv is already the newest version (2.4.9.1+dfsg1-2).
python-setuptools is already the newest version (33.1.1-1).
python-smbus is already the newest version (3.1.2-3).
sqlite3 is already the newest version (3.16.2-5).
wget is already the newest version (1.18-5+deb9u1).
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Searching for pip
Best match: pip 9.0.1
Processing pip-9.0.1-py2.7.egg
pip 9.0.1 is already the active version in easy-install.pth
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pi                              p-9.0.1-py2.7.egg

#### Updating pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pi                              p-9.0.1-py2.7.egg
#### Virtualenv already exists, skipping creation

[... 444 lines cut ...]
#### Could not connect to http://localhost. Waiting 60 seconds then trying again                               (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again                               (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again                               (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again                               (up to 5 times)...
#### Trying again...
#### Could not connect to http://localhost. Waiting 60 seconds then trying again (up to 5 times)...
#### Trying again...

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Tue  5 Dec 15:10:05 EST 2017

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.1.39/, or whatever your Pi's
IP address is, to create an admin user and log in.

**Some weirdness towards the end with connecting with the Mycodo database. Restarted it and it continued not to run. Running the install again, but not hopeful.**

kizniche, 2017-12-05:
Try deleting your entire ~/Mycodo directory and start over. I never test installing over a previous install.

hideouspenguinboy, 2017-12-05:
Well butter my biscuit. It worked. Website login works. Unsure what changed, but you can read the logs way better than I can.

pi@raspberrypi:~/Mycodo/install $ sudo /bin/bash ./setup.sh
### Mycodo installation beginning at 12-05-2017 15:28:11

#### Checking if swap size is 100 MB and needs to be changed to 512 MB
#### Swap not currently set to 100 MB. Not changing.

#### Updating apt repositories
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...

#### Uninstalling apt version of pip (if installed)
Reading package lists...
Building dependency tree...
Reading state information...
Package 'python-pip' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.

#### Installing prerequisite apt packages and update pip
Hit:1 http://archive.raspberrypi.org/debian stretch InRelease
Hit:2 http://mirrordirector.raspbian.org/raspbian stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
apache2 is already the newest version (2.4.25-3+deb9u3).
gawk is already the newest version (1:4.1.4+dfsg-1).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u2).
libapache2-mod-wsgi is already the newest version (4.5.11-1).
libav-tools is already the newest version (7:3.2.9-1~deb9u1).
libboost-python-dev is already the newest version (1.62.0.1+b4).
libffi-dev is already the newest version (3.2.1-6).
libgtk2.0-0 is already the newest version (2.24.31-2).
libi2c-dev is already the newest version (3.1.2-3).
logrotate is already the newest version (3.11.0-0.1).
moreutils is already the newest version (0.60-1).
python-dev is already the newest version (2.7.13-2).
python-numpy is already the newest version (1:1.12.1-3).
python-opencv is already the newest version (2.4.9.1+dfsg1-2).
python-setuptools is already the newest version (33.1.1-1).
python-smbus is already the newest version (3.1.2-3).
sqlite3 is already the newest version (3.16.2-5).
wget is already the newest version (1.18-5+deb9u1).
The following packages were automatically installed and are no longer required:
  libpython-all-dev python-all python-all-dev python-keyring
  python-keyrings.alt python-secretstorage python-wheel
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 4 not upgraded.
Searching for pip
Best match: pip 9.0.1
Processing pip-9.0.1-py2.7.egg
pip 9.0.1 is already the active version in easy-install.pth
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
Processing dependencies for pip
Finished processing dependencies for pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg

#### Updating pip
Requirement already up-to-date: pip in /usr/local/lib/python2.7/dist-packages/pip-9.0.1-py2.7.egg
[... 344 lines cut ...]
The user `mycodo' is already a member of `dialout'.
The user `mycodo' is already a member of `gpio'.
The user `mycodo' is already a member of `i2c'.
The user `mycodo' is already a member of `video'.

#### Setting permissions

#### Restarting the Mycodo web server
AH00558: apache2: Could not reliably determine the server's fully qualified domain name, using 127.0.1.1. Set the 'ServerName' directive globally to suppress this message

#### Creating Mycodo database if it doesn't exist
#### Successfully connected to http://localhost

#### Setting permissions

#### Restarting the Mycodo daemon
Mycodo previously shut down properly
Tue  5 Dec 15:33:03 EST 2017

***************************************************
** Mycodo successfully installed without errors! **
***************************************************

Go to https://192.168.1.39/, or whatever your Pi's
IP address is, to create an admin user and log in.

kizniche, 2017-12-06:
Looks good! I take it it's working fine now?

I added the 0.3.3 version to the requirements.txt to prevent anyone else from experiencing the same error during the install, until I can figure out if there's an issue with the python module or my code.

hideouspenguinboy, 2017-12-06:
Everything looks stable after a day of futzing with the web interface!

kizniche, 2017-12-06:
Great! Thanks for the update.