kizniche/Mycodo #340: pH calibration clear error

idlelion, 2017-12-05:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.11
 
#### Problem Description
Please list:
 tried calibrating pH probe- 
  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/calibration_routes.py", line 82, in calibration_atlas_ph
    atlas_command = AtlasScientificCommand(selected_input)
  File "/var/www/mycodo/mycodo/utils/calibration.py", line 20, in __init__
    self.interface = sensor_sel.interface
AttributeError: 'NoneType' object has no attribute 'interface'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. calibrate tab
 2. choose pH
 3. does not see sensor
4. clear calibration
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2017-12-05:
Thanks for the big report. I see the isseu and am working no a fix. I don't have this sensor, so you'll have to help me with the testing, if that's alright.

kizniche, 2017-12-05:
The following edits will fix this issue. Now, there may be other issues past this, so please try to do this and let me know if there are any other issues you encounter during the calibration process and I'll work on the fixes. Once we get everything working, I'll make a new release with the fixes for everyone to be able to use.

Open ~/Mycodo/mycodo/mycodo_flask/calibration_routes.py and make the following two edits:

After line 4, add the following import line, so it looks like:

```python
import logging
from sqlalchemy import or_
from flask_babel import gettext
```

Then, edit line 68 ("input_dev = Input.query...") so it looks like this:

```python
    input_dev = Input.query.filter(
        or_(Input.device == 'ATLAS_PH_UART',
            Input.device == 'ATLAS_PH_I2C')).all()
```

Save the file, then restart the web server with ```sudo service apache2 restart```

idlelion, 2017-12-06:
Thanks for the help. I'll try it in the morning.

Seaton

idlelion, 2018-02-22:
Hello,

I haven't been on mycodo in awhile is there a way to reset my password?

thanks

kizniche, 2018-02-22:
Unless you have another admin account that you can log in and change your other user's password, you're going to have to either A) Delete the admin use from the database then restart the web UI to induce the creation of another admin user, or B) Delete your entire database and restart eh web UI to create a new database and initilize the admin user creation page.

### A: Delete the current admin user

1. Download [SQLiteStudio](https://sqlitestudio.pl/index.rvt)
2. Open your mycodo database at ~/Mycodo/databases/mycodo.db
3. Delete the entry in the User table, then save
4. Restart the web UI with ```sudo service mycodoflask restart```
5. Visit your web interface and you will be presented to create a new admin user

### B: Delete your database

Note: Deleting your database will mean you will be starting from scratch and will need to add all your inputs, PIDs, etc. again.

1. Stop the Mycodo daemon with ```sudo service mycodo stop```
2. Delete your Mycodo database with ```rm -rf ~/Mycodo/databases/mycodo.db```
3. Restart the web UI with ```sudo service mycodoflask restart```
4. Visit your web interface and you will be presented to create a new admin user

idlelion, 2018-02-23:
Thanks for the advice.

Seaton

closed 2018-02-26 by kizniche