kizniche/Mycodo #341: Sql and new Math

RynFlutsch, 2017-12-09:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 upgraded vom 5.4.14 to 5.4.15
#### Problem Description

Error occured already on upgrade.

Already tried:
sudo ~/Mycodo/env/bin/pip install -r ~/Mycodo/install/requirements.txt --upgrade --force-reinstall --no-deps
and
cd ~/Mycodo/databases && sudo ~/Mycodo/env/bin/alembic upgrade head

Mycodo Log:
`Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/mycodo_daemon.py", line 758, in start_all_controllers
    'Math': db_retrieve_table_daemon(Math, entry='all'),
  File "/var/www/mycodo/mycodo/utils/database.py", line 63, in db_retrieve_table_daemon
    return_table = return_table.all()
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2724, in all
    return list(self)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2876, in __iter__
    return self._execute_and_instances(context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2899, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
OperationalError: (sqlite3.OperationalError) no such column: math.max_difference [SQL: u'SELECT math.id AS math_id, math.unique_id AS math_unique_id, math.name AS math_name, math.math_type AS math_math_type, math.is_activated AS math_is_activated, math.period AS math_period, math.inputs AS math_inputs, math.max_measure_age AS math_max_measure_age, math.measure AS math_measure, math.measure_units AS math_measure_units, math.max_difference AS math_max_difference, math.dry_bulb_t_id AS math_dry_bulb_t_id, math.dry_bulb_t_measure AS math_dry_bulb_t_measure, math.wet_bulb_t_id AS math_wet_bulb_t_id, math.wet_bulb_t_measure AS math_wet_bulb_t_measure, math.pressure_pa_id AS math_pressure_pa_id, math.pressure_pa_measure AS math_pressure_pa_measure \nFROM math']
2017-12-09 08:21:16,208 - mycodo.daemon - INFO - Mycodo daemon v5.4.15 started in 0.553 seconds
2017-12-09 08:21:16,213 - mycodo.daemon - INFO - 44.56 MB RAM in use`
 Error 500:
`Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 1005, in page_math
    math = Math.query.all()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2724, in all
    return list(self)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2876, in __iter__
    return self._execute_and_instances(context)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/orm/query.py", line 2899, in _execute_and_instances
    result = conn.execute(querycontext.statement, self._params)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/sql/elements.py", line 269, in _execute_on_connection
    return connection._execute_clauseelement(self, multiparams, params)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1060, in _execute_clauseelement
    compiled_sql, distilled_params
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
OperationalError: (sqlite3.OperationalError) no such column: math.max_difference [SQL: u'SELECT math.id AS math_id, math.unique_id AS math_unique_id, math.name AS math_name, math.math_type AS math_math_type, math.is_activated AS math_is_activated, math.period AS math_period, math.inputs AS math_inputs, math.max_measure_age AS math_max_measure_age, math.measure AS math_measure, math.measure_units AS math_measure_units, math.max_difference AS math_max_difference, math.dry_bulb_t_id AS math_dry_bulb_t_id, math.dry_bulb_t_measure AS math_dry_bulb_t_measure, math.wet_bulb_t_id AS math_wet_bulb_t_id, math.wet_bulb_t_measure AS math_wet_bulb_t_measure, math.pressure_pa_id AS math_pressure_pa_id, math.pressure_pa_measure AS math_pressure_pa_measure \nFROM math']`

 
### Additional Notes
Daemon is Running. 
This error occurs when i click on Math.

kizniche, 2017-12-09:
You seem to have a settings database issue. The easy way is to delete your database and let the web UI generate a new one. If you care about your past sensor data, it will be a bit more involved to repair your database, since each database issue is unique. Let me know if you want it repaired, and you can send it to me and I'll make the necessary edit.

kizniche, 2017-12-09:
Just curious, what was the version you installed Mycodo at?

Also, if you want me to edit the database, connect me at kylegabriel.com/contact and I'll reply with my email you can send your ~/Mycodo/databases/Mycodo.db

RynFlutsch, 2017-12-09:
Hello,

delete ist ok for me, just testing.
Startet with 5.x.x

Kind Regards
Richard

Gruß 
Richard

closed 2017-12-10 by kizniche

RynFlutsch, 2017-12-10:
How to properly create new db?
removed mycodo.db

RynFlutsch, 2017-12-10:
With deleted mycodo.db internal error occured.
So,the UI will not generate a new db automatic?

kizniche, 2017-12-10:
Try the instructions [here](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#incorrect-database-version). I think you're missing restarting the web server.

RynFlutsch, 2017-12-18:
> Also, if you want me to edit the database, connect me at https://kylegabriel.com/contact

Submit Button hung at Processing, without page load after solving captcha. Some Java bug ?
Tried second browser.

kizniche, 2017-12-18:
I just tested and the email form worked for me.

I found a way to fix the database ([reference](https://github.com/kizniche/Mycodo/issues/347))

Can you paste the output from both of these commands:

```bash
cd ~/Mycodo/databases
~/Mycodo/env/bin/alembic upgrade head
~/Mycodo/env/bin/alembic history
```

RynFlutsch, 2017-12-18:
Hi

INFO  [alembic.runtime.migration] Context impl SQLiteImpl.
INFO  [alembic.runtime.migration] Will assume non-transactional DDL.

b9712d4ec64e -> 8b36095c6cf9 (head), Add Math type: Input Verification
234283cc67f4 -> b9712d4ec64e, Add Math controller
8828a0074a44 -> 234283cc67f4, Add device option for camera
3dcf34dd7caf -> 8828a0074a44, Add Max Age option to LCD lines
321038f2e101 -> 3dcf34dd7caf, Add upgrade check to daemon
d0757b2ecd33 -> 321038f2e101, Add more Graph options
3c7c2b12389d -> d0757b2ecd33, Add unique_id for remote table
70b7016e107f -> 3c7c2b12389d, Add options to cycle multiple displays on one LCD
589ab40606d3 -> 70b7016e107f, Add user language option and delete misc language option
214c6bb4603a -> 589ab40606d3, Remove camera type, only use library instead
f4c0693f12a4 -> 214c6bb4603a, Add method_id as option for new PWM Timer
56ec4ceb9dfd -> f4c0693f12a4, Add sample time option for PWM and RPM Inputs
a0c55d19384c -> 56ec4ceb9dfd, Add PWM and RPM Input options
08a36ebf1a82 -> a0c55d19384c, Remove PID type option
ca975c26965c -> 08a36ebf1a82, Add Sensor option to inverse ADC unit scale
66db29288333 -> ca975c26965c, Add LCD I2C bus option
25676b9d5856 -> 66db29288333, Add new sensor type command
ed7e979852fa -> 25676b9d5856, Add option to end duration PID method with repeats after duration
c7b4a120a7bb -> ed7e979852fa, Add graph type
b604cf735be5 -> c7b4a120a7bb, add MH-Z19 CO2 sensor
9d7631079ac1 -> b604cf735be5, add options for PWM output
a1fcd7a4adf2 -> 9d7631079ac1, Add options for wireless relay control
a9a330ea0ccb -> a1fcd7a4adf2, Add calibration_sensor_measure option for Atlas Scientific pH sensors
f1c6b2901d45 -> a9a330ea0ccb, Add UART options: baud rate and device location
059a47f950b8 -> f1c6b2901d45, Add resolution and sensitivity to sensor table (for BH1750 sensor)
<base> -> 059a47f950b8, add option to show/hide tooltips

kizniche, 2017-12-18:
Try this:

```bash
cd ~/Mycodo/databases
~/Mycodo/env/bin/alembic stamp b9712d4ec64e
~/Mycodo/env/bin/alembic upgrade head
```

RynFlutsch, 2017-12-18:
Traceback (most recent call last):
  File "/root/Mycodo/env/bin/alembic", line 11, in <module>
    sys.exit(main())
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/config.py", line 479, in main
    CommandLine(prog=prog).main(argv=argv)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/config.py", line 473, in main
    self.run_cmd(cfg, options)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/config.py", line 456, in run_cmd
    **dict((k, getattr(options, k, None)) for k in kwarg)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/command.py", line 254, in upgrade
    script.run_env()
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/script/base.py", line 425, in run_env
    util.load_python_file(self.dir, 'env.py')
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/util/pyfiles.py", line 81, in load_python_file
    module = load_module_py(module_id, path)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/util/compat.py", line 141, in load_module_py
    mod = imp.load_source(module_id, path, fp)
  File "alembic/env.py", line 70, in <module>
    run_migrations_online()
  File "alembic/env.py", line 65, in run_migrations_online
    context.run_migrations()
  File "<string>", line 8, in run_migrations
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/runtime/environment.py", line 836, in run_migrations
    self.get_context().run_migrations(**kw)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/runtime/migration.py", line 330, in run_migrations
    step.migration_fn(**kw)
  File "/root/Mycodo/databases/alembic/versions/8b36095c6cf9_add_math_type_input_verification.py", line 27, in upgrade
    batch_op.add_column(sa.Column('pressure_pa_measure', sa.FLOAT))
  File "/usr/lib/python2.7/contextlib.py", line 24, in __exit__
    self.gen.next()
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/operations/base.py", line 300, in batch_alter_table
    impl.flush()
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/operations/batch.py", line 57, in flush
    fn(*arg, **kw)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/ddl/impl.py", line 172, in add_column
    self._exec(base.AddColumn(table_name, column, schema=schema))
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/alembic/ddl/impl.py", line 118, in _exec
    return conn.execute(construct, *multiparams, **params)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 948, in execute
    return meth(self, multiparams, params)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/sql/ddl.py", line 68, in _execute_on_connection
    return connection._execute_ddl(self, multiparams, params)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1009, in _execute_ddl
    compiled
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1200, in _execute_context
    context)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1413, in _handle_dbapi_exception
    exc_info
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/util/compat.py", line 203, in raise_from_cause
    reraise(type(exception), exception, tb=exc_tb, cause=cause)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/base.py", line 1193, in _execute_context
    context)
  File "/root/Mycodo/env/local/lib/python2.7/site-packages/sqlalchemy/engine/default.py", line 507, in do_execute
    cursor.execute(statement, parameters)
sqlalchemy.exc.OperationalError: (sqlite3.OperationalError) no such table: math [SQL: u'ALTER TABLE math ADD COLUMN max_difference FLOAT']

kizniche, 2017-12-18:
Try this:

```bash
cd ~/Mycodo/databases
~/Mycodo/env/bin/alembic stamp 234283cc67f4
~/Mycodo/env/bin/alembic upgrade head
```