kizniche/Mycodo #343: Math Live Page Issue

RynFlutsch, 2017-12-14:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.4.16
#### Problem Description
Please list

Created Math Average with 2x BME 280

If activated the live page show Internal 500
 
Daemon.log
`2017-12-14 09:58:38,236 - mycodo.daemon - INFO - 45.60 MB RAM in use
2017-12-14 09:58:48,259 - mycodo.math_1 - ERROR - Run Error: 
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/controller_math.py", line 170, in run
    success, measure = self.get_measurements_from_str(self.inputs)
  File "/var/www/mycodo/mycodo/controller_math.py", line 309, in get_measurements_from_str
    self.max_measure_age)
  File "/var/www/mycodo/mycodo/utils/influx.py", line 140, in read_last_influxdb
    last_measurement = client.query(query).raw
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/influxdb/client.py", line 381, in query
    expected_response_code=expected_response_code
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/influxdb/client.py", line 256, in request
    raise requests.exceptions.ConnectionError
ConnectionError`

error.log
`Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 681, in page_live
    maths_sorted=maths_sorted)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/templating.py", line 134, in render_template
    context, ctx.app)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/templating.py", line 116, in _render
    rv = template.render(context)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 3, in top-level template code
    {% set help_page = ["live-measurements", _('Live Measurements')] %}
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/layout.html", line 213, in top-level template code
    {%- block body %}{% endblock -%}
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 293, in block "body"
    {% if each_measure and measurement_units[each_measure]['meas'] == 'temperature' %}
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
UndefinedError: 'dict object' has no attribute u'Measurement'`

kizniche, 2017-12-14:
Thanks. I'll fix this.
In the meantime, to resolve the issue, change "Measurement" to "temperature"

RynFlutsch, 2017-12-14:
Working for Live Data.
Can‘t use as Pid setpoint
Error: Modify PID: A valid sensor is required
added `temperature` unit `°C`

kizniche, 2017-12-14:
Thanks. These should be easy fixes once I get to my development system.

berserktron3k, 2017-12-14:
This seems like it may be more prevalent. I was getting this same problem after the upgrade simply going to the Live page. Only sensor configured is a DHT22. Tried a clean install and have the same issue with Live data.

kizniche, 2017-12-14:
I haven't had an issue with the Live page in a long time, only since adding the Math controllers. All my fresh installs have worked fine. Though, if you have any errors or details to track down your issue, I'll investigate.

berserktron3k, 2017-12-14:
Should I open as a new issue?

Brady Hummel

kizniche, 2017-12-14:
If it's only the Live page, you can put all the information you have here. If it's an error on another page, start a new issue.

RynFlutsch, 2017-12-14:
Math vertification with set Measurement`temperature` and Unit `°C` result activated in Internal 500 error at live page:
`Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 681, in page_live
    maths_sorted=maths_sorted)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/templating.py", line 134, in render_template
    context, ctx.app)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/templating.py", line 116, in _render
    rv = template.render(context)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 3, in top-level template code
    {% set help_page = ["live-measurements", _('Live Measurements')] %}
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/layout.html", line 213, in top-level template code
    {%- block body %}{% endblock -%}
  File "/var/www/mycodo/mycodo/mycodo_flask/templates/pages/live.html", line 293, in block "body"
    {% if each_measure and measurement_units[each_measure]['meas'] == 'temperature' %}
  File "/var/www/mycodo/env/lib/python2.7/site-packages/jinja2/environment.py", line 411, in getitem
    return obj[argument]
UndefinedError: 'dict object' has no attribute u'temperature '`

kizniche, 2017-12-14:
You have a space after temperature. I needed to fix the issue of measurements not in the measurement dictionary causing problems.

kizniche, 2017-12-15:
[v5.4.17](https://github.com/kizniche/Mycodo/releases/tag/v5.4.17) was just released that addresses these issues.

closed 2017-12-15 by kizniche

RynFlutsch, 2017-12-15:
yes  i saw. Fixed and working without space.