kizniche/Mycodo #344: Method Time/Date, Edit Field?

RynFlutsch, 2017-12-14:
[image]

kizniche, 2017-12-14:
You're finding the bugs today! Thanks. I'll fix this along with the others and get a new release out as soon as I can.

RynFlutsch, 2017-12-14:
No rush ;)
Just testing my empty aquarium.
I am at the L2C Soil Moiste Sensor now. Maybe here is another bug ?
`2017-12-14 21:30:25,330 - mycodo.inputs.chirp - ERROR - ChirpSensor raised an exception when taking a reading: 121
Traceback (most recent call last):
  File "/var/www/mycodo/mycodo/inputs/chirp.py", line 96, in read
    self._lux, self._moisture, self._temperature = self.get_measurement()
  File "/var/www/mycodo/mycodo/inputs/chirp.py", line 83, in get_measurement
    lux = self.filter_average('lux', measurement=self.light())
  File "/var/www/mycodo/mycodo/inputs/chirp.py", line 133, in light
    self.bus.write_byte(self.address, 3)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/util.py", line 59, in validator
    return fn(*args, **kwdefaults)
  File "/var/www/mycodo/env/local/lib/python2.7/site-packages/smbus/smbus.py", line 121, in write_byte
    raise IOError(ffi.errno)
IOError: 121`

kizniche, 2017-12-14:
Did this happen one (or twice) and the sensor worked or did the sensor not return any readings and this error continually happened? I have a chirp sensor so I can hook it up to test.

kizniche, 2017-12-14:
Also, go to Config -> System Information and make sure the Database Version is green. If it's red, copy the error that's there.

RynFlutsch, 2017-12-14:
Moisture and Temp of all 3 sensor are looking reasonable.
Lux readings are very random and baloney
S1 0 lx 
Sensor 2 & S3 showing the same reading: 23,71 klx.
Lux-Sensor is covered by heatshrink tubing.
The Demon log shows continually happening.

RynFlutsch, 2017-12-14:
> Also, go to Config -> System Information and make sure the Database Version is green.
DB Version is green

kizniche, 2017-12-14:
I'll add the input number to the log lines in the next release so you can determine which input is causing the error. I'll also test myself, but if I can't replicate the issue, we'll have to wait to find out if all your chirp sensors cause this error or of it's a specific sensor that may be a hardware problem.

RynFlutsch, 2017-12-14:
I think its alwas the newst sensor wired. Or the last sensor wired, affect the sensor before it.

kizniche, 2017-12-14:
You can try setting up your multiplexer according to the Note at https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#i2c-multiplexers and see if it alleviates your issues. I set mine up this way and it's working great.

kizniche, 2017-12-15:
My chirp sensor worked fine with my multiplexer. I couldn't replicate the issue you're having.

You can edit ```~/Mycodo/mycodo/inputs/chirp.py``` and add after line 25 so it looks like:

```python
        if not testing:
            self.logger = logging.getLogger("mycodo.inputs.chirp_{b}_{a}".format(b=bus, a=address))
            self.bus = smbus.SMBus(bus)
            self.filter_average('lux', init_max=5)
```

Then restart the daemon and individually start each sensor to be able to determine which sensor is throwing the error.

[v5.4.17](https://github.com/kizniche/Mycodo/releases/tag/v5.4.17) was just released that addresses the method issue.

kizniche, 2017-12-15:
Actually, that logging line is not going to work if you don't change your multiplexer method to what I mentioned in my link. TO discover what chirp sensor is being read from, you're going to have to restart the daemon in debug mode and watch what channel is being selected before the attempted chirp reads.

RynFlutsch, 2017-12-16:
i am trying but i won‘t work.
`pi@raspberrypi:/boot $ ls
bcm2708-rpi-0-w.dtb     bcm2708-rpi-cm.dtb   bcm2710-rpi-cm3.dtb  config.txt     fixup.dat     issue.txt    LICENCE.broadcom  start_cd.elf  start_x.elf
bcm2708-rpi-b.dtb       bcm2709-rpi-2-b.dtb  bootcode.bin         COPYING.linux  fixup_db.dat  kernel7.img  LICENSE.oracle    start_db.elf
bcm2708-rpi-b-plus.dtb  bcm2710-rpi-3-b.dtb  cmdline.txt          fixup_cd.dat   fixup_x.dat   kernel.img   overlays          start.elf
pi@raspberrypi:/boot $ sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo i2c_gpio-pca9548.dts
FATAL ERROR: Couldn't open "i2c_gpio-pca9548.dts": No such file or directory
pi@raspberrypi:/boot $ cd
pi@raspberrypi:~ $ sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo i2c_gpio-pca9548.dts
FATAL ERROR: Couldn't open "i2c_gpio-pca9548.dts": No such file or directory
pi@raspberrypi:~ $`

kizniche, 2017-12-26:
Sorry for the long wait for a proper reply. This is how I installed the overlay:

```bash
sudo apt-get install git
git clone https://github.com/Theoi-Meteoroi/GPIO-pca9548
cd GPIO-pca9548
sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo i2c_gpio-pca9548.dts
```

Then edit /boot/config.txt with ```sudo nano /boot/config.txt``` and add the following lines to the end:

```
dtoverlay=i2c-gpio,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=4
dtoverlay=i2c_gpio-pca9548
```

Save the file, then reboot. Once the system starts back up, you should have 8 new I2C busses. I've done this on a few of my Pi 3s.

closed 2018-02-26 by kizniche