kizniche/Mycodo #345: Can not log into Mycodo after first time

cbaymac, 2017-12-15:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.4.17
 
#### Problem Description
Please list:
- trying to log into system from web browser on same wired network. I get the username a password page but the error below when i hit enter
 
### Errors
  - List any errors you encountered.
Error (Full Traceback):

Traceback (most recent call last):
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/var/www/mycodo/env/lib/python2.7/site-packages/flask_login/utils.py", line 228, in decorated_view
    return func(*args, **kwargs)
  File "/var/www/mycodo/mycodo/mycodo_flask/page_routes.py", line 661, in page_live
    for each_math_order in math_display_order:
TypeError: 'NoneType' object is not iterable

 
### Additional Notes
Same issue with Chrome and Firefox. Same wired network.

kizniche, 2017-12-15:
Well that's not good. The issue isn't that you're not logged in (you are) but that the landing page after logging in is throwing an error. Try going to https://yourIPaddress/input and you'll be able to reach the interface. From there, you can upgrade (Config -> Upgrade) to 5.4.18 that I'm going to be releasing shortly.

kizniche, 2017-12-15:
I just tested the latest release without any Math controllers and the error has been resolved. Let me know if you experience any more issues after an upgrade. Thanks for the bug report.

closed 2017-12-15 by kizniche

cbaymac, 2017-12-16:
Kyle,
  Thank you. You do great work. I'm looking forward to some excellent
mushrooms thanks to you.
Gordon

Gordon McPheter
757-790-9042

cbaymac, 2018-03-07:
Kyle,
   Good morning.  I am working with mycology and want to use the PID
controller in Mycodo to monitor the temperature in a pressure cooker via
type K thermocouple and a Max 31855 io board connected to my raspberry pi.
I then want to use this input to controll the output which is the hot plate
the pressure cooker is sitting on top of.
  I can buy a PID controller on amazon for $95 which will do this, but it
is expensive and I can’t monitor it via WiFi. This idea is based on a
Youtube tutorial by Eric Myers of Myers Mushrooms. This is the link.
https://youtu.be/0EE9QyLOjMs
Thank you, you do great work. I have built two control boxes using your
software.  They work great.
Regards,
Gordon McPheter
Hampton, Virginia
757-790-9042

Gordon McPheter 757-790-9042

kizniche, 2018-03-07:
It looks like it will work easily enough. I see [Adafruit makes a breakout board](https://www.adafruit.com/product/269) for the MAX31855K for reading type K thermocouples. Do you have this chip and are able to acquire measurements with your Raspberry Pi?

kizniche, 2018-03-07:
It also looks like Adafruit made a [Python library](https://github.com/adafruit/Adafruit_Python_MAX31855) for the MAX31855. If you have the chip, thermocouple, and a Pi, could you test if the code works? If so, I can add an Input module to Mycodo.

cbaymac, 2018-03-07:
Kyle,
  I have the chip on the way. I’ll give it a try and let you know how it
works.
Regards,
Gordon

cbaymac, 2018-03-08:
Kyle,
  The python script, Max31855 and RPi work as advertised. Photos attached.
If you could build a PID interface for Mycodo that would be great.
Gordon

Gordon McPheter
757-790-9042

kizniche, 2018-03-08:
I just added the MAX31855 as an Input. I'll make a release sometime tonight. After that you will be able to use the upgrade feature to add the new functionality to your Mycodo install.

> If you could build a PID interface for Mycodo that would be great.

I'm not sure what this means. Can you elaborate? Did you mean add the MAX31855 as an input? Because any Input can be used with the PID functionality with Mycodo.

cbaymac, 2018-03-08:
Kyle,
  Guess I’m confused now. Wha did you mean here?

cbaymac, 2018-03-08:
Kyle,
   Okay.  I now see your note where you did add the MAX 31855 as an input.
I’ll give it a try.
Thank you,
Gordon

kizniche, 2018-03-08:
So the new input (MAX31855) will store two measurements in the Mycodo database (the k-type thermocouple and an internal temperature from the MAX31855 chip itself). These measurements can then be used with PIDs and other parts of Mycodo. I made a commit to github that incorporates the new MAX31855 code and allows the new input to be used. However, I have not made a release yet. When I make a release, then you can go to Config -> Upgrade and your system will upgrade to the new code and you can see if the new input works. If it does, great. If not, we can investigate why and I can make a fix. I'll get the release out by tonight, so you can test after that.

cbaymac, 2018-03-08:
Nice,
  Thank you.

kizniche, 2018-03-10:
I find the MAX31856 pretty interesting because it can measure from several types of thermocouples (K, J, N, R, S, T, E, and B). There was a python library so I added this as an input as well, to give support to a full-range of thermocouple types.

cbaymac, 2018-03-10:
Kyle,
   Good stuff. Wonderful program. Thank you.
G.

Gordon McPheter
757-790-9042

cbaymac, 2018-03-11:
Kyle,
  I have a MAX31865 on order to provide amp and I/O to the raspberry pi for
a PT100 RDT  It is similar to the type K thermocouple which works with the
MAX 31855, but has three wires.  It might work with Mycodo and your MAX
31855 input?  i'll let you know.
Gordon

Gordon McPheter
757-790-9042

cbaymac, 2018-03-15:
Kyle, I see you have added some new inputs to Mycodo. Will the PT 100 work
with the PT1000 input.? I have the MAX 31865 board.

cbaymac, 2018-03-16:
No luck so far. This is the info I used to configure the PT100 with the
RPi.

cbaymac, 2018-03-16:
http://nicke.pe.hu/max31865/rpi.html

kizniche, 2018-03-16:
I'll look into it. If the MAX31865 can read the PT100, I should be able to add the PT100 to the sensor type dropdown I currently have with the other types of sensors.

cbaymac, 2018-03-16:
It looks like the RPi can interface with the MAX31865 as shown here.

https://www.raspberrypi.org/forums/viewtopic.php?t=69648

Also the folks at Craftbeer have a Raspberry PI PID controller which I
believe will work with the MAX31865 and the PT100.  Here is a link.

https://www.homebrewtalk.com/forum/threads/craftbeerpi-raspberry-pi-software.569497/page-23

kizniche, 2018-03-16:
Ah, I was confusing this with the MAX31856. This is a completely different chip. I'll look into integrating a module into Mycodo.

cbaymac, 2018-03-17:
> Can you test this code?

I checked it out and it provides the correct temp.  Screen shot attached.
The probe was in a glass of ice water. Van Dusen is correct. I had to
change the resistance value in the code from 400 to 430. I saw that in
another post and it worked for me.  I think it is the difference between
the PT100 and PT1000.  Do you think you can add this PID input to Mycodo?
That would be great. Thank you.

kizniche, 2018-03-17:
Can you test if this modified version works. I added the option to select the PT100 or PT1000. 

https://github.com/kizniche/Mycodo/blob/e608e56a54d85a06d1ba1e7329cbc2dfe078aaa0/mycodo/inputs/max31865.py

At the bottom of the file is where the pins should be changed to reflect your setup:

https://github.com/kizniche/Mycodo/blob/e608e56a54d85a06d1ba1e7329cbc2dfe078aaa0/mycodo/inputs/max31865.py#L230-L239

And the device type can me set to "PT100" or "PT1000"

https://github.com/kizniche/Mycodo/blob/e608e56a54d85a06d1ba1e7329cbc2dfe078aaa0/mycodo/inputs/max31865.py#L238

cbaymac, 2018-03-17:
Will do. I did get the PT100 working with CraftBeerPI by downloading this
https://github.com/thegreathoe/cbpi-pt100-sensor
There is also a PID auto-tune plug in which sounds interesting. Have not
tried it yet as the hot plate is not yet operational. It supposedly tells
you your optimal PID settings.
https://youtu.be/MveoQaDTAis
Thank you for all your efforts.

cbaymac, 2018-03-17:
Kyle,  your script works fine.  I had to change the PT100 resistance to 430
vs 400.  It seems to be an issue in many write ups. I’m not sure why 430 is
not the normal default.

kizniche, 2018-03-17:
> I did get the PT100 working with CraftBeerPI by downloading this https://github.com/thegreathoe/cbpi-pt100-sensor

Yes, that's the code I sent you previously. CraftBeerPi is using Stephen's code, as can be seen [here](https://github.com/thegreathoe/cbpi-pt100-sensor/blob/master/max31865.py#L4).

> There is also a PID auto-tune plug in which sounds interesting. Have not tried it yet as the hot plate is not yet operational. It supposedly tells you your optimal PID settings.

Auto-tuning is notoriously bad and there are a lot of pitfalls. For simple things like water temperature regulation it may work well because the response is slow.

It's interesting how similar CraftBeerPi looks like my interface. They've only been around recently, so I'd be surprised if they hadn't come across Mycodo.

cbaymac, 2018-03-17:
:o)

kizniche, 2018-03-17:
I released v5.6.6, so you can test the new input after obtaining the new code through the update menu.

cbaymac, 2018-03-17:
Excellent. I’ll give it a try.  I am using your Mycodo code to control
temperature, humidity and light in a fruiting chamber and incubation room.

The PT100 and the MAX31865 will PID control a pressure cooker on a heat
plate for sterilizing grain.  I will be using a normal thermocouple and the
MAX31855 for a 55 gallon sterilizer which I have yet to build. I plan to
run the sterilizer, which is a 55 gallon drum,  for 24 hours at 200 drgrees
Fahrenheit.

The pressure cooker PID is using a PI Zero W, MAX31865, PT100 and a 40 amp
SSR. Your Mycodo project really makes the job of controlling the
environment easy and accurate. The fact that you can control everything
over the internet and local WiFi network is great.  I can control the
entire operation from my iPhone while hanging out at the beach...sweet.
Mycodo has made a huge difference in my production and I’m just starting
out. Thank you for your great product  and rapid response to new features.

kizniche, 2018-03-18:
Glad to hear and glad to help. I've wanted to use Mycodo with a pressure cooker for a while, which would also double as a still.

zsole2, 2018-03-18:
I like the idea of adding autotuning, even though is not for all use cases. With both the sous-vide and the refrigerators, this would be a welcome addition.

cbaymac, 2018-03-19:
I’m about set up with the MaX31865 and the PT100 with the latest software
release but get an error that “no output defined” I don’t see how to say
that my GPIO pin 24 is the pin that controls the 110v outlet?  I can add
both the PT100 and the GPIO pin in function, but it won’t activate since
the data source is not...nor can I activate it. Photos attached.

kizniche, 2018-03-19:
I don't see any attachments.

kizniche, 2018-03-19:
Additionally, I did a search through the Mycodo code and there isn't a string “no output defined”, so you'll have to be a little more detailed with the steps you took and your settings.

cbaymac, 2018-03-19:
Two screen shots attached. I’m not seeing where to enter the GPIO PIN
number which controls the outlet for the hot plate. I can enter it in
function.

kizniche, 2018-03-19:
Still no images

kizniche, 2018-03-19:
You need to use a web browser to comment with images

cbaymac, 2018-03-19:
[image]
[image]

kizniche, 2018-03-19:
You don't have the options expanded so I can see what they are.

cbaymac, 2018-03-19:
[image]
[image]

cbaymac, 2018-03-19:
Pages expanded.  Look like I need a different skin to see the options better. Hope these pages help.

kizniche, 2018-03-19:
Okay. I found an issue with the MAX31865 code. I pushed a fix and made a new release that you can upgrade to.

Regarding the sensor, you should be able to activate it now.

Everything else looks good, although your setpoint for the PID is 25C, which I don't think is hot enough for your pressure cooker to pressure cook.

cbaymac, 2018-03-19:
Correct on the temp...I'm easing into this one ;o) Than you.

cbaymac, 2018-03-19:
Everything activates properly, but I am unable to change the “resistor
reference data”   It defaults back to 0 (zero) once saved. I believe the
proper way to calibrate the sensor in a glass of ice water.  Then adjust
the resistor value to read 0C.  I just can’t change the value.  Other
values in the table do change.

kizniche, 2018-03-19:
It's a minor fix. You can see the edit above in this thread. If you want to edit that file, add the new line, save, then restart the frontend from the Config -> Restart Frontend, you'll be able to use the new code.

cbaymac, 2018-03-19:
I think I can do that. Should be fun. I’ll let you know how it goes.

cbaymac, 2018-03-20:
That line of code did the trick.  Working like a champ. Thank you.

kizniche, 2018-03-20:
Good to hear. Thanks for testing and confirming it works.

cbaymac, 2018-03-24:
It seems this line of code did not make it into the last update or maybe
there has not been one. I still needed to add the code to get the 430
resistor value to stick in the PT100 code.

Gordon McPheter
757-790-9042

kizniche, 2018-03-24:
I haven't made a release since the fix was committed to github. I'll try to have a release go out today.

cbaymac, 2018-03-24:
Understand, I must have had an older backup I was looking at. No hurry.